networkupstools/nut #180: Smart-UPS RT 10000 XL support with apcsmart

surr, 2015-01-16:
I use Smart-UPS RT 10000 XL device with debian packaged nut.
After upgrading from debian 6 to 7 (and nut from 2.6.4 to 2.7.2) a lot of features disappear.
For example, there is only voltage value under input.* hierarchy in upsc output and upscmd -l returns an empty list.
All missing things worked fine with previous version of nut. I have to recalibrate device after battery replacement, but I do not know how to do it now.

[Link to device at vendor site.](http://www.apc.com/products/resource/include/techspec_index.cfm?base_sku=SURT10000RMXLI&tab=documentation)
The device is equipped with DB-9 connector, not a RJ-45 one.

It may be irrelevant, but somewhat similar model Smart-UPS 3000 RM works as before after upgrade. This device returns a lot of data and all commands are accessible.

clepple, 2015-01-18:
As mentioned in the NEWS file (available at /usr/share/doc/nut-server/NEWS.gz in Debian) and in the man page, we replaced the apcsmart driver with a new version. The version from NUT 2.6.4 is still available as apcsmart-old.

However, apcsmart-old will eventually be removed from NUT. Can you post the output of `upsc`?

Tagging @msoltyspl in case I missed something.

surr, 2015-01-18:
Output with apcsmart driver:

```
battery.charge: 100.0
battery.voltage: 218.1
device.mfr: APC
device.model: Smart-UPS RT 10000 XL   
device.serial: B21130002653  
device.type: ups
driver.name: apcsmart
driver.parameter.cable: 940-0024B
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ups
driver.parameter.sdtype: 1
driver.version: 2.6.4
driver.version.internal: 3.04
input.voltage: 223.2
output.current: 07.29
output.voltage: 231.6
ups.firmware: 476.18.W
ups.load: 015.0
ups.mfr: APC
ups.model: Smart-UPS RT 10000 XL   
ups.serial: B21130002653  
ups.status: OL
ups.temperature: 024.6
```

Output with apcsmart-old driver:

```
battery.alarm.threshold: 0
battery.charge: 100.0
battery.charge.restart: 00
battery.date: 07/22/11
battery.packs: 001
battery.packs.bad: 000
battery.runtime: 3960
battery.runtime.low: 120
battery.voltage: 219.2
battery.voltage.nominal: 192
device.mfr: APC
device.model: Smart-UPS RT 10000 XL   
device.serial: B21130002653  
device.type: ups
driver.name: apcsmart-old
driver.parameter.cable: 940-0024B
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ups
driver.parameter.sdtype: 1
driver.version: 2.6.4
driver.version.internal: 2.1
input.frequency: 49.99
input.quality: FF
input.sensitivity: A
input.transfer.high: 253
input.transfer.low: 161
input.transfer.reason: simulated power failure or UPS test
input.voltage: 224.6
input.voltage.maximum: 224.6
input.voltage.minimum: 220.3
output.current: 07.29
output.voltage: 230.4
output.voltage.nominal: 230
ups.delay.shutdown: 020
ups.delay.start: 000
ups.display.language: E
ups.firmware: 476.18.W
ups.firmware.old: 5YW
ups.id: UPS_IDEN
ups.load: 015.0
ups.mfr: APC
ups.mfr.date: 07/22/11
ups.model: Smart-UPS RT 10000 XL   
ups.serial: B21130002653  
ups.status: OL
ups.temperature: 024.6
ups.test.interval: 1209600
ups.test.result: NO
```

surr, 2015-01-18:
And the actual upgrade was from 2.4.3 to 2.6.4, of course.
2.7.2 is the version from the upcoming debian release.

clepple, 2015-01-21:
You could try the following: stop the driver, then run '/lib/nut/apcsmart -D -a name-of-UPS 2>&1 | tee /tmp/apc-log.txt', then post the log in a Gist or pastebin, and post the URL here.

I think there is a firmware table that isn't matching your version.

surr, 2015-01-21:
```
   0.000000	debug level is '3'
   0.076022	attempting firmware lookup using command 'V'
   0.156413	firmware: [5YW]
   0.156429	firmware not found in compatibility table - trying normal method
   0.156433	APC - attempting to find command set
   1.096035	valid_cmd: cmd [a] failed regex match
   1.096058	APC - trying to handle unknown model
   2.985510	detected Matrix-UPS [B21130002653  ] on /dev/ups
   3.055928	update_info_all: starting
   4.745038	update_info_all: done
   4.745136	dstate_init: sock /var/run/nut/apcsmart-ups open on fd 5
   4.865073	update_info_normal: starting
   5.764635	update_info_normal: done
```

clepple, 2015-01-22:
On Jan 18, 2015, at 9:25 AM, surr <notifications@github.com> wrote:
> And the actual upgrade was from 2.4.3 to 2.6.4, of course.
> 2.7.2 is the version from the upcoming debian release.
> 
Oh, then that means that we had a different APC driver for 2.4.3, 2.6.4 and 2.7.2.

Can you install the 2.7.2 version from testing? I would hate to try and fix a bug in what is now the old driver.

surr, 2015-01-22:
It is hard to install new version from debian, so I build nut from git with these parameters.

    ./configure --without-all --with-serial --prefix=/tmp/nut --with-statepath=/tmp/nut/_

upsc output

```
battery.charge: 100.0
battery.voltage: 219.2
device.mfr: APC
device.model: Smart-UPS RT 10000 XL   
device.serial: B21130002653  
device.type: ups
driver.name: apcsmart
driver.parameter.cable: 940-0024B
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ups
driver.parameter.sdtype: 1
driver.version: 2.7.2-signed-145-g26d096f
driver.version.internal: 3.1
input.voltage: 227.5
output.current: 07.29
output.voltage: 230.4
ups.firmware: 476.18.W
ups.load: 016.0
ups.mfr: APC
ups.model: Smart-UPS RT 10000 XL   
ups.serial: B21130002653  
ups.status: OL
ups.temperature: 022.9
```

Debug log:

```
Network UPS Tools - APC Smart protocol driver 3.1 (2.7.2-signed-145-g26d096f)
APC command table version 3.1
   0.000000	debug level is '1'
   0.076376	attempting firmware lookup using [V]
   0.156374	detected firmware version: 5YW
   0.156433	attempting var/cmdset lookup using [a]
   1.096819	very old or unknown APC model, support will be limited
   1.096848	preread_data: ups.temperature [C]
   1.216747	ups.temperature [C] - variable supported
   1.216761	preread_data: ups.load [P]
   1.326769	ups.load [P] - variable supported
   1.326797	preread_data: input.voltage [L]
   1.496736	input.voltage [L] - variable supported
   1.496748	preread_data: output.voltage [O]
   1.606750	output.voltage [O] - variable supported
   1.606763	preread_data: battery.charge [f]
   1.737142	battery.charge [f] - variable supported
   1.737153	preread_data: battery.voltage [B]
   1.896782	battery.voltage [B] - variable supported
   1.896813	preread_data: ups.model [0x01]
   2.207153	ups.model [0x01] - variable supported
   2.207179	preread_data: ups.serial [n]
   2.467158	ups.serial [n] - variable supported
   2.467185	preread_data: output.current [/]
   2.647135	output.current [/] - variable supported
   2.647151	preread_data: ups.firmware [b]
   2.777179	ups.firmware [b] - variable combination supported
   2.777194	detected Matrix-UPS [B21130002653  ] on /dev/ups
   2.777201	update_status: [Q]
   2.897141	update_info: starting scan (all vars)
   2.897161	poll_data: ups.temperature [C]
   3.007150	poll_data: ups.load [P]
   3.117135	poll_data: input.voltage [L]
   3.237123	poll_data: output.current [/]
   3.347127	poll_data: output.voltage [O]
   3.517558	poll_data: battery.charge [f]
   3.627115	poll_data: battery.voltage [B]
   3.797106	poll_data: ups.serial [n]
   4.007514	poll_data: ups.model [0x01]
   4.367490	poll_data: ups.firmware [b]
   4.547491	update_info: scan completed
   4.547585	update_status: [Q]
   4.617503	update_info: starting scan
   4.617517	poll_data: ups.temperature [C]
   4.767515	poll_data: ups.load [P]
   4.877499	poll_data: input.voltage [L]
   5.047491	poll_data: output.current [/]
   5.157508	poll_data: output.voltage [O]
   5.327496	poll_data: battery.charge [f]
   5.437505	poll_data: battery.voltage [B]
   5.607506	update_info: scan completed
```

clepple, 2015-01-23:
>    0.076376 attempting firmware lookup using [V]
>    0.156374 detected firmware version: 5YW
>    0.156433 attempting var/cmdset lookup using [a]
>    1.096819 very old or unknown APC model, support will be limited
What we need to do here is figure out if the "a" result looks good, or if we need to add "5YW" to the compatibility table.

I created a temporary branch to investigate: https://github.com/clepple/nut/compare/apcsmart_logging

Can you please switch to that branch (it's in my own repository rather than the main one, so that it doesn't spin up a bunch of Buildbot jobs), then rebuild and post just the output after "very old or unknown APC model" from the new branch?

surr, 2015-01-23:
```
   1.145806	APC_CMDSET: (86 bytes) => 34 2e 21 24 25 2b 3f 3d 23 7c 2e 01 05 09 0c 0e
   1.145816	 14 16 1a 27 29 2b 2d 2f 37 38 39 3c 3e 40 41 42 43 44 45 46 47 4b 4c 4d 4e
   1.145824	 4f 50 51 52 53 55 56 57 58 59 5a 5c 5e 61 62 63 65 66 67 6a 6b 6c 6d 6e 6f
   1.145832	 70 71 72 73 75 77 78 79 7a 7e 7f 2e 81 3a d0 e2 f0 fd fe ff
```

clepple, 2015-01-31:
It looks like the APC_CMDSET_FMT regex is only expecting two "." characters, and there are three:

   1.145806 APC_CMDSET: (86 bytes) => 34 2e 21 24 25 2b 3f 3d 23 7c 2e 01 05 09 0c 0e
   1.145816  14 16 1a 27 29 2b 2d 2f 37 38 39 3c 3e 40 41 42 43 44 45 46 47 4b 4c 4d 4e
   1.145824  4f 50 51 52 53 55 56 57 58 59 5a 5c 5e 61 62 63 65 66 67 6a 6b 6c 6d 6e 6f
   1.145832  70 71 72 73 75 77 78 79 7a 7e 7f 2e 81 3a d0 e2 f0 fd fe ff
I don't have any other logs showing what this should look like, but maybe I can find some more info in the apcupsd documentation.=

clepple, 2015-02-04:
On 2015-01-31 17:00, Charles Lepple wrote:
> It looks like the APC_CMDSET_FMT regex
> <https://github.com/networkupstools/nut/blob/master/drivers/apcsmart.h#L146> is
> only expecting two "." characters, and there are three:
>
> |    1.145806 [...]

Sorry for late reply. Yes it seems to be the cause of problem. Current 
version assumed that command set was returned by 'a' was:

version.alerts.commands

Old driver searched for '.' from the beginning of the string, the new 
one verified by regex and searched for '.' from the back, obviously 
failing, as new(?) version is apparently:

version.alerts.commands.something_apc_added

Version was also bumped to 4.

I'll fix it tommorow.

msoltyspl, 2015-02-13:
Pushed small update, it should be working fine now.

Sorry for delay.

surr, 2015-02-13:
I get a SEGV now.

```
Network UPS Tools - APC Smart protocol driver 3.1 (2.7.2-signed-163-g07e2f13)
APC command table version 3.1
   0.000000	debug level is '1'
   0.119102	attempting firmware lookup using [V]
   0.199099	detected firmware version: 5YW
   0.199221	attempting var/cmdset lookup using [a]
   1.139119	parsing out supported cmds/vars
   1.139136	preread_data: ups.model [0x01]
   1.519062	ups.model [0x01] - variable supported
   1.519089	[0x05] unrecognized
   1.519106	[0x09] unrecognized
   1.519123	preread_data: ups.display.language [0x0c]
   1.589085	ups.display.language [0x0c] - variable supported
   1.589114	load.on [0x0e] - command supported
   1.589121	[0x14] unrecognized
   1.589125	[0x16] unrecognized
   1.589130	preread_data: output.current [/]
   1.699069	output.current [/] - variable supported
   1.699096	[7] unrecognized
   1.699101	preread_data: input.quality [9]
   1.769024	input.quality [9] - variable supported
   1.769034	preread_data: battery.packs.bad [<]
   1.859486	battery.packs.bad [<] - variable supported
   1.859510	preread_data: battery.packs [>]
   1.999471	battery.packs [>] - variable supported
   1.999490	shutdown.return [@] - command supported
   1.999497	test.panel.start [A] - command supported
   1.999502	preread_data: battery.voltage [B]
   2.109059	battery.voltage [B] - variable supported
   2.109077	preread_data: ups.temperature [C]
   2.219043	ups.temperature [C] - variable supported
   2.219060	calibrate.start [D] - command supported
   2.219067	calibrate.stop [D] - command supported
   2.219071	preread_data: ups.test.interval [E]
   2.349494	ups.test.interval [E] - variable supported
   2.349522	preread_data: input.frequency [F]
   2.459033	input.frequency [F] - variable supported
   2.459068	preread_data: input.transfer.reason [G]
   2.519031	input.transfer.reason [G] - variable supported
   2.519048	shutdown.stayoff [K] - command supported
   2.519054	preread_data: input.voltage [L]
   2.659036	input.voltage [L] - variable supported
   2.659054	preread_data: input.voltage.maximum [M]
   2.769051	input.voltage.maximum [M] - variable supported
   2.769066	preread_data: input.voltage.minimum [N]
   2.939446	input.voltage.minimum [N] - variable supported
   2.939456	preread_data: output.voltage [O]
   3.049022	output.voltage [O] - variable supported
   3.049038	preread_data: ups.load [P]
   3.219003	ups.load [P] - variable supported
   3.219034	shutdown.return [S] - command supported
   3.219054	test.failure.start [U] - command supported
   3.219075	test.battery.start [W] - command supported
   3.219093	test.battery.stop [W] - command supported
   3.219111	preread_data: ups.test.result [X]
   3.289414	ups.test.result [X] - variable supported
   3.289429	load.off [Z] - command supported
   3.289435	[\] unrecognized
   3.289440	bypass.start [^] - command supported
   3.289446	bypass.stop [^] - command supported
   3.289451	preread_data: ups.id [c]
   3.438983	ups.id [c] - variable supported
   3.438993	preread_data: battery.charge.restart [e]
   3.519020	battery.charge.restart [e] - variable supported
   3.519043	preread_data: battery.charge [f]
   3.639433	battery.charge [f] - variable supported
   3.639457	preread_data: battery.voltage.nominal [g]
   3.719423	battery.voltage.nominal [g] - variable supported
   3.719436	preread_data: battery.runtime [j]
   3.828999	battery.runtime [j] - variable supported
   3.829016	preread_data: battery.alarm.threshold [k]
   3.939408	battery.alarm.threshold [k] - variable supported
   3.939421	preread_data: input.transfer.low [l]
   4.039397	input.transfer.low [l] - variable supported
   4.039407	preread_data: ups.mfr.date [m]
   4.189403	ups.mfr.date [m] - variable supported
   4.189415	preread_data: ups.serial [n]
   4.449398	ups.serial [n] - variable supported
   4.449417	preread_data: output.voltage.nominal [o]
   4.549396	output.voltage.nominal [o] - variable supported
   4.549411	preread_data: ups.delay.shutdown [p]
   4.629411	ups.delay.shutdown [p] - variable supported
   4.629421	preread_data: battery.runtime.low [q]
   4.699394	battery.runtime.low [q] - variable supported
   4.699409	preread_data: ups.delay.start [r]
   4.829814	ups.delay.start [r] - variable supported
   4.829829	preread_data: input.sensitivity [s]
   4.889373	input.sensitivity [s] - variable supported
   4.889398	preread_data: input.transfer.high [u]
   4.989367	input.transfer.high [u] - variable supported
   4.989392	[w] unrecognized
   4.989409	preread_data: battery.date [x]
   5.149380	battery.date [x] - variable supported
   5.149407	preread_data: ups.firmware [b]
   5.289440	ups.firmware [b] - variable combination supported
   5.289465	parsing out caps
   7.659747	output.voltage.nominal [o(W)] - capability supported
   7.659803	input.transfer.high [u(W)] - capability supported
   7.659826	input.transfer.low [l(W)] - capability supported
   7.659847	battery.charge.restart [e(4)] - capability supported
   7.659877	input.sensitivity [s(4)] - capability supported
   7.659887	battery.runtime.low [q(4)] - capability supported
   7.659918	ups.delay.shutdown [p(4)] - capability supported
   7.659946	battery.alarm.threshold [k(4)] - capability supported
   7.659962	ups.delay.start [r(4)] - capability supported
   7.659985	ups.test.interval [E(4)] - capability supported

Program received signal SIGSEGV, Segmentation fault.
0x0000000000404673 in apc_getcaps (qco=0) at apcsmart.c:1007
1007			} else if (vt->flags & APC_PACK)
(gdb) bt
#0  0x0000000000404673 in apc_getcaps (qco=0) at apcsmart.c:1007
#1  0x0000000000404f13 in getbaseinfo () at apcsmart.c:1262
#2  0x0000000000406a6c in upsdrv_initinfo () at apcsmart.c:2067
#3  0x0000000000407d4f in main (argc=0, argv=0x7fffffffeb88) at main.c:633
```

surr, 2015-02-13:
Local variables at the fail location.

```
(gdb) info locals 
ptr = 0x7fffffffe371 "\360\064\065\065AUTO 60/.160/3 50/.150/3 "
entptr = 0x7fffffffe375 "AUTO 60/.160/3 50/.150/3 "
upsloc = 87 'W'
temp = "#oW53230240200208220uW43253242265276lW43161196184173e4820015253550607590s411Aq4820205071012151820p483020060120240480720960000k4410TLNr483000020060120240480720960E443336168ON OFF\360\064\065\065AUTO 60/.160/3 50/.1"...
cmd = -16 '\360'
loc = 52 '4'
etmp = "OFF", '\000' <repeats 28 times>
endtemp = 0x7fffffffe38e ""
nument = 5
entlen = 5
i = 4
matrix = 0
ret = 206
valid = 0
vt = 0x0
__func__ = "apc_getcaps"
```

msoltyspl, 2015-02-14:
SEGV should be fixed. Check if it works fine when you have the opportunity.

surr, 2015-02-14:
It works fine now. Thanks.

closed 2015-02-14 by surr