networkupstools/nut #183: Ippon Smart Winner 1000 USB

Am1GO, 2015-02-04:
Greetings,

I can't get my UPS to work with nut.
I'm not pretty sure that it was a bug in NUT but I'll report it anyway.
Thank you for your analysis!

-
#### Repoducibility:
100%

#### Actual results:
##### ```# /usr/sbin/nutdrv_qx -a ismw1000_nutdrv_s```
```
Network UPS Tools - Generic Q* USB/Serial driver 0.06 (2.7.2)
USB communication driver 0.32
Device not supported!
Device not supported!
```
##### ```# /usr/sbin/blazer_usb -a ismw1000_blazer_s```
```
Network UPS Tools - Megatec/Q1 protocol USB driver 0.11 (2.7.2)
No supported UPS detected
```
#### Expected results:
##### ```# /usr/sbin/nutdrv_qx -a ismw1000_nutdrv_s```
```
Network UPS Tools - Generic Q* USB/Serial driver 0.06 (2.7.2)
USB communication driver 0.32
Voltage, current, etc..
```
##### ```# /usr/sbin/blazer_usb -a ismw1000_blazer_s```
```
Network UPS Tools - Megatec/Q1 protocol USB driver 0.11 (2.7.2)
Voltage, current, etc..
```
#### Additional information:
##### ```# cat /etc/issue```
```
Pidora release 2014 (Raspberry Pi Fedora Remix)
Kernel \r on an \m (\l)
```
##### ```# rpm -qa | grep -P '^nut'```
```
nut-client-2.7.2-1.fc20.armv6hl
nut-2.7.2-1.fc20.armv6hl
```
##### ```# cat /proc/version```
```
Linux version 3.18.5+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #746 PREEMPT Mon Feb 2 13:57:16 GMT 2015
```
##### ```# ll /dev/bus/usb/001/```
```
total 0
 3473 0 drwxr-xr-x 2 root root        120 фев  4 19:49 .
 3472 0 drwxr-xr-x 3 root root         60 янв  1  1970 ..
 3474 0 crw-rw-r-- 1 root root    189,  0 янв  1  1970 001
 3548 0 crw-rw-r-- 1 root root    189,  1 янв  1  1970 002
 3594 0 crw-rw-r-- 1 root root    189,  2 янв  1  1970 003
10001 0 crw-rw-r-- 1 root dialout 189, 20 фев  4 19:49 021
```
##### ```# cat /etc/group | grep nut```
```
dialout:x:18:nut
nut:x:57:
```
##### ```#``` [lsusb -v](https://gist.github.com/Am1GO/9728e372fb3dd83a43ee)
##### ```#``` [cat /etc/ups/ups.conf](https://gist.github.com/Am1GO/2c66514cd95515fa9dd0)
##### ```#``` [strace](https://gist.github.com/Am1GO/d92fad12bc24262055f7) -f -s 111111 -o blazer.strace [/usr/sbin/blazer_usb](https://gist.github.com/Am1GO/429d2adfe5f2949f05eb) -DDDDDDDDDDDDDDDDDDDDD -a ismw1000_blazer
##### ```#``` [strace](https://gist.github.com/Am1GO/50301729bf7909cdd904) -f -s 111111 -o nutdrv.strace [/usr/sbin/nutdrv_qx](https://gist.github.com/Am1GO/6ba31feb16da2723d020) -DDDDDDDDDDDDDDDDDDDDD -a ismw1000_nutdrv
##### ```#``` [strace](https://gist.github.com/Am1GO/c4d9ff3aaf5fd1de6c89) -f -s 111111 -o blazer_s.strace [/usr/sbin/blazer_usb](https://gist.github.com/Am1GO/9d04995b0c7cf0cc1b4f) -DDDDDDDDDDDDDDDDDDDDD -a ismw1000_blazer_s
##### ```#``` [strace](https://gist.github.com/Am1GO/638ea52b185f052de547) -f -s 111111 -o nutdrv_s.strace [/usr/sbin/nutdrv_qx](https://gist.github.com/Am1GO/21af92f3b9c564b34317) -DDDDDDDDDDDDDDDDDDDDD -a ismw1000_nutdrv_s

I've also tried to use two other USB<->USB-AB cables with no success.
#### Workaround
Serial connection with blazer_ser works fine but I need GPIO for other purposes.

clepple, 2015-02-09:
> Serial connection with blazer_ser works fine but I need GPIO for other purposes.

Did you completely power down the UPS after testing the serial connection? Many UPSes cannot switch between USB and serial on-the-fly.

Am1GO, 2015-02-09:
Yes, I've tried to power off UPS completely without any success.

clepple, 2015-02-09:
I assume the serial driver detected the megatec protocol variant?

Is this using libusb 1.0.x with libusb-compat? Most of the testing has been done with libusb-0.1.

Maybe @zykh has some other debugging ideas.

Am1GO, 2015-02-09:
Accordings to libusb package description [1] it was libusb 1.0.x:
```
This package provides a way for applications to access USB devices. Legacy libusb-0.1 is no longer supported by upstream, therefore content of this package was replaced by libusb-compat. It provides compatibility layer allowing applications written for libusb-0.1 to work with libusb-1.0.
```
But I'm not really sure because in Fedora it could be virtually anything and only maintainer knows the truth.

Here is the build logs of packages that I was using:
* [nut-2.7.2-1.fc20.armv6hl](http://japan.proximity.on.ca/kojifiles/packages/nut/2.7.2/1.fc20/data/logs/armv6hl/build.log)
* [libusb-0.1.5-3.fc20.armv6hl](http://japan.proximity.on.ca/kojifiles/packages/libusb/0.1.5/3.fc20/data/logs/armv6hl/build.log)

Which OS would you recommend to test my UPS with (only package-based distribs please)?

[1] http://koji.pidora.ca/koji/buildinfo?buildID=88980

clepple, 2015-02-10:
> Here is the build logs of packages that I was using:
> 
> nut-2.7.2-1.fc20.armv6hl
> libusb-0.1.5-3.fc20.armv6hl
The "libusb-0.1.5" log does mention libusb-compat. I am not saying that it won't work, just that it is an extra layer of complexity.
> Which OS would you recommend to test my UPS with (only package-based distribs please)?
> 

My personal preference would be Debian or Ubuntu (they still have libusb-0.1 as a separate library from libusb-compat), but even with Fedora it should be possible - it just might involve rebuilding the NUT package from source.

"megatec" uses some sort of "Q1" commands to query UPS status, but I think things are slightly different over USB. @zykh is much more familiar with this than I am.