networkupstools/nut #185: insecure storage of password in nut-monitor favorites file

bigon, 2015-02-13:
Hello,

As reported in [0], it seems that the favorite file can be readable by everybody in some situations.

Newly created directories are properly chmod'ed 0700, but the one created before the version 1.2 of nut-monitor are world readable. The debian bug is proposing a patch

[0] https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=777706

clepple, 2015-02-14:
David (cc'd),

I think the following patch from the Debian bug report is useful:

https://github.com/clepple/nut/commit/0d1b10de7e69cc99774db0573d5ad4ffdbc891c8

If you would like, I can bump the version number to 1.3.1.

- Charles

clepple, 2015-02-16:
Hi,

> I think the following patch from the Debian bug report is useful:
> 
> https://github.com/clepple/nut/commit/0d1b10de7e69cc99774db0573d5ad4ffdbc891c8
> [3]
> 
> If you would like, I can bump the version number to 1.3.1.

Yes it's a good way to have a minimum security on the config folder.

May be for future version, i can try to add some crypto to passwords to 
avoid
having them "human-readable".

For now i just encoded them as base64 that is not supposed to be 
"secure" but
just avoids reading the password in clear text when looking the 
favorites file.

Another solution would be to use a system keyring (Gnome/KDE) but it 
would
break the initial multi-platform goal of NUT-Monitor.

So applying this patch is a good starting point ;)

Thanks.

closed 2015-03-05 by clepple