networkupstools/nut #186: Can't get KIN2200APRM working

ikus060, 2015-02-15:
I'm struggling to get my PowerCom UPS model KIN2200APRM working with NUT.

My configuration `ups.conf`:
```
[myups]
        driver = powercom
        port = /dev/ttyS0
        desc = "PowerCom KIN 2200 AP RM"
        type = KIN
        manufacturer = PowerCom
        modelname = KIN2200APRM
```

When running powercom from command line:
```
# /lib/nut/powercom -D -D -D -D -D -q -a myups
Network UPS Tools - PowerCom protocol UPS driver 0.13 (2.6.4)
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/ttyS0"
   0.000056	send_to_all: SETINFO driver.parameter.type "KIN"
   0.000085	send_to_all: SETINFO driver.parameter.manufacturer "PowerCom"
   0.000103	send_to_all: SETINFO driver.parameter.modelname "KIN2200APRM"
   0.000132	debug level is '5'
   0.000702	Flow control disable
   0.000796	Num of bytes requested for reading from UPS: 16
   3.003855	data receiving error (0 instead of 16 bytes)
   3.003889	send_to_all: SETINFO device.type "ups"
   3.003898	send_to_all: SETINFO driver.version "2.6.4"
   3.003904	send_to_all: SETINFO driver.version.internal "0.13"
   3.003909	send_to_all: SETINFO driver.name "powercom"
   3.003915	send_to_all: SETINFO ups.mfr "PowerCom"
   3.003920	send_to_all: SETINFO ups.model "KIN2200APRM"
   3.003926	send_to_all: SETINFO ups.serial "Unknown"
   3.003931	send_to_all: SETINFO ups.model.type "KIN"
   3.003936	send_to_all: SETINFO input.voltage.nominal "230"
   3.003941	send_to_all: ADDCMD test.battery.start
   3.003945	send_to_all: ADDCMD shutdown.return
   3.003949	send_to_all: ADDCMD shutdown.stayoff
   3.004019	Num of bytes requested for reading from UPS: 16
   6.007064	data receiving error (0 instead of 16 bytes)
   6.007163	dstate_init: sock /var/run/nut/powercom-myups open on fd 5
   6.007176	send_to_all: SETINFO driver.parameter.pollinterval "2"
   6.007183	send_to_all: SETINFO device.mfr "PowerCom"
   6.007189	send_to_all: SETINFO device.model "KIN2200APRM"
   6.007195	send_to_all: SETINFO device.serial "Unknown"
   6.007265	Num of bytes requested for reading from UPS: 16
   9.010314	data receiving error (0 instead of 16 bytes)
   9.010405	Num of bytes requested for reading from UPS: 16
  12.013453	data receiving error (0 instead of 16 bytes)
  12.013545	Num of bytes requested for reading from UPS: 16
```

clepple, 2015-02-15:
> 6.007265 Num of bytes requested for reading from UPS: 16
> 9.010314 data receiving error (0 instead of 16 bytes)

Assuming the serial protocol hasn't changed, it looks like the UPS is not responding to the query character.

Some UPS vendors have different cables for their smart protocols (versus contact-closure signaling). If it is a custom-built cable, double-check the pinouts.

/dev/ttyS0 sounds like Linux, so you might be able to check the stats in /proc/sys/tty to see if the port is sending or receiving any characters. IRQ conflicts aren't common on the first port, but check the output of dmesg against any BIOS messages. You should also get locking-related error messages if another process is trying to use that port, but not all third-party packages do this properly.

ikus060, 2015-02-15:
You might be right about the cable. I've use a simple DB-9 cable. According to documentation, http://www.powercom-usa.com/techsup/UPSManual/kingpro.pdf I need to use a special cable. I will search my cable stack for this cable, but I might not have it anymore. Any suggestion ?

[image]

clepple, 2015-02-15:
We don't have it in our cable documentation, but let us know if this works: http://pinouts.ru/UPS/powercom_cable_pinout.shtml

ikus060, 2015-02-15:
hey hey, I found the cable and everything work like a charm. ! Thanks

closed 2015-02-15 by ikus060

clepple, 2015-02-15:
Excellent, glad that worked. If you get a chance to verify the pinout URL, we can include that link in the documentation.