networkupstools/nut #187: Wrong in/out voltage repported for PowerCom KIN2200APRM

ikus060, 2015-02-15:
I'm using a PowerCom KIN-2200-AP-RM. The input/ouput voltage reported is incorrect. Looking at current implementation and comparing it with the Java implementation provided by PowerCom, it doesn't take the line voltage into consideration. 

Solution:
The current value need to be divided by two when the line voltage is less then 200.

```
# /lib/nut/powercom -D -D -D -D -D -q -a myups
Network UPS Tools - PowerCom protocol UPS driver 0.13 (2.6.4)
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/ttyS0"
   0.000055	send_to_all: SETINFO driver.parameter.type "KIN"
   0.000077	send_to_all: SETINFO driver.parameter.manufacturer "PowerCom"
   0.000095	send_to_all: SETINFO driver.parameter.modelname "KIN2200APRM"
   0.000125	debug level is '5'
   0.000734	Flow control disable
   0.000825	Num of bytes requested for reading from UPS: 16
   0.145583	Num of bytes received from UPS: 16
Raw data from UPS:
 0 0x0b ( )
 1 0xd5 (�)
 2 0x98 (�)
 3 0x98 (�)
 4 0x51 (Q)
 5 0x21 (!)
 6 0x50 (P)
 7 0x00 ( )
 8 0x00 ( )
 9 0x00 ( )
10 0x00 ( )
11 0x4b (K)
12 0xb3 (�)
13 0x00 ( )
14 0x00 ( )
15 0x00 ( )
   0.145759	Detected: KIN-2200AP , 120V
   0.145838	Values of arguments:
   0.145856	 manufacturer            : 'PowerCom'
   0.145873	 model name              : 'KIN2200APRM'
   0.145890	 serial number           : 'Unknown'
   0.145907	 line voltage            : '120'
   0.145923	 type                    : 'KIN'
   0.145940	 number of bytes from UPS: '16'
   0.145957	 method of flow control  : 'no_flow_control'
   0.145976	 validation sequence: '{{11,0x4b},{8,0},{8,0}}'
   0.145994	 shutdown arguments: '{{1,30},y}'
   0.146017	send_to_all: SETINFO device.type "ups"
   0.146043	send_to_all: SETINFO driver.version "2.6.4"
   0.146063	send_to_all: SETINFO driver.version.internal "0.13"
   0.146084	send_to_all: SETINFO driver.name "powercom"
   0.146106	send_to_all: SETINFO ups.mfr "PowerCom"
   0.146126	send_to_all: SETINFO ups.model "KIN2200APRM"
   0.146147	send_to_all: SETINFO ups.serial "Unknown"
   0.146167	send_to_all: SETINFO ups.model.type "KIN"
   0.146188	send_to_all: SETINFO input.voltage.nominal "120"
   0.146208	send_to_all: ADDCMD test.battery.start
   0.146226	send_to_all: ADDCMD shutdown.return
   0.146245	send_to_all: ADDCMD shutdown.stayoff
   0.146327	Num of bytes requested for reading from UPS: 16
   0.299491	Num of bytes received from UPS: 16
Raw data from UPS:
 0 0x0b ( )
 1 0xd5 (�)
 2 0x97 (�)
 3 0x16 ( )
 4 0x50 (P)
 5 0x21 (!)
 6 0x50 (P)
 7 0x00 ( )
 8 0x00 ( )
 9 0x01 ( )
10 0x04 ( )
11 0x4b (K)
12 0xb3 (�)
13 0x00 ( )
14 0x00 ( )
15 0x00 ( )
   0.299607	input.frequency   (raw data): [raw: 80]
   0.299625	send_to_all: SETINFO input.frequency "60.09"
   0.299641	input.frequency: 60.09
   0.299652	output.frequency   (raw data): [raw: 80]
   0.299667	send_to_all: SETINFO output.frequency "60.09"
   0.299679	output.frequency: 60.09
   0.299690	ups.load  (raw data): [raw: 11]
   0.299705	send_to_all: SETINFO ups.load "17.3"
   0.299717	ups.load: 17.3
   0.299728	battery.charge (raw data): [raw: 213]
   0.299742	send_to_all: SETINFO battery.charge "100.0"
   0.299754	battery.charge: 100.0
   0.299765	input.voltage (raw data): [raw: 151]
   0.299780	send_to_all: SETINFO input.voltage "245.4"
   0.299792	input.voltage: 245.4
   0.299803	output.voltage (raw data): [raw: 22]
   0.299817	send_to_all: SETINFO output.voltage "228.2"
   0.299831	output.voltage: 228.2
   0.299845	send_to_all: SETINFO ups.status "OB TEST"
   0.299858	STATUS: OB TEST
   0.299869	send_to_all: DATAOK
   0.299924	dstate_init: sock /var/run/nut/powercom-myups open on fd 5
   0.299939	send_to_all: SETINFO driver.parameter.pollinterval "2"
   0.299953	send_to_all: SETINFO device.mfr "PowerCom"
   0.299966	send_to_all: SETINFO device.model "KIN2200APRM"
   0.299979	send_to_all: SETINFO device.serial "Unknown"
   0.300049	Num of bytes requested for reading from UPS: 16
   0.445233	Num of bytes received from UPS: 16
```

Java implementation:
```java
      if (iVoltage < 200) {
        if (iVA <= 625) {
          Involtage = (paramArrayOfInt[2] * 0.89D + 6.18D);
        }
        if ((iVA >= 800) && (iVA < 2000)) {
          Involtage = (paramArrayOfInt[2] * 1.61D / 2.0D);
        }
        if (iVA >= 2000) {
          Involtage = (paramArrayOfInt[2] * 1.625D / 2.0D);
        }
      }
      if (iVoltage >= 200) {
        if (iVA <= 625) {
          Involtage = (paramArrayOfInt[2] * 1.79D + 3.35D);
        }
        if ((iVA >= 800) && (iVA < 2000)) {
          Involtage = (paramArrayOfInt[2] * 1.61D);
        }
        if (iVA >= 2000) {
          Involtage = (paramArrayOfInt[2] * 1.625D);
        }
      }
```

clepple, 2015-02-16:
I agree that the input_voltage() calculation can be divided by 2:

https://github.com/networkupstools/nut/blob/master/drivers/powercom.c#L409

What about the output voltage? The logic is a bit more complicated there (stat* bits):

https://github.com/networkupstools/nut/blob/master/drivers/powercom.c#L467

ikus060, 2015-02-16:
Inside the `if (statINV==0)` the basically the same logic: divide by two.
https://github.com/networkupstools/nut/blob/master/drivers/powercom.c#L469

I have extract the relevant java code from ReceiveCOM1.java. At first look all the constants seams to be divided or multiply by two. 
```java
  double[] r1DataX1v = { 1.0D, 1.0D, 1.0D, 1.0D, 0.945D, 0.945D, 0.945D, 0.127D, 0.127D, 0.945D, 0.945D, 0.945D, 0.256D };
  double[] r1DataY1v = { 0.85D, 0.85D, 0.85D, 0.88D, 0.9D, 0.9D, 0.9D, 6.6D, 6.6D, 0.87D, 0.87D, 0.87D, 3.29D };
  double[] r1DataZ1v = { 1.03D, 0.78D, 0.78D, 0.72D, 0.55D, 0.55D, 0.55D, 0.5D, 0.5D, 0.43D, 0.43D, 0.43D, 0.3D };
  double[] r1DataX2v = { 1.0D, 1.0D, 1.0D, 1.0D, 1.89D, 1.89D, 1.89D, 0.127D, 0.127D, 1.89D, 1.89D, 1.89D, 0.256D };
  double[] r1DataY2v = { 1.73D, 1.74D, 1.74D, 1.77D, 0.9D, 0.9D, 0.9D, 13.204000000000001D, 13.204000000000001D, 0.88D, 0.88D, 0.88D, 6.645D };
  double[] r1DataZ2v = { 1.15D, 0.9D, 0.9D, 0.75D, 1.1D, 1.1D, 1.1D, 0.8D, 0.8D, 0.86D, 0.86D, 0.86D, 0.7D };

[...]

    if (INV == 0) {
      if (AVR == 0) {
        if (iUPS == 1) {
          if (iVoltage >= 200) {
            if (iVA <= 625) {
              OVoltage = (paramArrayOfInt[3] * 1.79D + 3.35D);
            }
            if ((iVA >= 800) && (iVA < 2000)) {
              OVoltage = (paramArrayOfInt[3] * 1.61D);
            }
            if (iVA >= 2000)
              OVoltage = (paramArrayOfInt[3] * 1.625D);
          }
          else
          {
            if (iVA <= 625) {
              OVoltage = (paramArrayOfInt[3] * 0.89D + 6.18D);
            }
            if ((iVA >= 800) && (iVA < 2000)) {
              OVoltage = (paramArrayOfInt[3] * 1.61D / 2.0D);
            }
            if (iVA >= 2000) {
              OVoltage = (paramArrayOfInt[3] * 1.625D / 2.0D);
            }
          }
        }
        [...]
      }
      else if (AVRMode == 1) {
        if ((iUPS == 1) || (iUPS == 13)) {
          if (iVoltage >= 200) {
            if (iVA <= 525) {
              OVoltage = (paramArrayOfInt[3] * 2.07D);
            }
            if (iVA == 625) {
              OVoltage = (paramArrayOfInt[3] * 2.07D + 5.0D);
            }
            if ((iVA >= 800) && (iVA < 2000)) {
              OVoltage = (paramArrayOfInt[3] * 1.87D);
            }
            if (iVA >= 2000)
              OVoltage = (paramArrayOfInt[3] * 1.87D);
          }
          else
          {
            if (iVA <= 625) {
              OVoltage = (paramArrayOfInt[3] * 2.158D / 2.0D);
            }
            if ((iVA >= 800) && (iVA < 2000)) {
              OVoltage = (paramArrayOfInt[3] * 1.842D / 2.0D);
            }
            if (iVA >= 2000) {
              OVoltage = (paramArrayOfInt[3] * 1.875D / 2.0D);
            }
          }
        }
        [...]
      }
      else
      {
        if ((iUPS == 1) || (iUPS == 13)) {
          if (iVoltage >= 200) {
            if (iVA <= 625) {
              OVoltage = (paramArrayOfInt[3] * 1.571D);
            }
            if ((iVA >= 800) && (iVA < 2000)) {
              OVoltage = (paramArrayOfInt[3] * 1.37D);
            }
            if (iVA >= 2000)
              OVoltage = (paramArrayOfInt[3] * 1.4D);
          }
          else
          {
            if (iVA <= 625) {
              OVoltage = (paramArrayOfInt[3] * 1.635D / 2.0D);
            }
            if ((iVA >= 800) && (iVA < 2000)) {
              OVoltage = (paramArrayOfInt[3] * 1.392D / 2.0D);
            }
            if (iVA >= 2000) {
              OVoltage = (paramArrayOfInt[3] * 1.392D / 2.0D);
            }
          }
        }
        [...]
      }
    }
    else
    {
      if ((iUPS == 1) && 
        (T != 0))
      {
        int i;
        if (iVoltage < 200) {
          i = paramArrayOfInt[12] / 16 - 1;
          if ((i <= 12) && 
            (i >= 0)) {
            rDataX = r1DataX1v[i];
            rDataY = r1DataY1v[i];
            rDataZ = r1DataZ1v[i];
          }
        }
        else
        {
          i = paramArrayOfInt[12] / 16 - 1;
          if ((i <= 12) && 
            (i >= 0)) {
            rDataX = r1DataX2v[i];
            rDataY = r1DataY2v[i];
            rDataZ = r1DataZ2v[(paramArrayOfInt[12] / 16)];
          }
        }

        BoostData = (1 + AVR * 20 / 135);
        if (iVA > 625) {
          sqrtest = (paramArrayOfInt[1] * rDataX * (paramArrayOfInt[1] * rDataX) * (T - paramArrayOfInt[3]) / T);
          if (sqrtest > 0.0D)
            OVoltage = (Math.sqrt(sqrtest) * rDataY * BoostData - paramArrayOfInt[0] * rDataZ * BoostData);
        }
        else
        {
          sqrtest = ((paramArrayOfInt[1] * rDataX - paramArrayOfInt[0] * rDataZ) * (paramArrayOfInt[1] * rDataX - paramArrayOfInt[0] * rDataZ) * (T - paramArrayOfInt[3]) / T);
          if (sqrtest > 0.0D) {
            OVoltage = (Math.sqrt(sqrtest) * rDataY);
          }
        }
        ErrorVal = 10;
        if (iVA < 200) {
          ErrorVal = 5;
        }
        if (OVoltage > iVoltage + ErrorVal) {
          OVoltage = (iVoltage + ErrorVal);
        }
        if ((OVoltage < iVoltage - ErrorVal) && (OVoltage > 50.0D)) {
          OVoltage = (iVoltage - ErrorVal);
        }
      }
    }
```

Here the original java file for reference.

aquette, 2015-12-29:
@ikus060 : bonjour Patrick.
I'd like to make a branch implementing the above.
However, for the input voltage, do you know where the *iVoltage* data comes from, in the Java implementation?
Also note that I've no such units, so I'd rely on your for the testing...

aquette, 2015-12-29:
@ikus060 : I've created a branch (https://github.com/networkupstools/nut/tree/PowerCom-KIN) which includes the tentative fix for both input and output voltages. As mentioned in the commit message and code comments, this may still need some further adjustments (esp. for output).
Thanks for testing and reporting back.

ikus060, 2015-12-29:
Hello,

I'm running debian wheezy. I'm having alot of trouble to get nut compile
from github source code. I can't even get 'autogen.sh' to run properly. Do
you have any recommendation to get the driver compile alone ?

ikus060, 2015-12-29:
Forget it, I manage to get it compile. I was missing "libtool" package.

ikus060, 2015-12-29:
I have two different revision of KIN2200 that behave differently. See the result:

## Before
```
$ upsc myups$ upsc myups
battery.charge: 0.0
device.mfr: PowerCom
device.model: KIN2200APRM
device.serial: Unknown
device.type: ups
driver.name: powercom
driver.parameter.manufacturer: PowerCom
driver.parameter.modelname: KIN2200APRM
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyS0
driver.parameter.type: KIN
driver.version: 2.6.4
driver.version.internal: 0.13
input.frequency: 80.12
input.voltage: 100.8
input.voltage.nominal: 120
output.frequency: 80.12
output.voltage: 100.8
ups.load: 15.2
ups.mfr: PowerCom
ups.model: KIN2200APRM
ups.model.type: KIN
ups.serial: Unknown
ups.status: OL
```

```
$ upsc myups2
battery.charge: 100.0
device.mfr: PowerCom
device.model: KIN2200APRM
device.serial: Unknown
device.type: ups
driver.name: powercom
driver.parameter.manufacturer: PowerCom
driver.parameter.modelname: KIN2200APRM
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyS0
driver.parameter.type: KIN
driver.version: 2.6.4
driver.version.internal: 0.13
input.frequency: 60.09
input.voltage: 248.6
input.voltage.nominal: 120
output.frequency: 60.09
output.voltage: 250.2
ups.load: 1.2
ups.mfr: PowerCom
ups.model: KIN2200APRM
ups.model.type: KIN
ups.serial: Unknown
ups.status: OL

```

## After
```
$ upsc myups
battery.charge: 100.0
device.mfr: PowerCom
device.model: KIN2200APRM
device.serial: Unknown
device.type: ups
driver.name: powercom
driver.parameter.manufacturer: PowerCom
driver.parameter.modelname: KIN2200APRM
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyS0
driver.parameter.synchronous: no
driver.parameter.type: KIN
driver.version: 2.7.3.1
driver.version.internal: 0.17
input.frequency: 60.00
input.voltage: 124.0
input.voltage.nominal: 120
output.frequency: 60.00
output.voltage: 124.0
ups.load: 13.0
ups.mfr: PowerCom
ups.model: KIN2200APRM
ups.model.type: IMP
ups.serial: Unknown
ups.status: OL
```

```
$ upsc myups2
battery.charge: 213.0
device.mfr: PowerCom
device.model: KIN2200APRM
device.serial: Unknown
device.type: ups
driver.name: powercom
driver.parameter.manufacturer: PowerCom
driver.parameter.modelname: KIN2200APRM
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyS0
driver.parameter.synchronous: no
driver.parameter.type: KIN
driver.version: 2.7.3.1
driver.version.internal: 0.17
input.frequency: 80.00
input.voltage: 306.0
input.voltage.nominal: 120
output.frequency: 81.00
output.voltage: 310.0
ups.load: 1.0
ups.mfr: PowerCom
ups.model: KIN2200APRM
ups.model.type: IMP
ups.serial: Unknown
ups.status: OL
```
## Debug output
""myups""
```
  12.310999	Num of bytes requested for reading from UPS: 16
  12.456115	Num of bytes received from UPS: 16
Raw data from UPS:
 0 0x0d ( )
 1 0x63 (c)
 2 0x3e (>)
 3 0x3e (>)
 4 0x3c (<)
 5 0xff (�)
 6 0x3c (<)
 7 0x00 ( )
 8 0x00 ( )
 9 0x00 ( )
10 0x08 ( )
11 0x4b (K)
12 0xb3 (�)
13 0x00 ( )
14 0x00 ( )
15 0x4b (K)
  12.456146	input.frequency   (raw data): [raw: 60]
  12.456153	input.frequency: 60.00
  12.456156	output.frequency   (raw data): [raw: 60]
  12.456160	output.frequency: 60.00
  12.456162	ups.load  (raw data): [raw: 13]
  12.456167	ups.load: 13.0
  12.456169	battery.charge (raw data): [raw: 99]
  12.456175	send_to_all: SETINFO battery.charge "99.0"
  12.456184	battery.charge: 99.0
  12.456189	input.voltage (raw data): [raw: 62]
  12.456194	input.voltage: 124.0
  12.456198	output.voltage (raw data): [raw: 62]
  12.456204	output.voltage: 124.0
  12.456210	STATUS: OL
```

""myusp2""
```
  18.311693	Num of bytes requested for reading from UPS: 16
  18.456963	Num of bytes received from UPS: 16
Raw data from UPS:
 0 0x01 ( )
 1 0xd5 (�)
 2 0x9b (�)
 3 0x9b (�)
 4 0x50 (P)
 5 0x21 (!)
 6 0x51 (Q)
 7 0x00 ( )
 8 0x00 ( )
 9 0x00 ( )
10 0x00 ( )
11 0x4b (K)
12 0xb3 (�)
13 0x00 ( )
14 0x00 ( )
15 0x00 ( )
  18.456997	input.frequency   (raw data): [raw: 80]
  18.457005	input.frequency: 60.09
  18.457009	output.frequency   (raw data): [raw: 81]
  18.457017	send_to_all: SETINFO output.frequency "59.35"
  18.457029	output.frequency: 59.35
  18.457036	ups.load  (raw data): [raw: 1]
  18.457043	ups.load: 1.2
  18.457047	battery.charge (raw data): [raw: 213]
  18.457054	battery.charge: 100.0
  18.457059	input.voltage (raw data): [raw: 155]
  18.457068	send_to_all: SETINFO input.voltage "125.9"
  18.457077	input.voltage: 125.9
  18.457082	output.voltage (raw data): [raw: 155]
  18.457090	send_to_all: SETINFO output.voltage "144.1"
  18.457099	output.voltage: 144.1
  18.457107	STATUS: OL
```

ikus060, 2015-12-29:
I was checking further into this issue. I didn't have the same output result for "myups2".
```
$ upsc myups
battery.charge: 100.0
device.mfr: PowerCom
device.model: KIN2200APRM
device.serial: Unknown
device.type: ups
driver.name: powercom
driver.parameter.manufacturer: PowerCom
driver.parameter.modelname: KIN2200APRM
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/ttyS0
driver.parameter.synchronous: no
driver.parameter.type: KIN
driver.version: 2.7.3.1
driver.version.internal: 0.17
input.frequency: 59.35
input.voltage: 124.3
input.voltage.nominal: 120
output.frequency: 60.09
output.voltage: 142.4
ups.load: 1.2
ups.mfr: PowerCom
ups.model: KIN2200APRM
ups.model.type: KIN
ups.serial: Unknown
ups.status: OL
```

```
 114.396365	Num of bytes requested for reading from UPS: 16
 114.541155	Num of bytes received from UPS: 16
Raw data from UPS:
 0 0x01 ( )
 1 0xd5 (�)
 2 0x98 (�)
 3 0x98 (�)
 4 0x51 (Q)
 5 0x21 (!)
 6 0x50 (P)
 7 0x00 ( )
 8 0x00 ( )
 9 0x00 ( )
10 0x00 ( )
11 0x4b (K)
12 0xb3 (�)
13 0x00 ( )
14 0x00 ( )
15 0x00 ( )
 114.541187	input.frequency   (raw data): [raw: 81]
 114.541197	send_to_all: SETINFO input.frequency "59.35"
 114.541208	input.frequency: 59.35
 114.541216	output.frequency   (raw data): [raw: 80]
 114.541224	output.frequency: 60.09
 114.541228	ups.load  (raw data): [raw: 1]
 114.541235	ups.load: 1.2
 114.541240	battery.charge (raw data): [raw: 213]
 114.541247	battery.charge: 100.0
 114.541253	input.voltage (raw data): [raw: 152]
 114.541259	send_to_all: SETINFO input.voltage "123.5"
 114.541265	input.voltage: 123.5
 114.541270	output.voltage (raw data): [raw: 152]
 114.541278	send_to_all: SETINFO output.voltage "141.5"
 114.541287	output.voltage: 141.5
 114.541295	STATUS: OL
```

Everything seams fine. At least it's better. The output voltage seams a bit high. I guess it's related to the load being very low...

aquette, 2015-12-30:
hi @ikus060 , thanks for your testing and feedback.
If possible, could you please counter check the output voltage with the Java Powercom SW?
That's probably the best way to ensure that the NUT code is now aligned and suitable.
Thanks and cheers, Arno

aquette, 2016-03-01:
hi @ikus060 , any news on this ticket?
I'd like to get it fixed, and have 2.7.4 released...

ikus060, 2016-03-01:
Hello @aquette 

I cannot run the Java version. It doesn't work with recent version of Java. Downgrading it is not an option. I've monitor it for some time and the data seams ok. It's way better then is was.

Consider this ticker closed. If I see any issue, I will re-open the ticket or create a new one. Thanks alot for your support.

aquette, 2016-03-01:
hi @ikus060 , thanks for the feedback, and "bonjour des cousins de France" ;)

aquette, 2016-03-01:
reopening to close through the branch merge...

closed 2016-03-02 by aquette