networkupstools/nut #190: upsd: NSS SSL only working in debug mode

clepple, 2015-03-04:
## Summary

Client SSL connections do not complete when connecting to upsd running as a daemon when linked to NSS. Running upsd in the foreground (`upsd -D`) works fine.

Originally identified on Ubuntu 14.04, which compiles NUT 2.7.1 against NSS (libnss3):

```
$ dpkg-query -W  nut-server libnss3 libnspr4
libnspr4:i386    2:4.10.7-0ubuntu0.14.04.1
libnss3:i386    2:3.17.4-0ubuntu0.14.04.1
nut-server    2.7.1-1ubuntu1
```

Verified with NUT 2.7.2.6+ (git master @ 197eafa4) on Linux Mint (also based on Ubuntu 14.04).

## References

mailing list thread: http://news.gmane.org/find-root.php?message_id=CAOXaFyxHtnp0X%2b%3dmrSHNzWcGv3q4TLufE3YjDjEPm1HqO0Q%5fjw%40mail.gmail.com

and http://news.gmane.org/find-root.php?message_id=CAOXaFyxfK32q4QZ9U%2bR%5foWqWJ8MJeWuAeDYe96jLnF4n%2bqvF4A%40mail.gmail.com

aquette, 2015-03-04:
@balooloo : you're expertize would be greatly appreciated here ;)

clepple, 2015-04-11:
Fixed in 2e3fb645.

closed 2015-04-11 by clepple