networkupstools/nut #193: make distcheck-light - nutclient_authenticate.3

nickma82, 2015-03-14:
I've got a problem with "make distcheck-light"
> ...
> make[4]: *** No rule to make target `nutclient_authenticate.3', needed by `distdir'. Stop.
> make[4]: Leaving directory /tmp/nut/docs/man'
> make[3]: *** [distdir] Error 1
> make[3]: Leaving directory /tmp/nut/docs'
> make[2]: *** [distdir] Error 1
> make[2]: Leaving directory /tmp/nut'
> make[1]: *** [dist] Error 2
> make[1]: Leaving directory /tmp/nut'
> make: *** [distcheck-light] Error 2

Reproduction: 
* git clone https://github.com/networkupstools/nut.git
* cd nut; ./autogen.sh
* ./configure (with that output: http://pastebin.com/tb3tfZN2 - expires 14.Apr.2015)
* make
* make distcheck-light (with that output: http://pastebin.com/uv5XHNfD - expires 14.Apr.2015)

Best, Nick

clepple, 2015-03-14:
To build the man pages, you need asciidoc, a2x, and a host of docbook-related dependencies:

```
checking for asciidoc... no
checking for a2x... no
checking for dblatex... no
checking for xsltproc... no
checking for xmllint... /usr/bin/xmllint
checking for xmllint version...  found
checking for source-highlight... no
checking if asciidoc version can build manpages (minimum required 8.6.3)... no
checking if a2x version can build manpages (minimum required 8.6.3)... no
checking if xsltproc is present (mandatory for man page regeneration)... no
checking if xmllint is present (mandatory for man page regeneration)... yes
checking if source-highlight is present (preferable for documentation generation)... no
checking whether to build and install documentation... no
```

If you are building on a system that has an older version of NUT in package form, there is probably a way to list (or automatically install) the dependencies needed to build that package. For instance, on Debian/Ubuntu you should be able to run `apt-get build-dep nut` to install all of the documentation-related dependencies, as well as the -dev packages needed for the other features.

nickma82, 2015-03-14:
Hi @clepple,
thanks for the quick response!

I'm new to nut and don't have any version installed yet. After installing the preconditions on Ubuntu ```apt-get build-dep nut``` totally did the job. 
For information, it installed the following packages:
```asciidoc cdbs dblatex debhelper dh-apparmor dh-systemd dh-translations freeipmi-common intltool libfontconfig1-dev libfreeipmi-dev libfreeipmi12 libfreetype6-dev libgd-dev ibipmimonitoring-dev libipmimonitoring5 libjbig-dev libjpeg-dev libjpeg-turbo8-dev libjpeg8-dev liblzma-dev libneon27-gnutls-dev libnspr4-dev libnss3-dev libpng12-dev libtiff5-dev libtiffxx5 libxpm-dev po-debconf python-scour xsltproc```

and after ```./configure``` and ```make clean;make;make distcheck-light``` it compiled without errors.

Is there any hint about preconditions in the [developer-guide](http://www.networkupstools.org/docs/developer-guide.chunked/ar01s03.html) I missed?
Thanks

clepple, 2015-03-14:
Not the Asciidoc-related dependencies, no. Usually, we field questions about earlier build problems, like not having autoconf, automake and libtool, so that is covered here:

http://www.networkupstools.org/download.html#_development_tree

I guess I also wasn't expecting non-maintainers to run `make distcheck*` targets-- usually if someone is building on their own, they would just do a regular 'make', and refer to documentation from older packages or the NUT website.

nickma82, 2015-03-14:
Well, I did had the "earlier build" dependencies already and followed the [developer-guide](http://www.networkupstools.org/docs/developer-guide.chunked/index.html)->[repository etiquette](http://www.networkupstools.org/docs/developer-guide.chunked/ar01s03.html#_repository_etiquette_and_quality_assurance) concerning "Before pushing your commits upstream" as I'm considering that. :) 

Best, Nick

closed 2015-03-14 by nickma82

clepple, 2015-03-14:
> Well, I did had the "earlier build" dependencies already and followed the developer-guide->repository etiquette concerning "Before pushing your commits upstream" as I'm considering that. :)

from "repository etiquette": we should probably expand on this: "...but will not work unless you have all the optional libraries and features installed"

We're trying not to maintain that list of dependencies in multiple places, especially across multiple *nix distributions. The documentation should probably say to look at the list at the end of the output of ./configure