networkupstools/nut #197: upsc on ePDU show ups.status only (snmp)

yclauzel, 2015-03-31:
@aquette 
Tried on a 2 eaton ePDUs G2 and G3.
Simple conf :
[epdu-name]
  driver = snmp-ups
  port = ip address

"upsc epdu-name" return "ups.status: WAIT"

ePDUs' web interface are working well.

aquette, 2015-03-31:
This bug cause the ePDU support to not be working at all, with ePDU that have many outlets.

I've identified the issue to be linked to the nonblocking I/O and O_NODELAY.
The issue being exactly that the driver sends too many data while upsd doesn't consume these.
The code is isolated to:
https://github.com/networkupstools/nut/blob/master/drivers/dstate.c#L240
up to L.256

Worth creating a "nonblocking" driver flag (for ups.conf), both global and per device

aquette, 2015-04-01:
Seems good to me for merging and fixing this issue.
It would however need more testing, to ensure non-regression and that performance are still satisfactory.
@clepple , @zykh ... feedback and comments appreciated...

zykh, 2015-04-02:
At a quick glance it seems good.

Just one thing...
Is 'synchronous' meant to be a flag (looking at the code, it seems so, but it's not completely clear reading the man -or I just need to sleep- as it says both that it's flag and that the value defaults to 'no')? If so:
- then the per-driver thing (https://github.com/networkupstools/nut/blob/9b9133fca4d2d41ba86b0be26b4bd838f0db4156/drivers/main.c#L367:L371) should be before flags bailout (https://github.com/networkupstools/nut/blob/9b9133fca4d2d41ba86b0be26b4bd838f0db4156/drivers/main.c#L342:L351);
- how are you supposed to override the global value (if enabled) on a driver entry (to disable it)?

Plus the comment on the setinfo() (https://github.com/networkupstools/nut/blob/9b9133fca4d2d41ba86b0be26b4bd838f0db4156/drivers/main.c#L677) doesn't sound right.

closed 2015-04-08 by aquette