networkupstools/nut #198: Windows port can't use port upper 9

Nefelim4ag, 2015-04-02:
Good time of day, on my setup i've 36 bugged ups, he won't work throught usd.
I've tried setup 6x Com cards on linux, and get some troubles =\
i've setup it on windows and it working not bad, but in code of NUT, hardcoded to support only first 9 com port, but i have COM{1..36} =[

Nefelim4ag, 2015-04-02:
I think what problem in
drivers/serial.c:
127: int ser_open_nf(const char *port)

{
	int	fd;

	fd = open(port, O_RDWR | O_NOCTTY | O_EXCL | O_NONBLOCK);

Function open, can't open ports upper COM9, but i can't find where open defined or mappend for win =\

aquette, 2015-04-02:
here is the logic followed:
https://github.com/networkupstools/nut/blob/WindowsPort-v2.6.5-6/drivers/serial.c#L140
https://github.com/networkupstools/nut/blob/WindowsPort-v2.6.5-6/drivers/serial.h#L99
https://github.com/networkupstools/nut/blob/WindowsPort-v2.6.5-6/common/wincompat.c#L661

however, I'm still failing to see where this limitation to COM9 max could come from...

@Nefelim4ag could you please send in some logs and traces? that may help to clarify things and catch the issue.

A quick search led me to this:
http://stackoverflow.com/questions/11775185/open-a-com-port-in-c-with-number-higher-that-9

Could you please try with eg "\\.\COM10" instead of "COM10" and report back?

clepple, 2015-04-02:
Due to the escaping in the NUT configuration parser, I think you will need to use `"\\\\.\\COM10"`

Nefelim4ag, 2015-04-02:
Guys, you are genious!

aquette, 2015-04-07:
Reopening, so that we track some documentation and examples improvements to deal with such Windows specific points.

@Nefelim4ag : would you be kind enough to support us through this process?
I would first need the confirmation of the exact syntax you've used, as per @clepple comment above.
i.e. *\.\COM10* or *\\\\.\\COM10*

aquette, 2015-12-29:
Closing again the ticket, since it's now addressed

closed 2015-12-29 by aquette