networkupstools/nut #200: nut-server doen't start if devices are disconnected

yclauzel, 2015-04-07:
If some devices are disconnected, the nut-server service will not start as nut-driver will end with an issue ans is required in /lib/systemd/system/nut-server.service
This leads to : 
 - We don't know what happened (just nut-driver didn't start)
 - We can't monitor any devices (even ones which are connected)

aquette, 2015-04-08:
Letter case issue, preventing from auto-closing the bug!
Actually closing manually...

rdoursenaud, 2015-05-04:
This change apparently broke my (Arch Linux) install.
Since nut-driver.service does not have an install section, is not required and has StopWhenUnneeded=yes it bails out leaving no access to the device.

aquette, 2015-05-05:
@rdoursenaud : thx for your report.
As per http://www.freedesktop.org/software/systemd/man/systemd.unit.html, could you test the following and feedback:
Wants=nut-driver.service

Since *Wants* is weaker, failing to start one or more driver should result in having upsd (nut-server.service) still starting ( @yclauzel : pinging for Debian test thx)

rdoursenaud, 2015-05-05:
Just tested, works a treat!

yclauzel, 2015-05-05:
Tested on debian-jessie. Works well.

aquette, 2015-05-07:
Thanks @rdoursenaud and @yclauzel.
I'm committing the fix and will be closing back again the issue.

closed 2015-05-07 by aquette