networkupstools/nut #203: [HCL] Powercool 350VA to 1600VA USB supported by nutdrv_atcl_usb

arthurdeco, 2015-04-13:
Hi Guys,

Been banging my head against this issue for many days now & I think I've got a handle on it.
Tried every driver in the apt- repository list, failed every one even with lots of variations.
lsusb verbose gave some clues - suggesting atcl (but blazer didn't work, close but no cigar).
More searching suggested driver "nutdrv_atcl_usb" but it's not in the ap package
So a steep learning curve on git / build-essential / and a dozen other undocumented bits got me a compiled nutdrv_atcl_usb sort of...
Anyway - /lib/nut/nutdrv_atcl_usb -a powercool -u root -DDD did produce info.
Just like in the windows USB debugger I looked at, there's only 2 messages of worth:
AC=OK and AC=FAILED and those are listed in the output from the driver:
Network UPS Tools - 'ATCL FOR UPS' USB driver 1.1 (2.7.2-signed-224-g2e3fb64)
Warning: This is an experimental driver.
  me features may not function correctly.
?
   0.000000     debug level is '3'
   0.000808     Searching for USB device...
   0.361940     Checking USB device [1d6b:0001] (006/001)
   0.426153     Checking USB device [1d6b:0001] (005/001)
   0.490164     Checking USB device [0001:0000] (004/002)
   0.503841     USB device [0001:0000] opened
   1.439730     read: (8 bytes) => 03 00 00 00 00 00 00 00
   1.439838     reply[0] = 0x03 -> OL
   1.440146     dstate_init: sock /var/state/ups/nutdrv_atcl_usb-powercool open on fd 5
   2.487781     read: (8 bytes) => 03 00 00 00 00 00 00 00
   2.487888     reply[0] = 0x03 -> OL
   3.519781     read: (8 bytes) => 03 00 00 00 00 00 00 00
   3.519885     reply[0] = 0x03 -> OL
   5.447784     read: (8 bytes) => 03 00 00 00 00 00 00 00
   5.447888     reply[0] = 0x03 -> OL
   7.447748     read: (8 bytes) => 03 00 00 00 00 00 00 00
   7.447803     reply[0] = 0x03 -> OL
   9.447723     read: (8 bytes) => 03 00 00 00 00 00 00 00
   9.447785     reply[0] = 0x03 -> OL
  11.447809     status interrupt read: error submitting URB: No such device
  11.447878     Communications with UPS lost: Query to UPS failed
  13.713950     Checking USB device [1d6b:0001] (006/001)
  13.778280     Checking USB device [1d6b:0001] (005/001)
  13.842253     Checking USB device [0001:0000] (004/003)
  13.853796     Can't set USB configuration: could not set config 1: Device or resource busy
  13.858421     USB device [0001:0000] opened
  14.362799     read: (8 bytes) => 01 00 00 00 00 00 00 00
  14.362855     Communications with UPS re-established
  14.362874     reply[0] = 0x01 -> OB
  15.458808     read: (8 bytes) => 01 00 00 00 00 00 00 00
  15.458866     reply[0] = 0x01 -> OB
  17.458745     read: (8 bytes) => 01 00 00 00 00 00 00 00
  17.458804     reply[0] = 0x01 -> OB
  19.458781     read: (8 bytes) => 01 00 00 00 00 00 00 00
  19.458841     reply[0] = 0x01 -> OB
  21.458805     read: (8 bytes) => 03 00 00 00 00 00 00 00
  21.458862     reply[0] = 0x03 -> OL
  23.458808     read: (8 bytes) => 03 00 00 00 00 00 00 00
  23.458872     reply[0] = 0x03 -> OL
  25.466809     read: (8 bytes) => 03 00 00 00 00 00 00 00

0x03 = "AC=OK"
0x01 = "AC=Failed"
You can see that the mains failed at 11 seconds & recovered again at 21 secs.
Is this helpful do you think, or should I abandon it & use a windows machine to monitor it?
Thanks

clepple, 2015-04-14:
> Been banging my head against this issue for many days now & I think I've got a handle on it.
> Tried every driver in the apt- repository list, failed every one even with lots of variations.
> lsusb [...]
Debian or Ubuntu? If the latter, there is a PPA that I plan to update every now and then: https://launchpad.net/~clepple/+archive/ubuntu/nut

For Debian, you might want to request a backport of NUT.
> So a steep learning curve on git / build-essential / and a dozen other undocumented bits got me a compiled nutdrv_atcl_usb sort of...
> 
For future reference, we have the archives for the email lists linked from the main NUT website - compiling debs has come up a few times.
> You can see that the mains failed at 11 seconds & recovered again at 21 secs.
> Is this helpful do you think, or should I abandon it & use a windows machine to monitor it?
> 
Have you had a chance to test the shutdown function? I realized that it is not listed in the instcmds, but if you move the computer to another outlet, then kill the existing driver, and run 'nutdrv_atcl_usb -a powercool -k', it should kill power for 30 seconds.

Either way, we can add your model to the HCL/DDL, but if shutdown doesn't work, that would be useful to note.

arthurdeco, 2015-04-16:
Hi clepple,

thanks hugely for replying, I was feeling really lost back there!
It's ubuntu trusty, so I will give your PPA a go I think...after removing all of my rubbish workings (just like me doing diy in the house I always leave stuff lying around...).

I tried the -k option & it didn't shutdown the ups.
Any other info you need from me?

Thanks again - I'll have a go at th PPA at the weekend & let you know...
all the best
art

clepple, 2015-04-18:
@arthurdeco I forgot to mention that the UPS has to be on battery for "-k" to work. (This is a potential problem if the power comes back after NUT starts to shut down.)

Check the bottom of http://www.networkupstools.org/stable-hcl.html for what we would like for the HCL. For your UPS, I think it will mostly be the `upsc` output.

closed 2017-05-22 by clepple