networkupstools/nut #208: Another systemd behavior regression (WRT sysV) with drivers startup

aquette, 2015-05-20:
following #200, @yclauzel has found another minor bug in the NUT driver startup flow, that is a behavior regression considering the previous sysV init one:
When 1 driver, at least, fails to start, upsdrvctl currently returns EXIT_FAILURE.
This results in systemd considering the whole nut-driver unit to have failed, and in turn nut-server unit is not started.
Previously, driver(s) were started in best effort mode, and then upsd.

Solution:
- Implement a verification in upsdrvctl: if at least 1 driver has succeeded, exit code will be 2. If all driver(s) have failed, return as currently with EXIT_FAILURE (code 1). Success code (0) is unchanged
- document these modifications in upsdrvctl manual page 
- Add the following field to the nut-driver.service file, in the [Service] section
SuccessExitStatus=2

I'll be pushing these proposed changes in a separate branch, while waiting for feedback.
This should not lead to regression in users integration, since most (if not all) people should be testing for a non-zero return code to check for errors. However, this will solve another systemd integration point.

@clepple , @zykh , @balooloo , @bigon : calling for review and approval. thx!

aquette, 2015-05-20:
branch available for review: https://github.com/networkupstools/nut/tree/upsdrvctl_partial_error

yclauzel, 2015-05-21:
I'm really interested into this one.
Things are working fine in a good environement but this fix should improve NUT robustness.

bigon, 2015-05-21:
One of the option could be to add some systemd specific functionality using sd_notify() http://www.freedesktop.org/software/systemd/man/sd_notify.html ??

clepple, 2015-05-22:
I have no opinions on the implementation yet, but we really need a section in the manual for systemd - especially since there are going to be several variations of the service files as installed by the various distributions.

jimklimov, 2022-08-19:
Current release 2.8.0 features better systemd integration (instance per driver, not one for all drivers) and ability to start upsd without drivers defined #766

closed 2022-08-19 by jimklimov