networkupstools/nut #209: systemd improvements and better integration

aquette, 2015-05-20:
Following the switch to systemd, a number of behavior regression have been identified and addressed (#200 and #208 particularly).

However, there are more visible improvements that can be addressed too, such as:
- ensure that systemd does well handle the UPS shutdown (upsdrvctl shutdown). This is not that obvious currently and is the main point of UPS protection systems! This probably requires a dedicated service file, with a "WantedBy: poweroff shutdown" or something alike.
@bigon : could you please comment (i.e. have you made some tests on your side?).
- emitting SIGPWR and check sigpwr.target (http://www.freedesktop.org/software/systemd/man/systemd.special.html)
- to be completed...

GreatEmerald, 2016-02-11:
Yes, please do!
As far as I can tell, the current way `upsdrvctl shutdown` is handled is... completely broken for any `usbhid-ups` systems. The `nutshutdown` script is executed right before shutdown - but `nut-driver.service` is stopped way sooner! It stops as soon as there is nothing left that requires it, and the only thing that requires it is `nut-driver.service`, which is turned off as soon as `systemctl poweroff` is called. That means the `usbhid-ups` drivers are turned off, and at shutdown `upsdrvctl` cannot do anything at all, since it no longer has communications to the UPS.

People on the systemd IRC channel suggested that the actually good way to implement this would be using a target (conveniently, there is `sigpwr.target`). `upsmon` should call `systemctl isolate sigpwr.target`; that will stop all services aside from the ones pulled in by said target. Then filesystems should be unmounted, `upsdrvctl shutdown` should be called (no need for any `/etc/killpower` files, the target would only be isolated when the UPS needs to be turned off anyway; in turn we don't need any mounted filesystems at all), and it would perhaps then proceed to `final.target` and `poweroff.target` (but not strictly necessary, as everything is off and unmounted at that point anyway, waiting for the UPS to shut down). That also prevents all power races.
Of course, that all will need quite a bit of testing to make sure it works right in all situations.

clepple, 2016-02-11:
> As far as I can tell, the current way `upsdrvctl shutdown` is handled is... completely broken for any `usbhid-ups` systems.

Are you sure of the failure mode? `upsdrvctl shutdown` **expects** the original driver to be stopped by the time it runs - it re-invokes the driver with `usbhid-ups -k` to skip any of the `upsd` connection code, and the driver simply sends the shutdown command.

Unfortunately, for a USB UPS, that means that `/dev/bus/usb` can't be unmounted at that point (but it shouldn't need to be unmounted, since it is a virtual filesystem).

GreatEmerald, 2016-02-11:
I'm sure that it failed to turn off my UPS, at least. If it does expect the drivers to be turned off and turns them back on, then that's not really documented in the manual (and is quite counter-intuitive).