networkupstools/nut #217: Drivers/ups.conf options

zykh, 2015-07-18:
- Global/main.c options (mostly) blindly accept what you feed them: we don't really check whether they are flags or if required value is present.
- Leftovers global directives are not checked, you can write whatever you want, it won't fail.
- Whether an entry is a flag or a variable is ups.conf(/-x)-decided: it should be instead driver (vartab) dependent.
- We should check that overrides/defaults have values before setting them.

clepple, 2015-07-26:
While I would generally agree that the command-line parsing is not NUT's strong suit, I think the majority of these items could be addressed by simply logging messages as each option is parsed. If a user changes an option, they would be able to see that in their logs (or not, if the option was mistyped). I am concerned that this would involve a big driver overhaul to keep track of which options have been parsed (item 2), and we don't have a driver testing infrastructure yet.

That said, I am open to ideas on this.

For the last item: perhaps I misunderstood, but defaults (and overrides, to some degree) are valid whether or not there is an existing value. In the case of overrides, the full NUT variable tree might not be populated when the driver initializes.

zykh, 2015-07-28:
Sorry if I wasn't clear, I meant "values to be set".
The following ups.conf section:

````
[ups]
        ...
        override.the.vaselines
        override.you.think
        default.you.re = ""
        default.a.man
````

..leads to this upsc output:

````
driver.flag.override.the.vaselines: enabled
driver.flag.override.you.think: enabled
driver.flag.default.a.man: enabled
the.vaselines: (null)
you.think: (null)
you.re: 
a.man: (null)
````

and this command-line:

````
<driver> -a <ups> \
         -x override.the.vaselines \
         -x override.you.think \
         -x default.you.re= \
         -x default.a.man
````

..to this upsc output:

````
the.vaselines: (null)
you.think: (null)
you.re: 
a.man: (null)
````

clepple, 2015-07-28:
ah, I was really confused - the examples help!

I have no strong feelings either way. Agreed that we could do better, but I still think that we could avoid this by adding some more logging to clarify whether the options are being parsed as the user intended.

I suppose this issue could also be fixed in such a way that wouldn't require too many code changes to the drivers, but I admit I haven't thought through all of that.