networkupstools/nut #218: TRIPP LITE not removed from model name on SMART500RT1U

tas50, 2015-07-23:
I have a Tripp Lite SMART500RT1U.  In the upsc output the manufacturer is correctly identified, but TRIPP LITE isn't stripped from the model name correctly.

ups.mfr: Tripp Lite
ups.model: TRIPP LITE SMART500RT1U

There seems to be a function in the driver that attempts to strip this out, but it doesn't appear to work on this model:

	/* trim "TRIPP LITE" from beginning of model */
	model = strdup(hd->Product);
	if(strstr(model, hd->Vendor) == model) {
		model += strlen(hd->Vendor);
	}

lsusb output;

Device: ID 09ae:0001 Tripp Lite
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x09ae Tripp Lite
  idProduct          0x0001
  bcdDevice            0.0a
  iManufacturer           1 Tripp Lite
  iProduct                2 TRIPP LITE SMART500RT1U
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      52
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0001
  Self Powered

clepple, 2015-07-23:
> There seems to be a function in the driver that attempts to strip this out, but it doesn't appear to work on this model:
> 
> /* trim "TRIPP LITE" from beginning of model */
> model = strdup(hd->Product);
> if(strstr(model, hd->Vendor) == model) {
>     model += strlen(hd->Vendor);
> }
> 

Would you mind writing and testing a patch to extend this to include the constant "TRIPP LITE" as well as hd->Product? (We'll leave the latter in case there is an OEM version.)

Thanks,
- Charles Lepple

tas50, 2015-12-30:
@clepple Thanks for the reply.  My programming experience ends at Ruby/Bash so this is a bit beyond me. I know enough to read through and determine where the issue is, but not enough to cleanup the problem.  Sorry.

zykh, 2016-01-31:
Something like this should work: https://github.com/zykh/nut/tree/issue-218
@tas50: can you confirm that it works properly for you?

tas50, 2016-02-03:
@zykh The logic seems sound, but I don't have access to that UPS anymore so I'm unable to test it on the real thing.