networkupstools/nut #219: Lost connection

vporton, 2015-07-23:
Below is a part of a terminal output on Debian Linux "testing" ("stretch") with Linux 4.0.0-2-686-pae.

NUT 2.7.2-4

---

Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 01:59:52 2015):
                                                                               
Communications with UPS advice@localhost lost                                  
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 01:59:57 2015):
                                                                               
Communications with UPS advice@localhost established                           
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:00:12 2015):
                                                                               
Communications with UPS advice@localhost lost                                  
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:00:17 2015):
                                                                               
Communications with UPS advice@localhost established                           
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:01:57 2015):
                                                                               
Communications with UPS advice@localhost lost                                  
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:02:02 2015):
                                                                               
Communications with UPS advice@localhost established                           
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:02:22 2015):
                                                                               
Communications with UPS advice@localhost lost                                  
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:02:27 2015):
                                                                               
Communications with UPS advice@localhost established                           
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:03:52 2015):
                                                                               
Communications with UPS advice@localhost lost                                  
                                                                               
                                                                               
Broadcast message from nut@victor.local (somewhere) (Fri Jul 24 02:03:57 2015):
                                                                               
Communications with UPS advice@localhost established

clepple, 2015-07-23:
> Below is a part of a terminal output on Debian Linux "testing" ("stretch") with Linux 4.0.0-2-686-pae.
> 
> NUT 2.7.2-4
> 
Not unexpected-- many UPSes have connectivity issues. Which driver?

I've been meaning to write up some examples of using upssched to filter short-duration power and communication events, but there are some ideas in the documentation.

vporton, 2015-07-24:
> 
> > Below is a part of a terminal output on Debian Linux "testing"
> ("stretch") with Linux 4.0.0-2-686-pae.
> > 
> > NUT 2.7.2-4
> > 
> Not unexpected-- many UPSes have connectivity issues. Which driver?
/etc/nut/ups.conf:
    driver = blazer_usb
Rebooting the computer helps.

> 
> I've been meaning to write up some examples of using upssched to filter short-duration power and communication events, but there are some ideas in the documentation.