networkupstools/nut #232: nutclient: stringset_to_strarr() and stringvector_to_strarr() only work for one element

clepple, 2015-09-03:
reported by Paul Vermeer: http://article.gmane.org/gmane.comp.monitoring.nut.user/9331

> Both functions stringset_to_strarr and stringvector_to_strarr have a similar bug in play. Although trivial in code, the coded function does not operate as intended.
> The pointer increment is missing (see added line marked yellow below [pstr++;]), which results in the same element being updated over and over again, and then only leaving the last element of the list in the array.

aquette, 2016-03-09:
@balooloo : if you have 5 mn to take a look at that one also, would be great. Please self assign if you can, thx.

aquette, 2016-05-09:
Closed through PR #284

closed 2016-05-09 by aquette