networkupstools/nut #238: [HCL] Eaton E Series DX UPS 1-20 kVA partially supported by blazer_ser

fiery-, 2015-10-05:
Link to product page:
http://www.eaton.com/Eaton/ESeriesUPS/DXUPS/

upsc
```
battery.charge: 100
battery.voltage: 109.44
battery.voltage.high: 104.00
battery.voltage.low: 83.20
battery.voltage.nominal: 96.0
device.type: ups
driver.name: blazer_ser
driver.parameter.pollinterval: 10
driver.parameter.port: /dev/ttyS0
driver.parameter.synchronous: no
driver.version: 2.7.3
driver.version.internal: 1.56
input.current.nominal: 10.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 236.3
input.voltage.fault: 215.4
input.voltage.nominal: 220
output.voltage: 221.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 4
ups.status: OL
ups.temperature: 25.0
ups.type: online
```
upscmd -l
```
Instant commands supported on UPS [xxx]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```
- Current HCL states that Eaton DX is supported by **mge-utalk**, which is not correct (tested with latest mge-utalk driver v2.7.3 compiled from src.rpm for Fedora).
- blazer_ser driver is v2.6.5 on CentOS 6.6 (EPEL repository). 
- We use upssched command to shutdown the machine after 5 min. on battery (schedule script sends *upsmon -c fsd*, which in turn calls *upsdrvctl shutdown* in /etc/init.d/halt script). This works almost fine, except that the UPS doesn't shutdown.
- **UPS doesn't offload at shutdown and the computer should be started manually. A BIG issues for us.** The same problem is reported here: 
https://lists.alioth.debian.org/pipermail/nut-upsuser/2013-April/008382.html

Is there any communication spec. for that UPS, where we can verify that blazer_ser 'SnRm' command (shutdown.return) is really supported?

clepple, 2015-10-06:
tagging @aquette - is there an issue with sub-60-second shutdown delays on this model?

fiery-, 2015-10-06:
As far as I understand, your question is if setting *ups.delay.shutdown* over 1 min. makes any difference?

According to the current *blazer.c* code, **shutdown.return** may send four different 'SnRm'  commands (the current one seems to be 'S.5R0003', which doesn't work). *ondelay* and *offdelay* are static variables and most probably I should recompile with different values. Do you know how exactly to change current delay values without recompiling?

[blazer.c](https://github.com/networkupstools/nut/blob/master/drivers/blazer.c)
```
static int	ondelay = 3;	/* minutes */
static int	offdelay = 30;	/* seconds */
...
if (!strcasecmp(cmdname, "shutdown.return")) {
		/*
		 * Sn: Shutdown after n minutes and then turn on when mains is back
		 * SnRm: Shutdown after n minutes and then turn on after m minutes
		 * Accepted values for n: .2 -> .9 , 01 -> 10
		 * Accepted values for m: 0001 -> 9999
		 * Note: "S01R0001" and "S01R0002" may not work on early (GE)
		 * firmware versions.  The failure mode is that the UPS turns
		 * off and never returns.  The fix is to push the return value
		 * up by 2, i.e. S01R0003, and it will return online properly.
		 * (thus the default of ondelay=3 mins)
		 */

		if (ondelay == 0) {
			if (offdelay < 60) {
				snprintf(buf, sizeof(buf), "S.%d\r", offdelay / 6);
			} else {
				snprintf(buf, sizeof(buf), "S%02d\r", offdelay / 60);
			}
		} else if (offdelay < 60) {
			snprintf(buf, sizeof(buf), "S.%dR%04d\r", offdelay / 6, ondelay);

		} else {
			snprintf(buf, sizeof(buf), "S%02dR%04d\r", offdelay / 60, ondelay);
		}
	}
```

clepple, 2015-10-07:
> As far as I understand, your question is if setting ups.delay.shutdown over 1 min. makes any difference?
> 
@fiery- if you have tested that, it would be interesting to know. I tagged Arno to see if he had any additional information from Eaton.
> Do you know how exactly to change current delay values without recompiling?
> 
Add `-x offdelay=60` to the command line. (From the output of `blazer_ser -h`)

zykh, 2015-10-14:
> - We use upssched command to shutdown the machine after 5 min. on battery (schedule script sends upsmon -c fsd, which in turn calls upsdrvctl shutdown in /etc/init.d/halt script). This works almost fine, except that the UPS doesn't shutdown. This is tested with blazer_ser v2.6.5 and now we wait a power outage to test v2.7.3 at shutdown.
> - UPS doesn't offload at shutdown and the computer should be started manually. A BIG issues for us. The same problem is reported here: https://lists.alioth.debian.org/pipermail/nut-upsuser/2013-April/008382.html

mmh... reading this and the mentioned topic, my best guess is that the shutdown sequence triggered by the `FSD` flag didn't work because the `C` command failed - it should now be fixed (since d079be2e4aab3c1999d99e87d70de5ee24309a31, landed in 2.7.1).
@fiery-, did you have a chance to test the shutdown sequence with current blazer_ser/nutdrv_qx? Do you still have problems?

fiery-, 2015-10-17:
All my previous tests was with v2.6.5, so it's either the 'S.5R0003' command or the problem which you've fixed in v2.7.1.
I recompiled the blaser v2.7.3 driver with hard coded "S01R0003" shutdown command in blazer.c. In the schedule script I put a "upscmd -u <user> -p <password> ups@localhost shutdown.return" command instead of "upsmon -c fsd".  Now, I'm waiting a power outage (it's a production server and there are 19 days since the last outage).

fiery-, 2015-10-25:
I can confirm that Eaton DX shutdown.return works properly with blazer_ser driver v2.7.3 (I modified blazer.c with a hard coded "S01R0003" shutdown command, may be `-x offdelay=60` command line option will do the same).  In the schedule script I put "upsmon -c fsd" as in first test and this time UPS offloaded and then started the attached computer just fine. The change comes either from the fix in v2.7.1 mentioned above or the hard coded "S01R0003" shutdown command as blazer.c comment suggests.

Please, update the HCL that Eaton DX works with blazer_ser driver and if possible put a note to this thread for those users that encounter some problem with the shutdown command.

closed 2015-11-11 by clepple