networkupstools/nut #241: [HCL] <Electrys> <UPS 2500> supported by <blazer_ser>

superlinux, 2015-11-08:
I am here just to report a UPS that works on NUT 2.7 from a company here in Beirut, Lebanon called Electrys. website is :  http://www.electrys.net . UPS Model is : Electrys UPS2500 . It uses the megatech protocol. Please list this UPS in the HCL page. 
here is the proof in this YouTube video that it works with the configuration steps:
https://www.youtube.com/watch?v=cn3DQ0ns9aQ

clepple, 2015-11-10:
Thanks for the information. Does it also work with the newer [nutdrv_qx](http://www.networkupstools.org/docs/man/nutdrv_qx.html) driver? (nutdrv_qx parameters are a little different, but more consistent -- seconds instead of minutes, etc.)

superlinux, 2015-11-10:
I will have to see. I was really lucky that it worked with blazer_ser.

superlinux, 2015-11-10:
OK! I can confirm that nutdrv_qx works. just replace :
driver=blazer_ser
with :
driver=nutdrv_qx
nothing else has to be done.

superlinux, 2015-11-10:
here is the proof it's working. watch this video.
https://www.youtube.com/watch?v=zb0CPkV6Pg8

superlinux, 2015-11-10:
Hi. I just want to confirm that you saw the second video. thanks.

closed 2015-11-11 by clepple

clepple, 2015-11-11:
@superlinux thanks, I just posted the first video to the nut-upsusers email list.

Also, if you want to provide additional information on your UPS, we are assembling a more detailed version of the HCL that includes the upsc output, as well as upscmd and upsrw: http://buildbot.networkupstools.org/~buildbot/cayman/docs/latest/ddl/#basic-syntax

Feel free to either reopen this issue, or file a new one on the DDL repo: https://github.com/networkupstools/nut-ddl (we could probably figure out some of this from the videos, but it would be much easier and less error-prone to cut-and-paste).

Thanks again!

superlinux, 2015-11-11:
Hi again.
There is another UPS from a company named PCE , I hope I can snipe and hunt it down. It uses a USB cable similar to the ones given to inkjet printers.
These ones actually:
http://www.pceups.com/products/ups/line-interactive/m8