networkupstools/nut #242: NUT with IPPON Back Pro and Smart Winner

grumbler66rus, 2015-11-15:
The blazer_USB peridocally looses connection with several IPPON UPS:
- IPPON Back Pro (desktop UPS)
- IPPON Smart Winner (rackmount UPS)

I make workaround: the script called by upsmon daemon (the token "NOTIFYCMD" of the upsmon.conf), this script unload blazer_usb and load it's again.

The script:
#!/bin/bash
# notifycmdhandler - script for NOTIFYCMD in the upsmon.conf
# It can use the environment variables NOTIFYTYPE and UPSNAME to know what has happened and on which UPS.
# Please insert into /etc/sudoers followed line:
# upsmon ALL=(root) NOPASSWD: /bin/service upsdrv restart
#
# NUT driver; vendor ID and product ID of USB device 
NUTDRIVERNAME=blazer_usb
UPSUSBVID=06da
UPSUSBPID=0003
# Programs
GREP=/bin/grep
SUDO="/usr/bin/sudo -S"
LSUSB=/usr/bin/lsusb
SERVICE=/bin/service
LOGGER=/usr/bin/logger
try_restart_blaser_usb(){
# check usb device presented if the blazer_usb driver is used
# if result of grep is zero then device is connected and we may restart of blazer_usb (this is should restore connection the NUT with UPS)
  [[ ${NUTDRIVERNAME} == "blazer_usb" && -n ${UPSUSBVID} && -n ${UPSUSBPID} ]] && ${LSUSB} | ${GREP} -q ${UPSUSBVID}:${UPSUSBPID} 
  if [[ $? -eq 0 ]]
  then
    ${SUDO} ${SERVICE} upsdrv restart 2>&1 | logger -p daemon.info -t notifycmdhander 
    ${LOGGER} -p daemon.info -t notifycmdhander "${SUDO} /bin/bash -c \"set | /bin/grep UID\" (restart service upsdrv) with result: $?"
  else 
    ${LOGGER} -p daemon.info -t notifycmdhander "USB device ${UPSUSBVID}:${UPSUSBPID}" not found (NUTDRIVERNAME=${NUTDRIVERNAME})
   fi
}
${LOGGER} -p daemon.debug -t notifycmdhander run: UPSNAME=${UPSNAME}, NOTIFYTYPE=${NOTIFYTYPE}, cmdline: $*
[[ ${UPSNAME} ]] || { ${LOGGER} -p daemon.error UPSNAME is not defined! ; exit 1 }
[[ ${NOTIFYTYPE} ]] || { ${LOGGER} -p daemon.error NOTIFYTYPE is nod defined! ; exit 1 }
if [[ "${NOTIFYTYPE}" == "NOCOMM" ]]; then
# Fix  event "NOCOMM" (wall message like "UPS IPPON@localhost is unavailable")
  [[ ${NUTDRIVERNAME} == "blazer_usb" ]] && try_restart_blaser_usb
fi

clepple, 2015-11-16:
> The blazer_USB peridocally looses connection with several IPPON UPS:
> 
> IPPON Back Pro (desktop UPS)
> IPPON Smart Winner (rackmount UPS)
A few questions:

 • Which version of NUT are you using?
 • What OS/kernel are you using?

This really seems like something that the driver should be able to handle, since it should get an error code from libusb when the UPS disconnects.

Can you try this UPS with nutdrv_qx? It is the successor to the blazer_* drivers.

grumbler66rus, 2015-11-16:
> • Which version of NUT are you using?
> • What OS/kernel are you using?

NUT  2.6.5
Linux 3.8.13.4-std-def-alt1.M70P.2 x86_64
libusb 1.0.9

> This really seems like something that the driver should be able to 
> handle, since it should get an error code from libusb when the UPS 
> disconnects.
>
> Can you try this UPS with nutdrv_qx? It is the successor to the 
> blazer_* drivers.

I may doing this some time later (it is need to build latest version of 
NUT from sources), and I may make this test  with IPPON Back Pro only.

clepple, 2015-11-18:
> > • Which version of NUT are you using?
> > • What OS/kernel are you using?
> 
> NUT 2.6.5
> Linux 3.8.13.4-std-def-alt1.M70P.2 x86_64
> libusb 1.0.9

Note that NUT has not been tested as extensively with libusb-1.x and libusb-compat (versus libusb-0.1.x). It should work, but since it is not easy to simulate USB errors, the equivalent error-handling code has not been exercised as much.

> I may doing this some time later (it is need to build latest version of
> NUT from sources), and I may make this test with IPPON Back Pro only.

With either nutdrv_qx or blazer_usb, you can kill the driver process, and start it manually (not via upsdrvctl) with extra -D flags to try to catch the error message when the UPS disconnects.

grumbler66rus, 2015-11-22:
> start it manually (not via upsdrvctl) with extra -D flags
> to try to catch the error message when the UPS disconnects.

I runned blazer_usb:
    /lib/nut/blazer_usb -a IPPON -r /var/lib/nut -u upsdrv -DDDD
Result:
...
293767.586757	send: Q1
293767.836896	read: (227.2 227.6 227.6 013 50.1 13.5 25.0 00001001
293769.628938	send: No such device or address
293769.628979	blazer_status: short reply
293769.628985	Communications with UPS lost: status read failed!
293771.598680	Checking device (8087/0024) (001/002)
293771.598834	Failed to open device, skipping. (No such device or address)
293771.613086	Checking device (1D6B/0002) (001/001)
293771.613100	Failed to open device, skipping. (No such device or address)
**293771.613105	Checking device (06DA/0003) (001/011)
293771.613110	Failed to open device, skipping. (No such device or address)**
293771.613113	Checking device (0419/3201) (001/012)
293771.613118	Failed to open device, skipping. (No such device or address)
293771.613122	Checking device (8087/0024) (002/002)
293771.613127	Failed to open device, skipping. (No such device or address)
293771.613131	Checking device (1D6B/0002) (002/001)
293771.613135	Failed to open device, skipping. (No such device or address)
293771.613138	Checking device (0409/0058) (002/003)
293771.613143	Failed to open device, skipping. (No such device or address)
293771.613147	Checking device (03F0/0A01) (002/008)
293771.613151	Failed to open device, skipping. (No such device or address)
293771.613154	No appropriate HID device found
293771.613164	blazer_status: short reply
293771.613168	Communications with UPS lost: status read failed!

At these time:
    Nov 21 21:34:34 sd kernel: [2307548.097872] usb 1-1.1: USB disconnect, device number 11
    Nov 21 21:34:34 sd kernel: [2307548.296645] usb 1-1.1: new low-speed USB device number 13 using ehci-pci
    Nov 21 21:34:34 sd kernel: [2307548.429161] hid-generic 0003:06DA:0003.0007: hiddev0,hidraw0: USB HID v1.11 Device [OMRON USB UPS] on usb-0000:00:1a.0-1.1/input0

Resume: device drops the USB connection and immediately restore it, but blazer_usb can't reconnect.

Also I compule nutdrv_qx from latest sources and try run it:
    # ./nutdrv_qx -a X -r /var/lib/nut -u upsdrv -DDDDD -x subdriver=ippon -x vendorid=06da -x productid=0003 -x bus=1
    Network UPS Tools - Generic Q* USB driver 0.23 (2.7.3-139-g638f1b8)
    USB communication driver 0.33
       0.000000	send_to_all: SETINFO driver.parameter.subdriver "ippon"
       0.000017	send_to_all: SETINFO driver.parameter.bus "1"
       0.000021	debug level is '5'
       0.000110	chrooted into /var/lib/nut
       0.000172	upsdrv_initups...
       0.000213	No appropriate HID device found
       0.000220	No supported devices found. Please check your device availability with 'lsusb'
    and make sure you have an up-to-date version of NUT. If this does not help,
    try running the driver with at least 'subdriver', 'vendorid' and 'productid'
    options specified. Please refer to the man page for details about these options
    (man 8 nutdrv_qx).