networkupstools/nut #244: Reduce logging frequency for upsmon-upsd comms issues

clepple, 2015-11-18:
Slightly different than #243.

From Larry Baker via <http://lists.alioth.debian.org/pipermail/nut-upsdev/2015-October/007104.html>:

...
> However, /var/log/messages is being filled with "upsmon[2391]: Poll UPS [tripplite@localhost] failed - Driver not connected" messages every five seconds.  I cannot seem to find a way to reduce that chatter.

aquette, 2016-03-01:
@clepple : can I assign this one to you?
the 2nd part of the patch looks good, the 1rst (upslog -> upsdebugx) is however subject to discussion

[nut-2.6.5-upsmon.patch.txt](https://github.com/networkupstools/nut/files/153003/nut-2.6.5-upsmon.patch.txt)

clepple, 2016-03-02:
@aquette, I was hoping someone else would pick it up first :-)

what about `upslog(LOG_DEBUG, ...)`?