networkupstools/nut #248: Legrand KEOR MULTIPLUG supported by blazer_usb

sliverc, 2015-12-10:
I have successfully tested the Legrand KEOR MULTIPLUG - nut recognized when the UPS is on battery and also when the battery is low whereas a shutdown signal is sent.

> Dec  7 10:01:30 maro-1 upsmon[1326]: UPS serverups@localhost on battery
Dec  7 10:02:30 maro-1 upsmon[1326]: UPS serverups@localhost battery is low
Dec  7 10:02:30 maro-1 upsmon[1326]: Executing automatic power-fail shutdown
Dec  7 10:02:30 maro-1 upsmon[1326]: Auto logout and shutdown proceeding

Specification:
http://ups.legrand.com/product/keor-multiplug/
http://ups.legrand.com/media/document/UPS_LGR_0083_GB.pdf

upsc:
> battery.charge: 100
battery.voltage: 13.70
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.1
driver.version.internal: 0.10
input.current.nominal: 3.0
input.frequency: 49.6
input.frequency.nominal: 50
input.voltage: 222.4
input.voltage.fault: 222.4
input.voltage.nominal: 230
output.voltage: 224.2
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665

upscmd:
> beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test

The commands are not really working though:
> sudo upscmd -u user -p pw serverups beeper.toogle 1
Unexpected response from upsd: ERR CMD-NOT-SUPPORTED

/lib/nut/blazer_usb -a serverups -DDDD:
> Network UPS Tools - Megatec/Q1 protocol USB driver 0.10 (2.7.1)
   0.000000	debug level is '4'
   0.549366	Checking device (1D6B/0001) (007/001)
   0.612493	- VendorID: 1d6b
   0.612611	- ProductID: 0001
   0.612674	- Manufacturer: unknown
   0.612736	- Product: unknown
   0.612797	- Serial Number: unknown
   0.612935	- Bus: 007
   0.612995	Trying to match device
   0.613070	Device does not match - skipping
   0.613251	Checking device (1D6B/0001) (006/001)
   0.676449	- VendorID: 1d6b
   0.676498	- ProductID: 0001
   0.676513	- Manufacturer: unknown
   0.676527	- Product: unknown
   0.676541	- Serial Number: unknown
   0.676554	- Bus: 006
   0.676567	Trying to match device
   0.676587	Device does not match - skipping
   0.676809	Checking device (0665/5161) (005/002)
   0.686007	- VendorID: 0665
   0.686098	- ProductID: 5161
   0.686164	- Manufacturer: INNO TECH
   0.686225	- Product: USB to Serial
   0.686285	- Serial Number: unknown
   0.686344	- Bus: 005
   0.686407	Trying to match device
   0.686473	Device matches
   0.688076	Trying megatec protocol...
   0.690006	send: Q1
   1.692034	read: error sending control message: Operation not permitted
   1.692170	blazer_status: short reply
   1.692311	Status read 1 failed
   1.694095	send: Q1
   1.828004	read: Q1
   1.828096	blazer_status: short reply
   1.828165	Status read 2 failed
   1.830023	send: Q1
   1.964030	read: Q1
   1.964153	blazer_status: short reply
   1.964259	Status read 3 failed
   1.964324	Trying mustek protocol...
   1.965942	send: QS
   2.283011	read: (217.0 217.0 217.0 --- 49.8 13.7 --.- 00001001
   2.283234	blazer_status: non numerical value [---]
   2.283334	blazer_status: non numerical value [--.-]
   2.283427	Status read in 1 tries
   2.283492	Supported UPS detected with mustek protocol
   2.284967	send: F
   2.493029	read: #230.0 003 12.00 50.0
   2.493232	Ratings read in 1 tries
   2.495005	send: I
   2.620005	read: I
   2.620044	blazer_vendor: short reply
   2.620061	Vendor information read 1 failed
   2.621992	send: I
   2.746926	read: I
   2.747054	blazer_vendor: short reply
   2.747126	Vendor information read 2 failed
   2.748996	send: I
   2.873948	read: I
   2.874061	blazer_vendor: short reply
   2.874130	Vendor information read 3 failed
   2.874193	Vendor information unavailable
   2.874253	No values provided for battery high/low voltages in ups.conf
   2.874367	Using 'guestimation' (low: 10.400000, high: 13.000000)!
   2.874432	Battery runtime will not be calculated (runtimecal not set)
   2.875986	send: QS
   3.192998	read: (217.0 217.0 218.8 --- 49.8 13.7 --.- 00001001
   3.193075	blazer_status: non numerical value [---]
   3.193101	blazer_status: non numerical value [--.-]
   3.193246	dstate_init: sock /var/run/nut/blazer_usb-serverups open on fd 6
   3.194978	send: QS
   3.512031	read: (217.0 217.0 217.0 --- 49.8 13.7 --.- 00001001
   3.512139	blazer_status: non numerical value [---]
   3.512179	blazer_status: non numerical value [--.-]
   5.195979	send: QS
   5.512989	read: (217.0 217.0 217.0 --- 49.8 13.7 --.- 00001001
   5.513153	blazer_status: non numerical value [---]
   5.513252	blazer_status: non numerical value [--.-]
   7.197972	send: QS
   7.514960	read: (217.0 217.0 217.0 --- 49.8 13.7 --.- 00001001
   7.515129	blazer_status: non numerical value [---]
   7.515225	blazer_status: non numerical value [--.-]
   9.199949	send: QS
   9.516958	read: (218.8 217.0 218.8 --- 49.8 13.7 --.- 00001001
   9.517073	blazer_status: non numerical value [---]
   9.517111	blazer_status: non numerical value [--.-]
  11.201936	send: QS
  11.518912	read: (218.8 217.0 217.0 --- 49.8 13.7 --.- 00001001
  11.518992	blazer_status: non numerical value [---]
  11.519014	blazer_status: non numerical value [--.-]

zykh, 2015-12-13:
@sliverc, can you test it with nutdrv_qx (preferably with current git HEAD)?

>The commands are not really working though:
>
>>    sudo upscmd -u user -p pw serverups beeper.toogle 1
>>    Unexpected response from upsd: ERR CMD-NOT-SUPPORTED

Aehm.. 'beeper.**toggle**', not 'beeper.**toogle**'

sliverc, 2015-12-16:
Oops. This explains. OK tested again and the beeper.toggle actually works with blazer_usb and nutdrv_qx.

As it is a production system the UPS is connected I am a bit reluctant to update to git HEAD. I did update to the newest stable release which is available as Debian package (now 2.7.2). Hope this is good enough.

Here are the test details for nutdrv_qx:

upsc:
> Init SSL without certificate database
battery.charge: 100
battery.voltage: 13.20
battery.voltage.high: 13.00
battery.voltage.low: 10.40
battery.voltage.nominal: 12.0
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.2
driver.version.data: Voltronic-QS 0.01
driver.version.internal: 0.06
input.current.nominal: 3.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 227.9
input.voltage.fault: 226.1
input.voltage.nominal: 230
output.voltage: 226.1
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.firmware.aux: PMV
ups.productid: 5161
ups.status: OL
ups.type: offline / line interactive
ups.vendorid: 0665

upscmd:
> beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.quick - Start a quick battery test

> sudo /lib/nut/nutdrv_qx -a serverups -DDD
Network UPS Tools - Generic Q* USB/Serial driver 0.06 (2.7.2)
USB communication driver 0.32
   0.000000	debug level is '3'
   0.011740	upsdrv_initups...
   0.528334	Checking device (1D6B/0001) (007/001)
   0.591465	- VendorID: 1d6b
   0.591579	- ProductID: 0001
   0.591644	- Manufacturer: unknown
   0.591706	- Product: unknown
   0.591766	- Serial Number: unknown
   0.591831	- Bus: 007
   0.591891	Trying to match device
   0.591959	Device does not match - skipping
   0.592198	Checking device (1D6B/0001) (006/001)
   0.655399	- VendorID: 1d6b
   0.655513	- ProductID: 0001
   0.655581	- Manufacturer: unknown
   0.655641	- Product: unknown
   0.655705	- Serial Number: unknown
   0.655765	- Bus: 006
   0.655825	Trying to match device
   0.655891	Device does not match - skipping
   0.656120	Checking device (0665/5161) (005/002)
   0.665903	- VendorID: 0665
   0.665988	- ProductID: 5161
   0.666052	- Manufacturer: INNO TECH
   0.666113	- Product: USB to Serial
   0.666174	- Serial Number: unknown
   0.666237	- Bus: 005
   0.666297	Trying to match device
   0.666365	Device matches
   0.669897	send: QGS
   1.670929	read: error sending control message: Operation not permitted
   1.671119	qx_process_answer: short reply (input.voltage)
   1.672890	send: QGS
   1.815911	read: QGS
   1.816040	qx_process_answer: short reply (input.voltage)
   1.817891	send: QGS
   1.961926	read: QGS
   1.962060	qx_process_answer: short reply (input.voltage)
   1.963899	send: M
   2.088823	read: V
   2.090846	send: QS
   2.407887	read: (222.4 222.4 220.6 --- 50.0 13.4 --.- 00001001
   2.408026	Using protocol: Voltronic-QS 0.01
   2.408068	upsdrv_initinfo...
   2.409894	send: QS
   2.726884	read: (220.6 222.4 222.4 --- 50.0 13.4 --.- 00001001
   2.727815	ups_infoval_set: non numerical value [ups.load: ---]
   2.727908	ups_infoval_set: non numerical value [ups.temperature: --.-]
   2.728857	send: F
   2.936843	read: #230.0 003 12.00 50.0
   2.937147	No values for battery high/low voltages
   2.937193	Using 'guesstimation' (low: 10.400000, high: 13.000000)!
   2.937227	Battery runtime will not be calculated (runtimecal not set)
   2.937272	upsdrv_updateinfo...
   2.937290	Quick update...
   2.938916	send: QS
   3.255896	read: (220.6 220.6 220.6 --- 50.0 13.4 --.- 00001001
   3.256190	dstate_init: sock /var/run/nut/nutdrv_qx-serverups open on fd 6
   3.256223	upsdrv_updateinfo...
   3.256240	Quick update...
   3.257874	send: QS
   3.574856	read: (220.6 220.6 224.2 --- 50.0 13.4 --.- 00001001
   5.258028	upsdrv_updateinfo...
   5.258093	Quick update...
   5.259875	send: QS
   5.576734	read: (220.6 220.6 222.4 --- 50.0 13.4 --.- 00001001
   7.259767	upsdrv_updateinfo...
   7.259837	Quick update...
   7.260812	send: QS
   7.577780	read: (220.6 220.6 220.6 --- 50.0 13.4 --.- 00001001
   9.261295	upsdrv_updateinfo...
   9.261369	Quick update...
   9.262764	send: QS
   9.579738	read: (220.6 220.6 220.6 --- 50.0 13.4 --.- 00001001
  11.263074	upsdrv_updateinfo...
  11.263170	Quick update...
  11.264746	send: QS
  11.581720	read: (220.6 220.6 220.6 --- 50.0 13.4 --.- 00001001
  13.264851	upsdrv_updateinfo...
  13.264926	Quick update...
  13.266610	send: QS
  13.582682	read: (222.4 220.6 220.6 --- 50.0 13.4 --.- 00001001
  15.266163	upsdrv_updateinfo...
  15.266236	Quick update...
  15.267647	send: QS
  15.584617	read: (220.6 220.6 220.6 --- 50.0 13.4 --.- 00001001
  17.267943	upsdrv_updateinfo...
  17.268014	Quick update...
  17.269609	send: QS
  17.586593	read: (220.6 220.6 220.6 --- 50.0 13.4 --.- 00001001
^[[B^[[B^[[B  19.269521	upsdrv_updateinfo...
  19.269598	Quick update...
  19.270589	send: QS
  19.588524	read: (220.6 220.6 220.6 --- 50.0 13.4 --.- 00001001

zykh, 2015-12-17:
Added to both HCL and DDL, thanks!

closed 2015-12-17 by zykh