networkupstools/nut #251: [HCL] <SWEEX> <INTELLIGENT UPS 1500VA P220> supported by <blazer_usb>

crazytiti, 2015-12-26:
SWEEX
INTELLIGENT UPS 1500VA
model P220
http://www.sweex.com/en/notebook-pc-accessoires/ups/PP220/

C:\Program Files (x86)\NUT\bin>upsc ups@127.0.0.1
battery.charge: 12
battery.voltage: 21.40
battery.voltage.high: 26.00
battery.voltage.low: 20.80
battery.voltage.nominal: 24.0
device.type: ups
driver.name: blazer_usb
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: Windows-v2.6.5-5-7-g72f380c
driver.version.internal: 0.10
input.current.nominal: 6.0
input.frequency: 50.0
input.frequency.nominal: 50
input.voltage: 6.4
input.voltage.fault: 6.4
input.voltage.nominal: 220
output.voltage: 216.0
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 3
ups.productid: 5161
ups.status: OB
ups.type: offline / line interactive
ups.vendorid: 0665

shutdown sequence : ups.status send OB when offline.

clepple, 2016-01-04:
Does the UPS send `OB LB` when the battery gets low?

crazytiti, 2016-01-04:
Yes, i just made some tests :
When online : OL
When on battery : OB
When battery goes low (<7%) OB LB
Last status before shutting down : ALARM OL FSD

clepple, 2016-01-06:
Forgot to ask: what do `upsrw` and `upscmd` list?

crazytiti, 2016-01-06:
C:\Program Files (x86)\NUT\bin>upscmd -l ups@127.0.0.1
Instant commands supported on UPS [ups]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test

crazytiti, 2016-01-06:
I have to set up rights to use upsrw, maybe tonight i can do it

crazytiti, 2016-01-06:
it seems upsrw is not supported :
C:\Program Files (x86)\NUT\bin>upsrw ups@127.0.0.1:3493

C:\Program Files (x86)\NUT\bin>

zykh, 2016-01-31:
Added to our DDL, too. Thanks!

closed 2016-01-31 by zykh