networkupstools/nut #252: Segfault on Ubuntu 14.04 with usbhid-ups on APC Backup-UPS 550G

tas50, 2015-12-30:
It's quite possible I'm doing something horribly wrong here, but here goes nothing.  I have Ubuntu 14.04 with the backported 4.2 kernel and the nut PPA to get nut "2.7.2.9~really2.7.3-2".  I have an APC Back-UPS ES 550G with USB.  I've copied in the UDEV script 62-nut-usbups.rules.  I have the following /etc/nut/ups.conf:

[apc]
	driver = usbhid-ups
	port = auto
	desc = APC Back-UPS BE550G

when I try to start nut I get a failure though:

upsdrvctl start apc -h
Network UPS Tools - UPS driver controller 2.7.3
Network UPS Tools - Generic HID driver 0.39 (2.7.3)
USB communication driver 0.32
kill: No such process
Driver exited abnormally

Syslog shows a segfault:
Dec 30 14:35:15 smith-0 kernel: [51973.368161] usbhid-ups[9719]: segfault at 0 ip 00007f936a18cff4 sp 00007ffe6ca287d8 error 4 in libc-2.19.so[7f936a04b000+1bb000] 

Any ideas?  Is there more information I can gather?

tas50, 2015-12-30:
Looks like a simple reboot actually fixed this.

closed 2015-12-30 by tas50

clepple, 2015-12-31:
> Is there more information I can gather?

In case this happens again, can you also check `/var/log/syslog` for any NUT messages containing "pid"? That code should be pretty solid, but I'll admit that I don't typically test with newer kernels than stock.

bandroidx, 2016-01-24:
same problem here i get:

Jan 23 19:06:05 librenms kernel: [   97.828615] usb 2-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
Jan 23 19:06:06 librenms kernel: [   98.831615] usb 2-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
Jan 23 19:06:07 librenms kernel: [   99.834556] usb 2-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
Jan 23 19:06:07 librenms kernel: [   99.834606] usbhid-ups[2600]: segfault at 0 ip 00007f0e49daf48c sp 00007ffffc38a148 error 4 in libc-2.19.so[7f0e49c6e000+1bb000]
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                
also ubuntu 14.04 server 64bit

clepple, 2016-01-24:
@bandroidx could you please open a new issue and provide some more details about the NUT package version and the running kernel?

bandroidx, 2016-01-24:
@clepple sure, just did.

clepple, 2016-01-25:
For future reference, this also seems to be the same point in the code as #258:
```
$ addr2line -f -e /usr/lib/debug/lib/x86_64-linux-gnu/libc-2.19.so $(printf "0x%x" $((0x00007f936a18cff4 - 0x7f936a04b000)))
__strncmp_ssse3
/build/buildd/eglibc-2.19/string/../sysdeps/x86_64/multiarch/../strcmp.S:1233
```

Same issue in an upstream bug: https://bugs.launchpad.net/ubuntu/+source/nut/+bug/1483615

AndreKR, 2017-09-15:
Since a reboot fixed it, it's possible that the udev rules were not reloaded. I just found out that the Ubuntu 16.04 install script does not run `udevadm control --reload` so `/dev/bus/usb/...` doesn't belong to `nut`, even if the UPS is unplugged and plugged. `udevadm control --reload` has to be run (or a reboot be done) to fix it. I'm going to file an issue with the Ubuntu package.

Unfortunately in this case usbhid-ups simply segfaults, I created #477 about adding a better error message.