networkupstools/nut #255: [HCL] Eaton Ellipse PRO 650VA (USB only) supported by usbhid-ups

fr3ddie, 2016-01-08:
Shutdown sequence tested OK on an old Debian Sarge (I suppose the device would work best on newer NUT releases...).

fr3ddie@NAS:~$ upsc -V
Network UPS Tools upsc 2.0.5

upsc dump:

battery.charge: 97
battery.chemistry: PbAc
battery.runtime: 2755
battery.voltage: 2.3
driver.name: hidups
driver.parameter.lowbatt_pct: 30
driver.version: 2.0.5
ups.mfr: EATON
ups.model: Ellipse PRO
ups.serial: G355FXXXXX
ups.status: OL

(I put the XXXXX manually)

clepple, 2016-01-09:
Can you please check with the `newhidups` driver instead? (`hidups` is linux-only, so it was replaced with `newhidups`, which was later renamed to `usbhid-ups` in v2.2.0)

I don't remember what Sarge did for USB permissions, but `newhidups` uses a different /dev node than `hidups`.

fr3ddie, 2016-01-11:
I will be able to try this after January 23rd, unfortunately I'm not where the UPS is located in this moment.
NUT v2.7.2 available on Jessie's repos is fine for the test?

Do you suspect it could not work with new versions of the driver?

clepple, 2016-01-12:
It is very likely that it will work. There seem to be a number of outstanding bugs related to upgrading and the udev scripts, but it should work with jessie.

fr3ddie, 2016-02-15:
I'm sorry for the delay but I'm still not back to the UPS location...
Please be patient.

fr3ddie, 2016-09-13:
I hope you can excuse the incredible delay......

As far as I understand it works pretty well:

xxx@yyy ~$ lsb_release -a
LSB Version:	core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch:security-4.0-amd64:security-4.0-noarch:security-4.1-amd64:security-4.1-noarch
Distributor ID:	Debian
Description:	Debian GNU/Linux 8.5 (jessie)
Release:	8.5
Codename:	jessie

xxx@yyy ~$ uname -a
Linux titanium 3.16.0-4-amd64 #1 SMP Debian 3.16.36-1+deb8u1 (2016-09-03) x86_64 GNU/Linux

                                                                               
xxx@yyy ~$ dpkg -l nut
Voluto=U (non noto)/I (installato)/R (rimosso)/P (rimosso totale)/H (in attesa)
| Stato=Non/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(nessuno)/R (reinstallazione richiesta) (Stato,Err: maiuscolo=grave)
||/ Nome                                      Versione                  Architettura              Descrizione
+++-=========================================-=========================-=========================-========================================================================================
ii  nut                                       2.7.2-4                   all                       network UPS tools - metapackage

xxx@yyy ~$ upsc UPS@localhost
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 20
battery.runtime: 2840
battery.type: PbAc
device.mfr: EATON
device.model: Ellipse PRO 650 
device.serial: Gxxxxxxxx
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/usb/hiddev0
driver.version: 2.7.2
driver.version.data: MGE HID 1.33
driver.version.internal: 0.38
input.frequency: 50.0
input.transfer.high: 285
input.transfer.low: 165
input.voltage: 233.0
input.voltage.extended: no
outlet.1.desc: PowerShare Outlet 1
outlet.1.id: 2
outlet.1.status: on
outlet.1.switchable: no
outlet.2.desc: PowerShare Outlet 2
outlet.2.id: 3
outlet.2.status: on
outlet.2.switchable: no
outlet.desc: Main Outlet
outlet.id: 1
outlet.switchable: no
output.frequency: 50.0
output.frequency.nominal: 50
output.voltage: 230.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.firmware: 01.10.0017
ups.load: 10
ups.mfr: EATON
ups.model: Ellipse PRO 650 
ups.power: 65
ups.power.nominal: 650
ups.productid: ffff
ups.realpower: 40
ups.serial: Gxxxxxxxxxx
ups.status: OL CHRG
ups.timer.shutdown: 0
ups.timer.start: 0
ups.vendorid: 0463

xxx@yyy ~$ upscmd -l UPS@localhost 
Instant commands supported on UPS [UPS]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
                                                                               

xxx@yyy ~$ upscmd UPS@localhost beeper.enable
Username (zzz): zzz
Password: 
OK

closed 2016-11-03 by clepple