networkupstools/nut #258: Segfault on Ubuntu 14.04 with usbhid-ups on APC Backup XS 1500G

bandroidx, 2016-01-24:
I tried with the default package in ubuntu 14.04 x64 server and also tried this PPA: https://launchpad.net/~clepple/+archive/ubuntu/nut which gave me version 2.7.2.9~really2.7.3-2

Both get this segfault and error:

Jan 23 19:06:05 librenms kernel: [ 97.828615] usb 2-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
Jan 23 19:06:06 librenms kernel: [ 98.831615] usb 2-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
Jan 23 19:06:07 librenms kernel: [ 99.834556] usb 2-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
Jan 23 19:06:07 librenms kernel: [ 99.834606] usbhid-ups[2600]: segfault at 0 ip 00007f0e49daf48c sp 00007ffffc38a148 error 4 in libc-2.19.so[7f0e49c6e000+1bb000]

After this I tried apcupsd to make sure hte issue wasnt hardware related in some way, apcupsd is working perfectly so i dont think its hardware related.

bandroidx, 2016-01-24:
I tried rebooting and double checked the udev rules are in place as well.

bandroidx, 2016-01-24:
Ok this is weird, i just installed centos 7 to test and installed the package from epel and i get the exact same segfault....
usbhid-ups[8429]: segfault at 0 ip 00007f7d60b5cdcc sp 00007ffc8ac1e398 error 4 in libc-2.17.so[7f7d60a2a000+1b6000]                                                                                                                      

bandroidx, 2016-01-24:
i should add this is running inside a vmware esxi 5.5 virtual machine with usb pass through enabled for the APC ups.  However liek i said, apcupsd works fine so I think that the vm setup is correct.

bandroidx, 2016-01-24:
Here is the version info:

Network UPS Tools - UPS driver controller 2.7.2
   0.000000     Starting UPS: myups
Network UPS Tools - Generic HID driver 0.38 (2.7.2)
USB communication driver 0.32
kill: No such process
   3.021595     Driver exited abnormally
                                                                                                                                                                                                                                             

clepple, 2016-01-24:
As root, please do the following:
```
killall usbhid-ups   # just in case
/lib/nut/usbhid-ups -DDD -a myups
```

and post the last 20 lines or so before it crashes.

Either OS should work, but I have a copy of Ubuntu 14.04 already running.

Also, which kernel version? Either `uname -a` or the package version. (Ubuntu has several options.)

bandroidx, 2016-01-24:
Linux librenms 3.19.0-47-generic #53~14.04.1-Ubuntu SMP Mon Jan 18 16:09:14 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux                                                                                                                     

Network UPS Tools - Generic HID driver 0.39 (2.7.3)
USB communication driver 0.32
   0.000000     debug level is '3'
   0.000530     upsdrv_initups...
   0.024464     Checking device (1D6B/0002) (002/001)
   0.024495     - VendorID: 1d6b
   0.024499     - ProductID: 0002
   0.024501     - Manufacturer: unknown
   0.024504     - Product: unknown
   0.024506     - Serial Number: unknown
   0.024508     - Bus: 002
   0.024510     Trying to match device
   0.024517     Device does not match - skipping
   0.024523     Checking device (051D/0002) (001/004)
   4.044137     - VendorID: 051d
   4.044172     - ProductID: 0002
   4.044178     - Manufacturer: unknown
   4.044183     - Product: unknown
   4.044187     - Serial Number: unknown
   4.044192     - Bus: 001
   4.044197     Trying to match device
Segmentation fault (core dumped)
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               

(PS. i managed to get it working on centos7 some how but prefer to use this ubuntu vm, let me know what lese you need)

bandroidx, 2016-01-24:
dmesg when the ups is connected:

[ 8757.427455] usb 1-2.1: new full-speed USB device number 4 using uhci_hcd
[ 8757.774277] usb 1-2.1: New USB device found, idVendor=051d, idProduct=0002
[ 8757.774285] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8757.774289] usb 1-2.1: Product: Back-UPS XS 1500G FW:866.L8 .D USB FW:L8
[ 8757.774293] usb 1-2.1: Manufacturer: American Power Conversion
[ 8757.774295] usb 1-2.1: SerialNumber: 4B1309P30999
[ 8758.250172] hid-generic 0003:051D:0002.0006: hiddev0,hidraw2: USB HID v1.00 Device [American Power Conversion Back-UPS XS 1500G FW:866.L8 .D USB FW:L8 ] on usb-0000:02:01.0-2.1/input0
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    

clepple, 2016-01-24:
> 0.024523 Checking device (051D/0002) (001/004)
> 4.044137 - VendorID: 051d
> 4.044172 - ProductID: 0002
> 4.044178 - Manufacturer: unknown
> 4.044183 - Product: unknown
> 4.044187 - Serial Number: unknown

This is odd. It should not take four seconds to retrieve the device descriptor and three string descriptors. Does this match the time of the "USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110" dmesg errors? (-110 is ETIMEDOUT)

What does `sudo lsusb -d051d:` return?

Also: `ls -l /dev/bus/usb/001/004`

bandroidx, 2016-01-24:
yes it matches the USBDEVFS_CONTROL failed message time.

root@librenms:/etc/nut# sudo lsusb -d051d:
Bus 001 Device 004: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
                                                                                                                                       
root@librenms:/etc/nut# ls -l /dev/bus/usb/001/004
crw-rw-r-- 1 root nut 189, 3 Jan 23 21:54 /dev/bus/usb/001/004
                                                                                                                      

bandroidx, 2016-01-24:
it was doing the same thing on centos 7.  oddly, i ran upsdrvinstall with -u root and it worked but didnt work, then after  that i was able to run it normally and everything started working.

clepple, 2016-01-24:
> sudo lsusb -d051d:

Sorry, I forgot a flag: `lsusb -vv -d051d:`

I see a potential problem in the code, but if there is some setting in ESXi that will give full control of the USB device to the guest, that would be the simplest way. Those "unknown"s are NULL pointers, and real hardware would return an actual string.

bandroidx, 2016-01-24:
root@librenms:/etc/nut# lsusb -vv -d051d:

Bus 001 Device 004: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x051d American Power Conversion
  idProduct          0x0002 Uninterruptible Power Supply
  bcdDevice            0.90
  iManufacturer           1 American Power Conversion
  iProduct                2 Back-UPS XS 1500G FW:866.L8 .D USB FW:L8
  iSerial                 3 4B1309P30999
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength    1134
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             100
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 

As far as i know esxi gives full control of the usb device, its a pass-through.

mkuron, 2016-01-24:
I'm seeing the same after upgrading from Debian 6 to Debian 8. I am also running this on VMware ESXi, however I'm on version 6.0.

```
[   15.552464] usb 1-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
[   17.926078] usb 1-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
[   18.926713] usb 1-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
[   18.926732] usbhid-ups[489]: segfault at 0 ip b76f0aa4 sp bffcdc08 error 4 in libc-2.19.so[b75b0000+1a7000]
```

```
$ /lib/nut/usbhid-ups -DDD -a apc
Network UPS Tools - Generic HID driver 0.38 (2.7.2)
USB communication driver 0.32
   0.000000	debug level is '3'
   0.000441	upsdrv_initups...
   0.033178	Checking device (1D6B/0002) (002/001)
   0.045042	- VendorID: 1d6b
   0.045148	- ProductID: 0002
   0.045274	- Manufacturer: unknown
   0.045381	- Product: unknown
   0.045434	- Serial Number: unknown
   0.045483	- Bus: 002
   0.045532	Trying to match device
   0.045596	Device does not match - skipping
   0.056994	Checking device (051D/0002) (001/004)
   3.060760	- VendorID: 051d
   3.060779	- ProductID: 0002
   3.060782	- Manufacturer: unknown
   3.060784	- Product: unknown
   3.060786	- Serial Number: unknown
   3.060788	- Bus: 001
   3.060790	Trying to match device
Segmentation fault
```

```
$ lsusb -vv -d051d:

Bus 001 Device 004: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x051d American Power Conversion
  idProduct          0x0002 Uninterruptible Power Supply
  bcdDevice            0.90
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 3 (error)
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength    1134
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             100
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```
Perhaps it's worth noting that _lsusb_ takes about 6 seconds to run, which is a lot longer than usual. Also, _iManufacturer_, _iProduct_ and _iSerial_ are reported as errors, unlike what @bandroidx saw. Do you thing this is a bug in libusb or the kernel that only shows up on ESXi?

clepple, 2016-01-24:
> Perhaps it's worth noting that lsusb takes about 6 seconds to run, which is a lot longer than usual.

Agreed - that extra delay is probably the sum of all of the string descriptor request timeouts.

> Also, iManufacturer, iProduct and iSerial are reported as errors, unlike what @bandroidx saw. Do you thing this is a bug in libusb or the kernel that only shows up on ESXi?

Unclear at this point.

For Ubuntu, `lsusb` seems to be linked against `libusb-1.0`. What does "ldd $(which lsusb)" return on Debian?

mkuron, 2016-01-24:
Same: `/lib/i386-linux-gnu/libusb-1.0.so.0`

clepple, 2016-01-24:
@mkuron do you get the same error when you run lsusb as root? Installing NUT tends to bring in udev rules that change permissions on the UPS /dev/bus/usb node.

mkuron, 2016-01-24:
Yes, lsusb was running as root.

clepple, 2016-01-24:
@bandroidx the Ubuntu 14.04 x64 segfault seems to be in `__strncmp_ssse3()`:
```
addr2line -f -e /usr/lib/debug/lib/x86_64-linux-gnu/libc-2.19.so $(printf "0x%x" $((0x00007f0e49daf48c - 0x7f0e49c6e000)))
__strncmp_ssse3
/build/buildd/eglibc-2.19/string/../sysdeps/x86_64/multiarch/../strcmp.S:209
```

How do you want to fix this? Since your UPS isn't on the `tweak_max_report` list (Back-UPS ES 525, Back-UPS CS), we should be able to zero out a few bytes in the driver executable. You could do that in a hex editor, or with `dd`. I can also try to push a new build to the PPA, but that could take a while (I have switched machines since I last did that).

clepple, 2016-01-24:
@mkuron I don't have a 32-bit Debian 8 VM on hand. Can you install `libc6-dbg` and run the following command to verify that you are seeing the exact same crash?

```
$ addr2line -f -e /usr/lib/debug/lib/i386-linux-gnu/libc-2.19.so $(printf "0x%x" $((0xb76f0aa4-0xb75b0000)))
```

Same question for you, except that the PPA isn't an option. How do you want to fix this? Do you have everything set up to rebuild NUT from source? Binary patching is an option for quick testing.

clepple, 2016-01-24:
Looks like a similar bug was filed on Launchpad: https://bugs.launchpad.net/ubuntu/+source/nut/+bug/1483615

mkuron, 2016-01-25:
That commit fixes it, thanks!
I cherry-picked f679a2820e49cbdec4138295de5e8947ee953a8a on top of v2.7.2 and rebuilt just the driver. Do you want to file a bug report with Debian for them to include the patch, or shall I do that?

clepple, 2016-01-25:
> Do you want to file a bug report with Debian for them to include the patch, or shall I do that?

@mkuron since it's on your system, could you please file it, and let us know the bug number?

I vaguely recall Debian's reportbug grabbing some other package versions to include in its report for NUT bugs, so you might want to try that (even if you just submit the output manually).

Thanks for testing!

bandroidx, 2016-01-25:
@mkuron if it was an issue that showed up with esxi specifically I would expect apcupsd to have the same issue though, no?  Have you tried apcupsd to see if it works for you?  I personally can't use apcupsd because one of the devices I am working with is a synology nas which needs NUT, but it was working for me.

bandroidx, 2016-01-25:
@clepple what does added to 2.7.4 milestone mean?  Does that mean the bug will be fixed in the next release or has it already been patched in git master?

mkuron, 2016-01-25:
Haven't tried apcupsd yet, but considering that lsusb takes so long I'm pretty sure this is an issue in the underlying libusb or kernel. lsusb uses libusb too.

clepple, 2016-01-25:
@bandroidx yes, the plan is to include the fix as part of 2.7.4 (as opposed to a larger bug which would break things in the 2.7.x series, and would need to be deferred). I was going to wait to merge it to master until after you confirm that this fixes your issue (although I am fairly certain it will), since GitHub will close the issue when I merge from the branch with commit f679a28. However, since Ubuntu is still on 2.7.1, that's why I was asking how you wanted to work around the issue for now.

clepple, 2016-01-25:
@mkuron note that NUT still uses libusb-0.1, so I think the underlying issue is elsewhere.

mkuron, 2016-01-25:
Here's the Debian bug report: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=812604. I've included your patch.

bandroidx, 2016-01-25:
@clepple sorry github refreshed strangely and i missed some of your responses, just saw them now when i did a full refresh of the page.  I am open to fixing it in any manor really, I am using your PPA now but if you want me to pull master from github and then patch it, I could do that as well.  I guess the question is, what do you need me to do?

bandroidx, 2016-01-25:
@clepple I am running in to a lot of errors trying to compile from git master so if i can just make a simple hex edit to the driver that would be good

bandroidx, 2016-01-25:
@clepple i managed to get the driver built finally and replaced usbhid driver in /lib/nut, it doesnt crash when loading the driver now but I still get this in dmesg:
[168838.162712] usb 1-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
[168839.165206] usb 1-2.1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -110
                                                                                                                                                                                                                  

Also, even though the driver is running i get 
Error: Driver not connected                             

when trying to run upsc

bandroidx, 2016-01-25:
I got it working!  even though i had --with-pidpath=/var/run/nut it was putting the pids in /var/state/nut with the new driver i built.  i symlinked them together and now its working!

clepple, 2016-01-26:
> @clepple i managed to get the driver built finally and replaced usbhid driver in /lib/nut, it doesnt crash when loading the driver now but I still get this in dmesg:
> [168838.162712] usb 1-2.1: [...]

These are going to keep showing up until the underlying problem with the USB passthrough is fixed.

clepple, 2016-01-26:
> I got it working! even though i had --with-pidpath=/var/run/nut it was putting the pids in /var/state/nut with the new driver i built. i symlinked them together and now its working!

Glad it's working!

Not sure what to tell you on the PID path, except that sometimes you have to run "make clean" after changing ./configure options.

bandroidx, 2016-01-26:
@clepple the symbolic link of the directory is working good enough, I am just going to leave it.  Thanks so much for your help!

closed 2016-01-26 by clepple

mkuron, 2016-02-14:
Just wanted to add that installing kernel 4.3 from the Debian Jessie backports fixes the `lsusb` problems (it's slow and the output shows errors for _iManufacturer_, _iProduct_ and _iSerial_). However, `usbhid-ups` without the patch still crashes. With the patch, of course, it's all good.

clepple, 2016-02-14:
> However, usbhid-ups without the patch still crashes.

Does usbhid-ups print the strings at startup with the new kernel, or does it print "null"?

mkuron, 2016-02-15:
```
$  lsusb -vv -d051d:

Bus 001 Device 004: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x051d American Power Conversion
  idProduct          0x0002 Uninterruptible Power Supply
  bcdDevice            0.90
  iManufacturer           1 American Power Conversion
  iProduct                2 Back-UPS RS 900G FW:879.L4 .I USB FW:L4  
[...]
$ /lib/nut/usbhid-ups -a apc -DDD
[...]
   0.059700	Checking device (051D/0002) (001/004)
   3.064133	- VendorID: 051d
   3.064334	- ProductID: 0002
   3.064444	- Manufacturer: unknown
   3.064528	- Product: unknown
   3.064579	- Serial Number: unknown
   3.064631	- Bus: 001
   3.064681	Trying to match device
   3.064735	device->Product is NULL so it is not possible to determine whether to activate max_report_size workaround
   3.064790	Device matches
```
This is version 2.7.2 with the patch cherry-picked.

maaboo, 2016-02-16:
Please, could you tell me how to apply the patch, step by step? At least in general. I'm not an expert in Linux.

I have this issue right now. When I switch to xHCI, NUT starts to lose UPSes, but can see them. When I use eHCI - I'm getting segfault.

clepple, 2016-02-17:
@maaboo which distribution? (Be sure to file a bug upstream if `lsusb` has stopped showing USB strings, too.)

maaboo, 2016-02-17:
@clepple Nevermind, I've found manual here:

http://vladimir-stupin.blogspot.ru/2015/09/nut-apc-smart-ups-1500va.html

It's exactly about the issue. I can confirm that it's working correctly now. at least for 5 minutes. :)