networkupstools/nut #260: Support for Automatic Transfer Switches

maaboo, 2016-02-17:
Is there any chance to get NUT supporting ATS (at least from APC)? I have one and it's the only UPS related thing that isn't centrally monitored.

I've tried to add it like usual UPS:

[ups.conf]
`[UPS-APCATS]`
        `driver = snmp-ups privprotocol=AES`
        `port = 10.200.0.250`
        `desc = "RACK-01: APC Automatic Transfer Switch"`

[hosts.conf]
`MONITOR UPS-APCATS@mntr "RACK-01: APC Automatic Transfer Switch"`

but got errors like:

> Feb 17 11:03:59 mntr upsmon[1136]: Fatal error: insufficient power configured!
Feb 17 11:03:59 mntr upsmon[1136]: Sum of power values........: 0
Feb 17 11:03:59 mntr upsmon[1136]: Minimum value (MINSUPPLIES): 1

I could provide any info at your request, I want to get my equip working and also help NUT grow.

Please, mark this as feature request.

aquette, 2016-02-18:
hi @maaboo ,support for ATS / STS is indeed something planned, and for soon now.
However, I'd need to ensure your use case, to better understand.
Also, if you could extract the info using this procedure, that would help a lot:
http://networkupstools.org/docs/developer-guide.chunked/ar01s04.html#snmp-subdrivers

If you need more info, don't hesitate to tell me.

aquette, 2016-02-25:
hi @maaboo 

in case you have some issues to collect the data I asked:
- download and save https://raw.githubusercontent.com/networkupstools/nut/master/scripts/subdriver/gen-snmp-subdriver.sh into a directory "A"
- also put the MIB file of the APC ATS in the same directory "A"
- from that directory "A", execute the following command
./gen-snmp-subdriver.sh -n apc_ats -H <ip address of the ATS>

and send me the resulting .c/.h file

maaboo, 2016-02-25:
Yes, thanks. I need some time to do it. I was busy with another product bug request.

I'll send you info in a few days.

aquette, 2016-02-26:
perfect, thanks. Note that I'm about to push the preliminary (though already complete enough) support for Eaton ATS. Will update the comment to point at the commit, probably today

aquette, 2016-03-04:
@maaboo : we're approaching the end of the tasks list. Any news or should I go ahead with releasing 2.7.4?

maaboo, 2016-03-04:
Could you give me two more days?

aquette, 2016-03-04:
@maaboo : should be good, I expect to release beginning of next week.

maaboo, 2016-03-05:
For some reason I haven't managed to get info with first method, so I used snmpwalk and then the script. As well as I couldn't get the name like "apc_ats". I've attached the resulting files along the MIB I used.

[https://www.dropbox.com/s/3xgk9i00xsswwlu/apcats.7z?dl=0](url)

aquette, 2016-03-07:
Hi @maaboo 
thx for your feedback, I've done a 1rst version of APC ATS support.
Could you please test the branch and send back the output of upsc / upsrw?

The branch is: https://github.com/networkupstools/nut/tree/ats_support
just "git clone https://github.com/networkupstools/nut.git", then "git checkout ats_support" and configure / compile as usual. If you need more directions, please ping me back.

Note on this 1rst implementation:
- ups.status maps on the output voltage, so either "OL" (ok) or "OFF" (fail). This data is needed for upsmon, however since this is a preliminary support, not sure if it will be enough to actually do something.
- since there is something for that in NUT now, I've added outlet.group collection to handle the output bank.
- there are 2 commands available, to select the preferred input source and the sensitivity. Would be good if you can test it too.

maaboo, 2016-03-07:
Unfortunately there are holydays in our coutry until 9th of May. I shut down my test server (it's a physical machine) so I can't do these tests. :(

aquette, 2016-03-07:
@maaboo : that's sad. I'll merge the pull request, mentioning that APC ATS should also be supported, and calling for tests. I'll also leave the present ticket opened, and will move it to 2.7.5. Please update it whenever you can test.

aquette, 2016-08-12:
@maaboo did you have a chance to test this finally. We're approaching a candidate for 2.7.5...

maaboo, 2016-08-12:
Okay. What should I do? Just install the app and check for error?

aquette, 2016-08-12:
@maaboo indeed, install 2.7.4 (either packaged or from source), configure an entry 
[apc-ats]
 driver = snmp-ups
 port = <ip@>

then run upsd and snmp-ups, and send back an upsc output and possibly a driver debug output if a problem arise (if so, launch snmp-ups manually using "-DDD -a apc-ats)
thx and cheers

maaboo, 2016-08-12:
Sorry, but could you provide deb package of the latest version?

aquette, 2016-08-16:
@maaboo attached is a patched x86_64 build which includes support for APC ATS too.
I've only shipped nut-snmp, not the rest.
please, remove the suffix .png after .deb (otherwise, upload is blocked)
[image]

maaboo, 2016-08-16:
Running upsd gives that:

`listening on 127.0.0.1 port 3493
listening on 10.10.100.10 port 3493
Connected to UPS [M-UPS-04]: snmp-ups-M-UPS-04
`
I haven't noticed any errors at the moment. If you could say where to look, I check for them again.

Web-console shows:
[image]
[image]

aquette, 2016-08-16:
looks great, thanks for your feedback.
Could you please also post the output of upsc / upsrw for this unit, so that I can publish it on the [Device Dumps Library](www.networkupstools.org/ddl/)? Thx

maaboo, 2016-08-16:
Sure:

=========
~# upsc M-UPS-04@10.239.7.14:3493
Init SSL without certificate database
device.mfr: APC
device.model: AP7724
device.serial: 5A1516T15268
device.type: ats
driver.name: snmp-ups
driver.parameter.pollinterval: 2
driver.parameter.port: 10.239.1.250
driver.parameter.synchronous: no
driver.version: 2.7.4.1
driver.version.data: apc_ats MIB 0.2
driver.version.internal: 0.99
input.1.frequency: 50.00
input.1.voltage: 221.00
input.2.frequency: 50.00
input.2.voltage: 221.00
input.count: 2.00
input.frequency.nominal: 50.00
input.sensitivity: high
input.source: 2.00
input.source.preferred: 2.00
input.voltage.nominal: 230.00
outlet.group.1.current: 7.70
outlet.group.1.id: 1.00
outlet.group.1.name: total
outlet.group.1.realpower: 1701.00
outlet.group.1.status: OL
outlet.group.1.voltage: 221.00
outlet.group.2.current: 4.00
outlet.group.2.id: 2.00
outlet.group.2.name: bank1
outlet.group.2.realpower: 884.00
outlet.group.2.status: OL
outlet.group.2.voltage: 221.00
outlet.group.3.current: 3.80
outlet.group.3.id: 3.00
outlet.group.3.name: bank2
outlet.group.3.realpower: 839.00
outlet.group.3.status: OL
outlet.group.3.voltage: 221.00
outlet.group.count: 3.00
output.frequency: 50.00
output.voltage: 221.00
ups.firmware: R01
ups.firmware.aux: 3.0.5
ups.id: m-ups-04
ups.mfr.date: 04/18/2015
ups.status: OL
========

========
:~# upsrw -u user -p password M-UPS-04
[input.sensitivity]
Input power sensitivity
Type: ENUM
Option: "high" SELECTED
Option: "low"

[input.source.preferred]
Description unavailable
Type: UNKNOWN (unrecognized)
========

aquette, 2016-08-17:
@maaboo perfect, thanks.

closed 2016-08-17 by aquette