networkupstools/nut #261: Weird upsd users behavior on Windows

Yvan-Masson, 2016-02-17:
First, thank you very much for this software ! Also, please apologize for the long story…

I have upsd (nut-client/server 2.7.2-4) installed on Debian Jessie, which monitors 2 UPS:
```
[DellH914n1]
        driver = usbhid-ups
        port = auto
        serial = "CN-0H914N-75162-05T-0085-000"
        desc = "blabla"
        sdorder = 0

[DellH914n2]
       driver = usbhid-ups
       port = auto
       serial = "CN-0H914N-75162-12Q-0129-001"
       desc = "blabla"
       sdorder = 1
```

Two users are configured (password are changed):
```
[writeuser]
        password = xxxx
        actions = SET
        instcmds = ALL
        upsmon master
[readuser]
        password = yyyy
        upsmon slave
```

On this same server, upsmon is configured like this:
```
MONITOR DellH914n1@localhost 1 writeuser xxxx master
MONITOR DellH914n2@localhost 1 writeuser xxxx master
```

On another Debian Jessie server (nut-client 2.7.2-4), upsmon is configured like this:
```
MONITOR DellH914n1@<master's-IP-adress> 1 readuser yyyy slave
MONITOR DellH914n2@<master's-IP-adress> 1 readuser yyyy slave
```

On a pfSense (FreeBSD), upsmon is configured by web interface, but with equivalent settings than first line above (only one UPS is monitored).

I also wanted to configure Windows servers (2003 and 2008) as nut slaves (Beta NUT 2.6.5-6), so I tried to replicate the upsmon configuration, but I had to mix user and password to make it work:
```
MONITOR `DellH914n1@<master's-IP-adress>` 1 readuser xxxx slave
MONITOR `DellH914n2@<master's-IP-adress>` 1 readuser xxxx slave
```

Without that, access was denied: I found this working setting by mistake…

This week, I did a real test, which … mostly worked in my sense:

- All servers detected that the two UPS where "On battery".
- DellH914n1 failed just before being on "Low battery" state, killing the nut master and the others Unix servers (I did not check batteries before doing that, my fault)
- Windows servers could not reach upsd anymore and so shut down properly.
- DellH914n2 detected that there was no charge anymore and so shutdown on his own.

After reboot, the situation evolved:

- DellH914n2 is not detected anymore on USB, either with "lsusb", either with official Windows software. I have tried a reset to factory setting with no luck.
- To make my nut installation kind of work again, I removed DellH914n2 from all files, to keep only the "working" UPS, DellH914n1. This also made me change "MINSUPPLIES 2" to "MINSUPPLIES 1" in upsmon.conf.
- And finally, on Windows servers, I had to change the password in upsmon config to make it work again, so now the right password is needed…:
```
MONITOR `DellH914n1@<master's-IP-adress>` 1 readuser yyyy slave
```

So, there is in some code a weird things with users, which is linked with UPS definitions or upsmon definitions or Windows port. Unfortunately, as my second UPS is not detected anymore, I can't test again (but I am sure of this, the automatic shutdown worked !). Do not hesitate to ask if needed.

Regards,
Yvan

aquette, 2016-02-18:
hi @Yvan-Masson , not had time to go through your whole feedback, but a thing to try to restore the comm with your USP (n2):
- power off the UPS
- unplug the power cord of the UPS 
- wait 10 seconds,
- plug everything back and restart operations...

Yvan-Masson, 2016-02-19:
Thanks for the advice: I will try this, but I need to shut down many servers, so it will not be before two weeks. If USB connection works again and so if I can try to reproduce the bug, I will let you know.

aquette, 2016-02-23:
thx @Yvan-Masson 
btw, you shouldn't mix the administrative and upsmon users.
I.e, you should have something like that:
[writeuser]
        password = xxxx
        actions = SET
        instcmds = ALL
[readuser]
        password = yyyy
[monmaster]
        password = xxxx
        upsmon master
[monslave]
        password = yyyy
        upsmon slave

that separates clearly between the protection side (upsmon) and the administrative (users to get / set data)

Yvan-Masson, 2016-03-07:
Hi,
Thanks again Aquette for advices, I could revive my UPS by unplugging and rebooting.

"As expected", I can not reproduce the bug: passwords now match users even on Windows slaves. I really don't understand what happened. Anyway, sorry for the noise, you can close this bug.

Best regards,
Yvan

aquette, 2016-03-08:
Hi @Yvan-Masson 
thanks for your feedback. Glad to hear that this solution worked.
I'm closing this bug, don't hesitate to reopen if the issue happens again

closed 2016-03-08 by aquette