networkupstools/nut #262: Please improve Windows port documentation/packaging

Yvan-Masson, 2016-02-17:
Hi,

Thanks very much for this great software.

I just set up a nut infrastructure with a Linux master and Linux / Windows slaves. The configuration of upsmon on Windows (2003 and 2008 servers, 32 bits) was difficult, and I wanted to report how I did so that documentation / packaging could be improved. Please apologize if all of this is already written somewhere.

- Download and install normally the "Beta NUT 2.6.5-6 package for Windows" (usb driver is not needed in our case)
- Download from OpenSSL GitHub (https://github.com/openssl-net/openssl-net/tree/master/native/windows/x86) libeay32.dll and ssleay32.dll, and put these in `<NUT-install-directory>\sbin\`. It seems that Microsoft Visual C++ 2010 Redistributable Package (x86) is needed (I did not have to install it, so I think it is default on Windows).
- Copy libgcc_s_dw2-1.dll from `<NUT-install-directory>\bin\` to `<NUT-install-directory>\sbin\` (this is not always needed, the library might be already installed by another software and available globally).
- Edit nut configuration files as usual.
- Start service with the entry created in the menu.
- Result can be monitored from "Applications" in the "Event viewer".

Regards,
Yvan

mrudat, 2016-09-18:
It sounds more like a packaging issue than a documentation issue to me; if the SSL libraries are a requirement for NUT's operation, then they should be installed as a part of installing NUT, either by bundling the libraries into the installer, or downloading and installing the latest version during installation.

int-red, 2018-02-04:
Great work! Your findings helped me to get NUT running on Windows Server 2016.

The SSL Libraries you where referencing did not work. I got the error that they are not valid Win32 binaries.
I got the "official" libraries from OpenSSL.org: https://wiki.openssl.org/index.php/Binaries
Direkt link is
https://indy.fulgan.com/SSL/openssl-1.0.2n-i386-win32.zip
and they worked fine.

But a shame that two years later the installer is still not fixed.

kevin201, 2020-03-06:
+1 

I found instructions here: http://jtkdev.com/wp/2016/04/10/nut-for-windows/ before looking here.

Using nut as a slave on Windows.

rchovan, 2020-03-10:
@kevin201 HI, I'm trying windows version too, I want to have my windows server as slaves, and shut them down with upssched, but it does not work. I can monitor ups, communicate with ups, see logs in event viewer, but can't shutdown client after specific time to preserve runtime for more important servers. 

my upssched.conf
```
CMDSCRIPT C:\\NUT\bin\\upssched-cmd
PIPEFN C:\\NUT\\upssched.pipe
LOCKFN C:\\NUT\\upssched.lock
AT ONBATT ups_1000 START-TIMER onbattwarn 30
AT ONLINE ups_1000 EXECUTE ups-back-on-line
```

kevin201, 2020-03-10:
I suggest you look at the SHUTDOWN command. I never got that far, as when I tested the 'upsmon -c fsd', my APC UPS went into a continuous cycle of 'onbatt, power off, power on' and messed up my systems! I changed to apcupsd, which is actively maintained and aimed at APC kit.

Had I not gone down that route, I was going to look at the old Windows binaries here: 
https://code.google.com/archive/p/winnut/downloads
to see if that would give me any more clues.

dotRyan, 2020-08-19:
@kevin201 I was unable to connect the windows port as a slave to a linux server running 2.7.4.
Is that what you were running when you had it working?

kevin201, 2020-08-19:
@dotRyan Sorry, I can't tell. I would have been using whatever was the current release at the time, but I never got as far as testing it properly before I moved to apcupsd.