networkupstools/nut #266: Study the possibility to default to type 'STRING' for variables

aquette, 2016-03-04:
Quoting @zykh from #265:
"I know that most of our vars are numbers and so it's easier to only edit the type for the few string ones, but maybe it would be better to have 'STRING' as the default type, since a number can always be stored in a string (well.. not always true, but pretend it to be), but not viceversa: if, for some reason, we lie (like me forgetting to set a var as the correct base type STRING/NUMERIC), then if the default type is STRING and the var is not, we're not doing something as bad as if the default type were NUMERIC and the var a string."

From @aquette:
"defaulting to string would be interesting, but would require to define explicitly a ST_FLAG_NUMBER and to apply it everywhere (where there is no ST_FLAG_STRING) to be useful."

aquette, 2016-08-16:
Moving this one for 2.7.6, too intrusive to complete #267

jimklimov, 2024-07-22:
On a related note:
````
LIST VAR eco650
...
VAR eco650 driver.version.usb "libusb-1.0.24 (API: 0x1000108)"
...

GET TYPE eco650 driver.version.usb
TYPE eco650 driver.version.usb NUMBER
````

jimklimov, 2024-07-22:
While here, the "IMMUTABLE" flag is also an internal detail that can impact remote users (values that can only be set once).