networkupstools/nut #291: daisychain: documentation does not build

clepple, 2016-05-22:
Related to #287: the daisychain branch does not currently build.

From [buildbot](http://buildbot.networkupstools.org/public/nut/builders/MacOSX-10.11-x64/builds/1272/steps/compile/logs/stdio):

```
...
a2x: executing: "xmllint" --nonet --noout --valid "/sw.mav/var/lib/buildbot/buildslave-minime/minime-full.build/build/nut-v2.7.4-49-g6625539/_build/sub/docs/user-manual.xml"

/sw.mav/var/lib/buildbot/buildslave-minime/minime-full.build/build/nut-v2.7.4-49-g6625539/_build/sub/docs/user-manual.xml:3773: element link: validity error : IDREF attribute linkend references an unknown ID "daisychain"
```

clepple, 2016-05-22:
@aquette feel free to reassign; I saw your name on a few of the commits.

clepple, 2016-05-22:
Looks like this is not specific to the asciidoc version on the OS X builder: http://buildbot.networkupstools.org/public/nut/one_line_per_build?branch=daisychain

aquette, 2016-05-23:
@clepple : thanks for the report. I'm currently fixing it.

clepple, 2016-05-24:
Looks good to me.

closed 2016-05-24 by clepple