networkupstools/nut #292: Chloride deskpower 650 issue

SilentT-FR, 2016-05-29:
ive see this

https://github.com/networkupstools/nut/blob/master/data/driver.list.in#L173

i have test some serial cable but nothing work so i test with USB cable and blazer_usb and same probleme nothing work ..

clepple, 2016-05-29:
> but nothing work

Please provide more details. 

Which version of NUT?

Was the serial cable straight-through or special? Is there a pinout in the UPS manual?

Before switching to the USB port, did you completely power down the UPS?

Were there any debug messages from either driver ?

SilentT-FR, 2016-05-29:
After more test with serial cable its ok sorry for noise, but when i switch on usb cable no respond. 
I just replace blazer_sr to blazer_usb and the port

SilentT-FR, 2016-05-29:
Nut 2.7.2 from jessie deb

SilentT-FR, 2016-05-29:
And i have shutdown psu by cut powerline and on/off button

SilentT-FR, 2016-05-29:
If you want more info for debug say me how and i execute :)

aquette, 2016-05-30:
@SilentT-FR: quick question: when plugging the USB cable, have you removed the serial cable, at least from the UPS side? Otherwise, please test again USB without the serial cable plugged.
Then for USB communication with NUT:
- from a privileges standpoint, that should be good since "ID 06da:0002" is already listed for udev.
- the best would be to launch the driver in debug mode and attach the trace here, so that we have enough visibility to help.
For example, "blazer_usb -DDDDD -a <ups id in ups.conf>"
Another option would also be to test the new driver, that is replacing blazer_*: nutdrv_qx

SilentT-FR, 2016-05-30:
Yes i remove serial cable i reboot it and i plug usb cable

Network UPS Tools - Megatec/Q1 protocol USB driver 0.10 (2.7.1)
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/bus/usb/003/004"
   0.000134	debug level is '5'
   0.323218	Checking device (1D6B/0003) (004/001)
   0.323430	- VendorID: 1d6b
   0.323474	- ProductID: 0003
   0.323485	- Manufacturer: unknown
   0.323495	- Product: unknown
   0.323528	- Serial Number: unknown
   0.323568	- Bus: 004
   0.323605	Trying to match device
   0.323644	Device does not match - skipping
   0.323808	Checking device (06DA/0002) (003/004)
   0.324850	- VendorID: 06da
   0.324875	- ProductID: 0002
   0.324887	- Manufacturer: unknown
   0.324896	- Product: unknown
   0.324906	- Serial Number: unknown
   0.324915	- Bus: 003
   0.324923	Trying to match device
   0.324936	Device matches
   0.325585	send_to_all: SETINFO ups.vendorid "06da"
   0.325615	send_to_all: SETINFO ups.productid "0002"
   0.325636	send_to_all: SETINFO device.type "ups"
   0.325651	send_to_all: SETINFO driver.version "2.7.1"
   0.325666	send_to_all: SETINFO driver.version.internal "0.10"
   0.325680	send_to_all: SETINFO driver.name "blazer_usb"
   0.325692	Trying megatec protocol...
   0.326149	send: error sending control message: Protocol error
   0.326169	blazer_status: short reply
   0.326180	Status read 1 failed
   0.326590	send: error sending control message: Protocol error
   0.326613	blazer_status: short reply
   0.326626	Status read 2 failed
   0.327093	send: error sending control message: Protocol error
   0.327116	blazer_status: short reply
   0.327128	Status read 3 failed
   0.327137	Trying mustek protocol...
   0.327575	send: error sending control message: Protocol error
   0.327598	blazer_status: short reply
   0.327610	Status read 1 failed
   0.328094	send: error sending control message: Protocol error
   0.328116	blazer_status: short reply
   0.328129	Status read 2 failed
   0.328536	send: error sending control message: Protocol error
   0.328559	blazer_status: short reply
   0.328571	Status read 3 failed
   0.328581	Trying megatec/old protocol...
   0.329095	send: error sending control message: Protocol error
   0.329117	blazer_status: short reply
   0.329129	Status read 1 failed
   0.329559	send: error sending control message: Protocol error
   0.329582	blazer_status: short reply
   0.329594	Status read 2 failed
   0.330116	send: error sending control message: Protocol error
   0.330134	blazer_status: short reply
   0.330144	Status read 3 failed
   0.330153	Trying zinto protocol...
   0.330561	send: error sending control message: Protocol error
   0.330585	blazer_status: short reply
   0.330597	Status read 1 failed
   0.331093	send: error sending control message: Protocol error
   0.331116	blazer_status: short reply
   0.331128	Status read 2 failed
   0.331566	send: error sending control message: Protocol error
   0.331589	blazer_status: short reply
   0.331601	Status read 3 failed
   0.331611	No supported UPS detected

Network UPS Tools - Generic Q* USB/Serial driver 0.01 (2.7.1)
USB communication driver 0.32
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/bus/usb/003/005"
   0.000142	debug level is '5'
   0.001303	upsdrv_initups...
   0.001450	tcgetattr(/dev/bus/usb/003/005): Inappropriate ioctl for device

/dev/bus change because a un/plug usb cable on each reboot

SilentT-FR, 2016-05-30:
With another machine

root@pi:/lib/nut# ./blazer_usb -DDDDDDDDDD -a test
Network UPS Tools - Megatec/Q1 protocol USB driver 0.11 (2.7.2)
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/bus/usb/001/006"
   0.004062	debug level is '10'
   0.022080	Checking device (06DA/0002) (001/006)
   0.040324	- VendorID: 06da
   0.042274	- ProductID: 0002
   0.044884	- Manufacturer: ONEAC
   0.046807	- Product: USB Cable (V2.00)
   0.047034	- Serial Number: unknown
   0.047279	- Bus: 001
   0.048743	Trying to match device
   0.049051	Device matches
   0.051922	send_to_all: SETINFO ups.vendorid "06da"
   0.052294	send_to_all: SETINFO ups.productid "0002"
   0.053656	send_to_all: SETINFO device.type "ups"
   0.053955	send_to_all: SETINFO driver.version "2.7.2"
   0.054288	send_to_all: SETINFO driver.version.internal "0.11"
   0.055002	send_to_all: SETINFO driver.name "blazer_usb"
   0.055410	Trying megatec protocol...
   5.054867	send: error sending control message: Connection timed out
   5.055051	blazer_status: short reply
   5.055130	Status read 1 failed
   5.056511	send: error sending control message: Broken pipe
   5.625500	Device reset handled
   5.627766	blazer_status: short reply
   5.630421	Status read 2 failed
   5.636343	Checking device (06DA/0002) (001/006)
   5.653035	- VendorID: 06da
   5.654906	- ProductID: 0002
   5.656212	- Manufacturer: ONEAC
   5.657500	- Product: USB Cable (V2.00)
   5.658755	- Serial Number: unknown
   5.659059	- Bus: 001
   5.659401	Trying to match device
   5.660090	Device matches
   5.665505	send: Q1
   5.668406	read: timeout
   5.669112	blazer_status: short reply
   5.669922	Status read 3 failed
   5.670164	Trying mustek protocol...
  10.670807	send: error sending control message: Connection timed out
  10.671096	blazer_status: short reply
  10.671188	Status read 1 failed
  10.672475	send: error sending control message: Broken pipe
  11.240941	Device reset handled
  11.241241	blazer_status: short reply
  11.241332	Status read 2 failed
  11.246060	Checking device (06DA/0002) (001/006)
  11.263136	- VendorID: 06da
  11.265861	- ProductID: 0002
  11.267392	- Manufacturer: ONEAC
  11.268049	- Product: USB Cable (V2.00)
  11.269398	- Serial Number: unknown
  11.269667	- Bus: 001
  11.269865	Trying to match device
  11.271633	Device matches
  16.270892	send: error sending control message: Connection timed out
  16.271081	blazer_status: short reply
  16.271209	Status read 3 failed
  16.271352	Trying megatec/old protocol...
  16.272771	send: error sending control message: Broken pipe
  16.837214	Device reset handled
  16.839440	blazer_status: short reply
  16.841461	Status read 1 failed
  16.847789	Checking device (06DA/0002) (001/006)
  16.864510	- VendorID: 06da
  16.866307	- ProductID: 0002
  16.867668	- Manufacturer: ONEAC
  16.869117	- Product: USB Cable (V2.00)
  16.869394	- Serial Number: unknown
  16.870195	- Bus: 001
  16.870458	Trying to match device
  16.871456	Device matches
  16.876361	send: D
  16.878877	read: timeout
  16.879609	blazer_status: short reply
  16.880047	Status read 2 failed
  21.878822	send: error sending control message: Connection timed out
  21.879061	blazer_status: short reply
  21.879154	Status read 3 failed
  21.879240	Trying zinto protocol...
  21.880609	send: error sending control message: Broken pipe
  22.445414	Device reset handled
  22.447655	blazer_status: short reply
  22.449584	Status read 1 failed
  22.454832	Checking device (06DA/0002) (001/006)
  22.471602	- VendorID: 06da
  22.473583	- ProductID: 0002
  22.475593	- Manufacturer: ONEAC
  22.475846	- Product: USB Cable (V2.00)
  22.476676	- Serial Number: unknown
  22.476884	- Bus: 001
  22.477723	Trying to match device
  22.477952	Device matches
  27.478867	send: error sending control message: Connection timed out
  27.479063	blazer_status: short reply
  27.479272	Status read 2 failed
  27.480529	send: error sending control message: Broken pipe
  28.048111	Device reset handled
  28.050363	blazer_status: short reply
  28.052441	Status read 3 failed
  28.053616	No supported UPS detected
root@pi:/lib/nut# nano /etc/nut/ups.conf 
root@pi:/lib/nut# ./nutdrv_
nutdrv_atcl_usb  nutdrv_qx        
root@pi:/lib/nut# ./nutdrv_qx -DDDDDDDDD -a test
Network UPS Tools - Generic Q* USB/Serial driver 0.06 (2.7.2)
USB communication driver 0.32
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/bus/usb/001/007"
   0.002630	debug level is '9'
   0.014639	upsdrv_initups...
   0.016928	tcgetattr(/dev/bus/usb/001/007): Inappropriate ioctl for device
root@pi:/lib/nut# ./nutdrv_qx -DDDDDDDDD -a test

SilentT-FR, 2016-05-30:
some syslog 

May 30 10:51:27 pi kernel: [47508.429976] usb 1-1.3: reset low-speed USB device number 6 using dwc_otg
May 30 10:51:32 pi kernel: [47513.769906] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd blazer_usb rqt 33 rq 9 len 8 ret -110
May 30 10:51:32 pi kernel: [47514.029953] usb 1-1.3: reset low-speed USB device number 6 using dwc_otg
May 30 10:51:38 pi kernel: [47519.377834] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd blazer_usb rqt 33 rq 9 len 8 ret -110
May 30 10:51:38 pi kernel: [47519.637981] usb 1-1.3: reset low-speed USB device number 6 using dwc_otg
May 30 10:51:43 pi kernel: [47524.977888] usb 1-1.3: usbfs: USBDEVFS_CONTROL failed cmd blazer_usb rqt 33 rq 9 len 8 ret -110
May 30 10:51:44 pi kernel: [47525.238016] usb 1-1.3: reset low-speed USB device number 6 using dwc_otg

aquette, 2016-05-30:
@SilentT-FR thx for the traces.
Some more questions (I always have some ;)):
I see that dwc_otg in your syslog.
Could you try "dwc_otg.speed=1" as per:
http://raspberrypi.stackexchange.com/questions/4090/how-can-dwc-otg-speed-1-be-made-to-work

Or otherwise try a different USB port... We've already seen some such issue, tied to USB versions supported (v1.1 Vs v2 Vs v3)

SilentT-FR, 2016-05-30:
this option remove May 30 10:51:44 pi kernel: [47525.238016] usb 1-1.3: reset low-speed USB device number 6 using dwc_otg lines

root@pi:/lib/nut# ./blazer_usb -DDDDDDDDDD -a test
Network UPS Tools - Megatec/Q1 protocol USB driver 0.11 (2.7.2)
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/bus/usb/001/007"
   0.002112	debug level is '10'
   0.017469	Checking device (06DA/0002) (001/007)
   0.034758	- VendorID: 06da
   0.036516	- ProductID: 0002
   0.038217	- Manufacturer: ONEAC
   0.040153	- Product: USB Cable (V2.00)
   0.040454	- Serial Number: unknown
   0.041401	- Bus: 001
   0.041764	Trying to match device
   0.041983	Device matches
   0.044790	send_to_all: SETINFO ups.vendorid "06da"
   0.045094	send_to_all: SETINFO ups.productid "0002"
   0.045383	send_to_all: SETINFO device.type "ups"
   0.045620	send_to_all: SETINFO driver.version "2.7.2"
   0.046939	send_to_all: SETINFO driver.version.internal "0.11"
   0.047947	send_to_all: SETINFO driver.name "blazer_usb"
   0.048207	Trying megatec protocol...
   0.050748	send: Q1
   0.052820	read: timeout
   0.053082	blazer_status: short reply
   0.053320	Status read 1 failed
   5.050832	send: error sending control message: Connection timed out
   5.051066	blazer_status: short reply
   5.051149	Status read 2 failed
   5.052432	send: error sending control message: Broken pipe
   5.621115	Device reset handled
   5.623208	blazer_status: short reply
   5.625037	Status read 3 failed
   5.626716	Trying mustek protocol...
   5.630041	Checking device (06DA/0002) (001/007)
   5.647153	- VendorID: 06da
   5.648952	- ProductID: 0002
   5.650195	- Manufacturer: ONEAC
   5.651498	- Product: USB Cable (V2.00)
   5.652580	- Serial Number: unknown
   5.652811	- Bus: 001
   5.653474	Trying to match device
   5.653736	Device matches
   5.658021	send: QS
   5.660550	read: timeout
   5.660808	blazer_status: short reply
   5.661579	Status read 1 failed
  10.658821	send: error sending control message: Connection timed out
  10.659109	blazer_status: short reply
  10.659213	Status read 2 failed
  10.660422	send: error sending control message: Broken pipe
  11.225352	Device reset handled
  11.227423	blazer_status: short reply
  11.229197	Status read 3 failed
  11.230929	Trying megatec/old protocol...
  11.235007	Checking device (06DA/0002) (001/007)
  11.251880	- VendorID: 06da
  11.253476	- ProductID: 0002
  11.254665	- Manufacturer: ONEAC
  11.255825	- Product: USB Cable (V2.00)
  11.256554	- Serial Number: unknown
  11.257300	- Bus: 001
  11.257503	Trying to match device
  11.258337	Device matches
  11.262344	send: D
  11.265266	read: timeout
  11.265532	blazer_status: short reply
  11.265775	Status read 1 failed
  16.266901	send: error sending control message: Connection timed out
  16.267093	blazer_status: short reply
  16.267184	Status read 2 failed
  16.268434	send: error sending control message: Broken pipe
  16.833319	Device reset handled
  16.836169	blazer_status: short reply
  16.837927	Status read 3 failed
  16.839035	Trying zinto protocol...
  16.841989	Checking device (06DA/0002) (001/007)
  16.858991	- VendorID: 06da
  16.860718	- ProductID: 0002
  16.862934	- Manufacturer: ONEAC
  16.864084	- Product: USB Cable (V2.00)
  16.864740	- Serial Number: unknown
  16.865077	- Bus: 001
  16.865672	Trying to match device
  16.866076	Device matches
  21.866843	send: error sending control message: Connection timed out
  21.867063	blazer_status: short reply
  21.867205	Status read 1 failed
  21.868585	send: error sending control message: Broken pipe
  22.433029	Device reset handled
  22.433311	blazer_status: short reply
  22.433405	Status read 2 failed
  22.437821	Checking device (06DA/0002) (001/007)
  22.454765	- VendorID: 06da
  22.456490	- ProductID: 0002
  22.458403	- Manufacturer: ONEAC
  22.459816	- Product: USB Cable (V2.00)
  22.460027	- Serial Number: unknown
  22.460264	- Bus: 001
  22.461352	Trying to match device
  22.461706	Device matches
  22.466818	send: Q1
  22.468757	read: timeout
  22.469036	blazer_status: short reply
  22.469330	Status read 3 failed
  22.469535	No supported UPS detected

Network UPS Tools - Generic Q* USB/Serial driver 0.06 (2.7.2)
USB communication driver 0.32
   0.000000	send_to_all: SETINFO driver.parameter.port "/dev/bus/usb/001/009"
   0.002482	debug level is '10'
   0.013587	upsdrv_initups...
   0.016115	tcgetattr(/dev/bus/usb/001/009): Inappropriate ioctl for device

SilentT-FR, 2016-05-30:
i change the port to auto, and with nutdrv_qx : 
root@pi:/lib/nut# ./nutdrv_qx -DDDDDDDDD -a test
Network UPS Tools - Generic Q* USB/Serial driver 0.06 (2.7.2)
USB communication driver 0.32
   0.000000	send_to_all: SETINFO driver.parameter.port "auto"
   0.001996	debug level is '9'
   0.011992	upsdrv_initups...
   0.016750	Checking device (06DA/0002) (001/014)
   0.033381	- VendorID: 06da
   0.035215	- ProductID: 0002
   0.037178	- Manufacturer: ONEAC
   0.038651	- Product: USB Cable (V2.00)
   0.038989	- Serial Number: unknown
   0.040233	- Bus: 001
   0.040807	Trying to match device
   0.041008	Device matches
   0.043682	send_to_all: SETINFO ups.vendorid "06da"
   0.044696	send_to_all: SETINFO ups.productid "0002"
   5.043174	send: error sending control message: Connection timed out
   5.043437	qx_process_answer: short reply (input.voltage)
   5.044584	send: error sending control message: Broken pipe
   5.605354	Device reset handled
   5.607495	qx_process_answer: short reply (input.voltage)
   5.612401	Checking device (06DA/0002) (001/014)
   5.628866	- VendorID: 06da
   5.630441	- ProductID: 0002
   5.632449	- Manufacturer: ONEAC
   5.633894	- Product: USB Cable (V2.00)
   5.635108	- Serial Number: unknown
   5.636295	- Bus: 001
   5.637910	Trying to match device
   5.638134	Device matches
   5.640751	send: error sending control message: Broken pipe
   6.200323	Device reset handled
   6.201954	qx_process_answer: short reply (input.voltage)
   6.206580	Checking device (06DA/0002) (001/014)
   6.223369	- VendorID: 06da
   6.224887	- ProductID: 0002
   6.226602	- Manufacturer: ONEAC
   6.228467	- Product: USB Cable (V2.00)
   6.229900	- Serial Number: unknown
   6.230772	- Bus: 001
   6.231146	Trying to match device
   6.231346	Device matches
   6.235426	send: M
   6.238326	read: timeout
   6.238802	qx_process_answer: short reply (ups.firmware.aux)
  11.239164	send: error sending control message: Connection timed out
  11.239400	qx_process_answer: short reply (ups.firmware.aux)
  11.240487	send: error sending control message: Broken pipe
  11.801292	Device reset handled
  11.803045	qx_process_answer: short reply (ups.firmware.aux)
  11.807826	Checking device (06DA/0002) (001/014)
  11.824431	- VendorID: 06da
  11.825991	- ProductID: 0002
  11.828272	- Manufacturer: ONEAC
  11.830041	- Product: USB Cable (V2.00)
  11.831459	- Serial Number: unknown
  11.832647	- Bus: 001
  11.832858	Trying to match device
  11.833059	Device matches
  16.835125	send: error sending control message: Connection timed out
  16.835396	qx_process_answer: short reply (input.voltage)
  16.836575	send: error sending control message: Broken pipe
  17.397275	Device reset handled
  17.399011	qx_process_answer: short reply (input.voltage)
  17.403692	Checking device (06DA/0002) (001/014)
  17.420035	- VendorID: 06da
  17.421581	- ProductID: 0002
  17.423412	- Manufacturer: ONEAC
[... 89 lines cut ...]
  36.006537	- Bus: 001
  36.006743	Trying to match device
  36.007062	Device matches
  41.007149	send: error sending control message: Connection timed out
  41.007345	qx_process_answer: short reply (input.voltage)
  41.008190	send: error sending control message: Broken pipe
  41.568441	Device reset handled
  41.570167	qx_process_answer: short reply (input.voltage)
  41.573985	Checking device (06DA/0002) (001/014)
  41.590960	- VendorID: 06da
  41.592506	- ProductID: 0002
  41.593937	- Manufacturer: ONEAC
  41.595142	- Product: USB Cable (V2.00)
  41.595363	- Serial Number: unknown
  41.595545	- Bus: 001
  41.596002	Trying to match device
  41.596246	Device matches
  41.601054	send: Q1
  41.603129	read: timeout
  41.603424	qx_process_answer: short reply (input.voltage)
  41.604392	send: error sending control message: Broken pipe
  42.164112	Device reset handled
  42.165762	qx_process_answer: short reply (input.voltage)
  42.168007	Device not supported!
  42.169952	Device not supported!

SilentT-FR, 2016-05-30:
in fact, its work with serial cable but, i have 1 usb/serial converter and 2 straight cable and a  cross adaptor. it ugly, i can replace this by one and unique cable but i dont have this for the moment so i have test with usb to be more simple but it not ;)

SilentT-FR, 2016-06-21:
I recycle my two old chloride against Liebert PSP. I close

closed 2016-06-21 by SilentT-FR