networkupstools/nut #293: Improve upssched example in documentation

clepple, 2016-06-05:
Per a [conversation with Roger Price](http://news.gmane.org/find-root.php?message_id=alpine.LSU.2.11.1606051224460.3552%40maria), the example at http://networkupstools.org/docs/user-manual.chunked/ar01s07.html chapter 7.2 does not clearly distinguish between timers, and the file created as part of the example timer scenario.

clepple, 2016-06-05:
Related: transient event filtering mentioned here: https://github.com/networkupstools/nut/issues/219

aquette, 2016-08-12:
@clepple can you please take care of that one? thx and cheers

clepple, 2016-08-12:
@aquette I did not assign myself to this issue because I don't use upssched. I will see if I have time to go through Roger's documentation after I set up an environment to test the libusb-1.0 branch.

Luo2Jie2, 2017-04-04:
This ticket is due to my misunderstanding of the upssched timers.  I wrongly assumed that they were files.  I now understand that they are in-memory, and that the 2.7.5 documentation will explain this.

This SIGUSR1 and SIGUSR2 support patches which were a byproduct of the mailing list discussions do not address specific UPS's, and for this reason I withdraw the patches.  A general solution to the problem of stopping a timer from a Bash script could be achieved with SIGQUEUE, but since there appears to be no demand for such a feature, I will continue with my private solution.   

Roger

aquette, 2017-11-27:
@Luo2Jie2 and @clepple : is this ticket still worth?
I already made some clarification in 7a7f941 that should be sufficient...

jimklimov, 2023-01-06:
To me, the goal seems achieved, and no new complaints logged :)

closed 2023-01-06 by jimklimov