networkupstools/nut #296: [HCL] Liebert PSP supported by usbhid-ups

SilentT-FR, 2016-06-21:
root@pi:/lib/nut# upsc LiebertBox@maison
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 38
battery.charge.warning: 38
battery.type: PbAc
battery.voltage: 13.7
battery.voltage.nominal: 12.0
device.mfr: Emerson Network Power
device.model: LiebertPSP
device.serial:
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/bus/usb/001/005
driver.version: 2.7.2
driver.version.data: Belkin HID 0.16
driver.version.internal: 0.38
input.frequency: 49.9
input.voltage: 235.6
output.voltage: 234.4
ups.load: 1
ups.mfr: Emerson Network Power
ups.model: LiebertPSP
ups.productid: 0001
ups.serial:
ups.status: OL CHRG
ups.vendorid: 10af

http://www.emersonnetworkpower.com/en-US/Products/ACPower/DesktopWorkstationUPS/Pages/liebertpspstandbyups350650va.aspx

clepple, 2016-08-25:
Does shutdown work properly?

clepple, 2017-05-20:
@SilentT-FR assuming that shutdown works; please open another issue if not.

closed 2017-05-20 by clepple