networkupstools/nut #302: Make 64 bit version of the Windows port available

lf-, 2016-08-09:
The current situation of providing a 32 bit Windows version makes excellent sense, as it has compatibility with both 32 and 64 bit Windows. The reason I'm raising this issue is because Nano Server has no WoW64, so the 32 bit build doesn't run at all.

It would be excellent if a 64 bit distribution of NUT for Windows was provided (zip would be nicer too, less extraction due to Nano Server having no msi support).

clepple, 2016-08-11:
planned, see https://github.com/networkupstools/nut/issues/5

gdt, 2023-01-07:
Is this still a valid issue, or is nut-for-windows built for x86_64?

jimklimov, 2023-01-07:
Good question... on Appveyor CI - I think x86_64 in MSYS2 MinGW.

With linux+mingw cross-builds both are possible, some caveats (for prereq builds) were documented in scripts/Windows/README IIRC.

Either way, it is an effort that needs completion (packaging, esp. of privileged libusb device access) to hit prime-time.