networkupstools/nut #306: UPS information server should start if an UPS is unreachable

sebelk, 2016-08-16:
Hi,

I've found that upsd (2.7.2) does not start if an UPS is unrecheable:

I run upsd manually, so I think it's not a problem of systemd:
```

210.921843 Signal 2: exiting
 210.921902 Disconnect from 127.0.0.1
 210.921972 Disconnect from 127.0.0.1
 210.922022 Disconnect from 127.0.0.1
 210.922070 Disconnect from 127.0.0.1
 210.922117 Disconnect from 127.0.0.1
 210.922164 Disconnect from 127.0.0.1
 210.922212 Disconnect from 127.0.0.1
 210.922259 Disconnect from 127.0.0.1
 210.922305 Disconnect from 127.0.0.1
 210.922352 Disconnect from 127.0.0.1
Network UPS Tools upsd 2.7.2
```
The complete debug is at http://pastebin.com/QDNfWWin

Could be it a bug?

Thanks in advance

clepple, 2016-08-16:
Signal 2 usually corresponds to Ctrl-C, so it looks like `upsd` ran for over three minutes before it was killed.

What behavior are you expecting?

When you mention systemd, does upsd behave differently than when running from the command line? If so, we would need to know more about the system (distribution, version, etc.) to comment effectively.

aquette, 2016-08-17:
I've been doing some improvements around systemd and startup, especially WRT to the historic (sysV) behavior. There is a fix that shipped with NUT 2.7.4 that should be of interest for you @sebelk: https://github.com/networkupstools/nut/issues/208

jimklimov, 2022-08-19:
Current release 2.8.0 features better systemd integration (instance per driver, not one for all drivers) and ability to start upsd without drivers defined with #766 

upsd per se should be startable if driver does not respond, just has no readings to report; `check-NIT` verifies that now. (another matter is if systemd units would enable without drivers - a distro matter)