networkupstools/nut #307: USB driver Eurocase EA200LED 1200VA

jcanderle, 2016-08-17:
Can you point me where to start to write driver/sub_driver for the Eurocase EA200LED 1200VA?
I have collected USB communication (Send/Receive on beginning of line are not part of comm) :

S,80 06 03 03 09 04 00 66 
R,(003.0 000.0 230.0 032 00.0 24.4 29.0 10001000
S,80 06 0c 03 09 04 00 66 
R,#                           V2.8      
S,80 06 0d 03 09 04 00 66 
R,#   .        .     . 
S,80 06 04 03 09 04 00 66 
R,UPS No Ack
S,80 06 06 03 09 04 00 74 
R,UPS No Ack
S,80 06 05 03 09 04 00 66 
R,UPS No Ack
S,80 06 07 03 09 04 00 66 
R,UPS No Ack
S,80 06 03 03 09 04 00 66 
R,*003.0 000.0 230.0 026 00.0 25.0 29.0 10001000
S,80 06 06 03 09 04 00 65 
R,UPS No Ack
S,80 06 03 03 09 04 00 66 
R,(237.0 000.0 237.0 024 50.1 26.0 29.0 00001000
S,80 06 03 03 09 04 00 66 
R,(237.0 000.0 237.0 024 49.8 26.8 29.0 00001000

I am trying to figure out the meaning of commands and returned results meanwhile.
The USB USP device report itself as:
ID 0001:0000 Fry's Electronics 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                1 MEC0003
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           41
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     624
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
Device Status:     0x0000

for which I have created udev rule to create symlink /dev/usbhid-ups to the usb device:
ATTR{idVendor}=="0001", ATTR{idProduct}=="0000", ATTRS{product}=="MEC0003", MODE="0774", GROUP="nut", SYMLINK+="usbhid-ups"

I have tried other available drivers, but  it seems that this device is not using any supported protocol.

zykh, 2016-08-17:
This sounds like something that should be supported by nutdrv_qx with 'fabula' USB subdriver (set protocol to 'megatec', if you want to skip the initial match procedure, which otherwise should give you some failed queries before the actual start of communication with the device), or, in case you experience any problem with it, you may want to try the 'krauler' one (which doesn't support shutdowns).

In case of success, please test all the available instant commands (or, at least, the shutdowns and the shutdown procedure) and report back the output of upsc / upsrw / upscmd for our [DDL](http://networkupstools.org/ddl/).

jcanderle, 2016-08-30:
I have tried nutdrv_qx with none or "fabula" or "krauler" subdrivers specified. To make it working, I had to specify the port to /dev/usbhid-ups (see SYMLINK above). Than all settings always worked (kind of - communication starts, some info is pulled from UPS), but only for the first packets, all consequent readings show same values, so it looks, like some buffer is not flushed. Even when I pull the plug from wall socket, it reads online (not on battery).
I have attached the wireshark logs: one is using original app and one is using nut driver:
[USB-UPS-nut.txt](https://github.com/networkupstools/nut/files/444305/USB-UPS-nut.txt)
[USB-UPS-orig.txt](https://github.com/networkupstools/nut/files/444306/USB-UPS-orig.txt)

alexz-kh, 2016-11-17:
Hm, looks like i have something-around issue...let me paste it there, and do not create another one issue :)

A have device:
```
$ lsusb
Bus 001 Device 003: ID 0001:0000 Fry's Electronics 
$ usb-devices
T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=01 Dev#=  3 Spd=1.5 MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=0001 ProdID=0000 Rev=00.00
S:  Manufacturer=ATCL FOR UPS
S:  Product=ATCL FOR UPS
S:  SerialNumber=ATCL FOR UPS
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=98mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=(none)
```

First, i had 'device not found\not detected' and etc issues from 'blazer_usb'

/lib/nut/blazer_usb -DD -a frys
```
Network UPS Tools - Megatec/Q1 protocol USB driver 0.11 (2.7.2)
   0.000000     debug level is '2'
   0.017068     Checking device (1D6B/0003) (002/001)
   0.028542     - VendorID: 1d6b
   0.028572     - ProductID: 0003
   0.028668     - Manufacturer: unknown
   0.028676     - Product: unknown
   0.028682     - Serial Number: unknown
   0.028688     - Bus: 002
   0.028694     Trying to match device
   0.028702     Device does not match - skipping
   0.028841     Checking device (0001/0000) (001/003)
   0.033773     - VendorID: 0001
   0.033816     - ProductID: 0000
   0.033830     - Manufacturer: ATCL FOR UPS
   0.033841     - Product: ATCL FOR UPS
   0.033853     - Serial Number: ATCL FOR UPS
   0.033864     - Bus: 001
   0.033874     Trying to match device
   0.033888     Device matches
   0.035544     Trying megatec protocol...
   0.036992     Stall condition cleared
   0.037030     blazer_status: short reply
   0.037039     Status read 1 failed
   0.038110     Stall condition cleared
   0.038124     blazer_status: short reply
   0.038132     Status read 2 failed
   0.038973     Stall condition cleared
   0.038986     blazer_status: short reply
   0.038993     Status read 3 failed
   0.038999     Trying mustek protocol...
   0.039012     blazer_status: short reply
   0.039019     Status read 1 failed
   0.039026     blazer_status: short reply
   0.039032     Status read 2 failed
   0.039038     blazer_status: short reply
   0.039044     Status read 3 failed
   0.039049     Trying megatec/old protocol...
   0.039056     blazer_status: short reply
   0.039062     Status read 1 failed
   0.039214     blazer_status: short reply
   0.039221     Status read 2 failed
   0.039228     blazer_status: short reply
   0.039234     Status read 3 failed
   0.039239     Trying zinto protocol...
   0.040200     Stall condition cleared
   0.040213     blazer_status: short reply
   0.040220     Status read 1 failed
   0.041192     Stall condition cleared
[... 107 lines cut ...]
   0.035935     blazer_status: short reply
   0.035943     Status read 3 failed
   0.035950     Trying megatec/old protocol...
   0.035957     blazer_status: short reply
   0.036005     Status read 1 failed
   0.036047     blazer_status: short reply
   0.036087     Status read 2 failed
   0.036294     blazer_status: short reply
   0.036304     Status read 3 failed
   0.036311     Trying zinto protocol...
   0.037300     Stall condition cleared
   0.037316     blazer_status: short reply
   0.037325     Status read 1 failed
   0.038149     Stall condition cleared
   0.038162     blazer_status: short reply
   0.038169     Status read 2 failed
   0.039036     Stall condition cleared
   0.039050     blazer_status: short reply
   0.039057     Status read 3 failed
   0.039063     No supported UPS detected
```

http://lists.alioth.debian.org/pipermail/nut-upsuser/2014-November/009348.html (chmod \ -u root fixes also tryed)

Could you suggest me something else to try ?

vtacquet, 2020-04-22:
> I have tried blazer_usb and nutdrv_qx with none or "fabula" or "krauler" subdrivers specified. To make it working, I had to specify the port to /dev/usbhid-ups (see SYMLINK in udev conf file above, auto-detection always failed). Then all drivers settings always worked (kind of - communication starts, some info is pulled from UPS), but only for the first packets, all consequent readings show same values, so it looks, like some buffer is not flushed. Even when I pull the plug from wall socket, it reads online (not on battery).
> I have attached the wireshark logs: one is using original app and one is using nut driver:
> [USB-UPS-nut.txt](https://github.com/networkupstools/nut/files/444305/USB-UPS-nut.txt)
> [USB-UPS-orig.txt](https://github.com/networkupstools/nut/files/444306/USB-UPS-orig.txt)
> where you can see, the request sequencing and "URB_INTERRUPT in" after each "I", "Q1" sequence. It looks, like the "URB_INTERRUPT in" is getting values refreshed into UPS USB buffer.

Hey, did you ever find a solution for this ? After 1 day of troubleshooting and finding the same interrupt crap for the graphical tool (UPsmart) in my tcpdump files I stumbled on this thread. I have an Energenie UPS - in USB name that would be Fry's Electronics. Tried all possible combinations with blazer_ups, nutdrv_qx ...

CyborTronik, 2020-10-14:
A setup like this worked for me:

```
[LogicPowerUPS]
        driver = nutdrv_qx
        subdriver = fabula
        productid = 0000
        vendorid = 0001
        port = auto
```
Having a Logic Power UPS 2000 which gets identified as same Fry's Electronics thing

gukandrew, 2020-12-21:
@CyborTronik thank you! **nutdrv_qx** works perfectly for my Logic Power UPS (LK-800 USB?).

I'm using Home Assistant with Network UPS Tools Home Assistant Community Add-on.
So if somebody find this page and struggling with similar model here is my configuration for that addon (Home Assistant -> Supervisor -> Dashboard Tab -> Network UPS Tools -> Configuration Tab):
```
users:
  - username: 'blahuser'
    password: 'blahpass'
    instcmds:
      - all
    actions: []
devices:
  - name: myups
    driver: nutdrv_qx
    subdriver: fabula
    port: auto
    config:
      - vendorid = 0001
      - productid = 0000
mode: netserver
shutdown_host: 'false'
list_usb_devices: 'true'
```

Also there should be nut sensor in general HA config:
```
  # UPS
  - platform: nut
    name: "Logic Power UPS 800"
    host: a0d7b954-nut
    username: nutty
    password: "!SDF-ee11eq"
    resources:
      - battery.charge
      - battery.voltage
      - battery.voltage.high
      - battery.voltage.low
      - ups.load
      - ups.temperature
      - ups.status.display
      - input.frequency
      - input.voltage
      - output.voltage
```

pejotes, 2021-09-19:
> @CyborTronik thank you! **nutdrv_qx** works perfectly for my Logic Power UPS (LK-800 USB?).
> 
> I'm using Home Assistant with Network UPS Tools Home Assistant Community Add-on.
> So if somebody find this page and struggling with similar model here is my configuration for that addon (Home Assistant -> Supervisor -> Dashboard Tab -> Network UPS Tools -> Configuration Tab):
> 
> ```
> users:
>   - username: 'blahuser'
>     password: 'blahpass'
>     instcmds:
>       - all
>     actions: []
> devices:
>   - name: myups
>     driver: nutdrv_qx
>     subdriver: fabula
>     port: auto
>     config:
>       - vendorid = 0001
>       - productid = 0000
> mode: netserver
> shutdown_host: 'false'
> list_usb_devices: 'true'
> ```
> 
> Also there should be nut sensor in general HA config:
> 
> ```
>   # UPS
>   - platform: nut
>     name: "Logic Power UPS 800"
>     host: a0d7b954-nut
>     username: "blahuser"
>     password: "blahpass"
>     resources:
>       - battery.charge
>       - battery.voltage
>       - battery.voltage.high
>       - battery.voltage.low
>       - ups.load
>       - ups.temperature
>       - ups.status.display
>       - input.frequency
>       - input.voltage
>       - output.voltage
> ```

This worked like a charm!!!

dinojascur, 2022-09-22:
> I have tried blazer_usb and nutdrv_qx with none or "fabula" or "krauler" subdrivers specified. To make it working, I had to specify the port to /dev/usbhid-ups (see SYMLINK in udev conf file above, [...]

Hey @jcanderle any further conclusion? I hit the same issue...