networkupstools/nut #310: APC Back-UPS RS 550G output

gkl1368, 2016-08-19:
the APC Back-UPS RS 550G in nut-cgi shows that output not supported. How can i solve this problem.
[image]

clepple, 2016-08-19:
If you run `upsc apc-ups`, what does it return?

gkl1368, 2016-08-19:
root@NUT-Server:~# upsc apc-ups
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.date: 2001/09/25
battery.mfr.date: 2016/04/06
battery.runtime: 1348
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 13.8
battery.voltage.nominal: 12.0
device.mfr: American Power Conversion
device.model: Back-UPS RS 550G
device.serial: * 
device.type: ups
driver.flag.Maxreport: enabled
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
input.sensitivity: medium
input.transfer.high: 282
input.transfer.low: 176
input.transfer.reason: input voltage out of range
input.voltage: 217.0
input.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: 857.L5 .I
ups.firmware.aux: L5  
ups.load: 23
ups.mfr: American Power Conversion
ups.mfr.date: 2016/04/06
ups.model: Back-UPS RS 550G
ups.productid: 0002
ups.realpower.nominal: 330
ups.serial: * 
ups.status: OL
ups.test.result: No test initiated
ups.timer.reboot: 0
ups.timer.shutdown: -1
ups.vendorid: 051d

clepple, 2016-08-20:
There are lower-level debug dumps we can try, but given the recent manufacturing date, it is possible that this UPS only supports some variables via MODBUS (which NUT does not currently implement; see #139). 

Output is similar to this other report:
http://networkupstools.org/ddl/APC/Back-UPS_RS_550GI.html

j-ed, 2016-10-01:
It might be possible to use code snippets from the apcupsd project because they seem to have already implemented MODBUS support with the support of APC:
https://sourceforge.net/p/apcupsd/mailman/message/33333592/
https://sourceforge.net/p/apcupsd/svn/HEAD/tree/branches/Branch-3_14/src/drivers/modbus/

closed 2023-02-08 by gkl1368