networkupstools/nut #311: Test documentation tools such as dblatex in configure script

clepple, 2016-08-25:
Currently we check for tools such as `asciidoc` and `dblatex`, but we do not verify that they produce output.

Reference: https://github.com/networkupstools/nut/pull/305#issuecomment-242352514

Vaguely related: #162 (indicate required versions)

jimklimov, 2016-08-25:
Example failed build log: http://buildbot.networkupstools.org/public/nut/builders/FreeBSD-x86/builds/529/steps/compile/logs/stdio

````
...
asciidoc: include: /usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/docs/history.txt
asciidoc: user-manual.txt: line 187: reading: /usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/docs/history.txt

a2x: executing: "xmllint" --nonet --noout --valid "/usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs/user-manual.xml"

a2x: executing: "dblatex" -t pdf -p "/usr/local/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/usr/local/etc/asciidoc/dblatex/asciidoc-dblatex.sty"  -V  "/usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs/user-manual.xml"

Build the book set list...
xsltproc -o /tmp/tmpTEw1YW/doclist.txt --xinclude --xincludestyle doclist.xsl /usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs/user-manual.xml
Build the listings...
xsltproc -o /tmp/tmpTEw1YW/listings.xml --xinclude --xincludestyle --param current.dir '/usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs' /usr/local/share/dblatex/xsl/common/mklistings.xsl /usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs/user-manual.xml
xsltproc -o user-manual.rtex --xinclude --xincludestyle --catalogs --param current.dir '/usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs' --param listings.xml '/tmp/tmpTEw1YW/listings.xml' /tmp/tmpTEw1YW/custom.xsl /usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs/user-manual.xml
XSLT stylesheets DocBook - LaTeX 2e (0.3.7)
===================================================
Image 'dblatex' not found
Build user-manual.pdf
built-in module pdftex registered
no support found for ifxetex
no supporta2x: ERROR: "dblatex" -t pdf -p "/usr/local/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/usr/local/etc/asciidoc/dblatex/asciidoc-dblatex.sty"  -V  "/usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs/user-manual.xml" returned non-zero exit status 1
 found for fontspec
no support found for xltxtra
no support found for fontenc
no support found for inputenc
no support found for fancybox
built-in module makeidx registered
no support found for asciidoc-dblatex
building additional files...
checking if compiling is necessary...
the output file doesn't exist
pdflatex -interaction=batchmode user-manual.tex
pdflatex failed
Unexpected error occured
Error: pdflatex compilation failed

Makefile:824: recipe for target 'user-manual.pdf' failed
make[3]: *** [user-manual.pdf] Error 1
make[3]: Leaving directory '/usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub/docs'
Makefile:524: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/usr/home/_buildbot/nut-slave/FreeBSD-x86/build/nut-v2.7.4-18-ga12f3a8/_build/sub'
Makefile:730: recipe for target 'distcheck' failed
make[1]: *** [distcheck] Error 1
make[1]: Leaving directory '/usr/home/_buildbot/nut-slave/FreeBSD-x86/build'
Makefile:927: recipe for target 'distcheck-light' failed
make: *** [distcheck-light] Error 2
program finished with exit code 2
elapsedTime=4548.513437
````

jimklimov, 2016-09-14:
Candidate solution posted in PR #315 .

clepple, 2017-09-08:
Just noticed that this was still open - I am closing it on the assumption that PR #315 was sufficient.

closed 2017-09-08 by clepple