networkupstools/nut #312: Can not compile nut-scanner missing reference to str_rtrim

theroch, 2016-08-26:
It is not possible to compile the current master.
Builds of nut fail with:
```bash
make[3]: Entering directory './nut.src/nut/tools/nut-scanner'
/bin/bash ../../libtool --tag=CC   --mode=link gcc -I../../clients -I../../include -g -O2 -Wall -Wsign-compare   -o nut-scanner nut_scanner-nut-scanner.o libnutscan.la ../../common/libcommon.la -lpthread
libtool: link: gcc -I../../clients -I../../include -g -O2 -Wall -Wsign-compare -o .libs/nut-scanner nut_scanner-nut-scanner.o  ./.libs/libnutscan.so ../../common/.libs/libcommon.a -lpthread -Wl,-rpath -Wl,//lib
./.libs/libnutscan.so: undefined reference to `str_rtrim'
collect2: error: ld returned 1 exit status
Makefile:524: recipe for target 'nut-scanner' failed
make[3]: *** [nut-scanner] Error 1
make[3]: Leaving directory './nut.src/nut/tools/nut-scanner'
Makefile:390: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory './nut.src/nut/tools/nut-scanner'
Makefile:384: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory './nut.src/nut/tools'
Makefile:448: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1
```
Maybe it is the same problem like this:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=831740

aquette, 2016-09-02:
hi @theroch , this is strange and was supposed to be fixed with https://github.com/networkupstools/nut/commit/5d98d5536699222bc93c58c1e6f6df43dc9bfcff done last March
I'm also running the latest Jessie, and can't reproduce the issue.

Have you rerun autogen.sh and configure on your latest master, before doing a make?

theroch, 2016-09-02:
Thx @aquette I missed the autogen.sh part.

With the following steps, it compiles now correct.
```
./autogen.sh
./configure ...
make
```

closed 2016-09-02 by theroch

aquette, 2016-09-02:
hi @theroch , thanks for the confirmation, I was a bit puzzled ;)