networkupstools/nut #316: Ensure that all drivers print version information before they go into the background

clepple, 2016-09-15:
In the "Starting the driver(s)" section in the user manual, the example says `Detected EATON - Ellipse MAX 1100 [...]`.

Not all drivers do this. For instance, `usbhid-ups` uses upsdebugx(1, ...) which won't print to the console unless a user has debugging turned on.

After fixing the drivers, update the documentation in case the format has changed.

jimklimov, 2023-01-06:
At least currently `usbhid-ups` does report the following details:
````
# ./drivers/usbhid-ups -a eco650
Network UPS Tools - Generic HID driver 0.49 (2.8.0-Windows-316-g0a5cfdfe6)
USB communication driver (libusb 1.0) 0.43
Using subdriver: MGE HID 1.46
````

But indeed gotta check that everybody does.

gdt, 2023-01-09:
I think it's a bug for programs to print version if not asked to (via some -v -or -D).   Most daemons do not do this.

jimklimov, 2024-04-10:
See also PR #2386 which should ensure that specifically NUT version is printed early.

jimklimov, 2024-09-12:
With the PRs listed above, driver banner is always printed unless disabled explicitly, early before parsing the bulk of command-line arguments (or in help/verbose handling, if it was disabled). Stated scope goals seem achieved.

closed 2024-09-12 by jimklimov