networkupstools/nut #322: apcsmart driver option(s?) ignored

dark-penguin, 2016-09-16:
Some APC UPS-es ignore the "Kill power now!" command if power is already back at the moment.

According to man ups.conf and man apcsmart.conf, it's possible to specify "sdtype = 4" option in ups.conf to use an "CS hack" to work around this. (There is even an example in man ups.conf with this very option!)

Unfortunately, it does not work. Even with "sdtype = 4" specified, issuing "/sbin/upsdrvctl shutdown" (which is called after the shutdown procedure to kill the power) does NOT use the hack (which would be to start a simulated power failure to switch to battery power before issuing the forced shutdown command). It just sends a normal shutdown command, which is ignored.

However, issuing "upscmd -u myuser -p mypwd apc shutdown.return cs" does work as described in the manual (namely, you must issue this command TWICE, the second time being 3~15 sec after the first one.)

clepple, 2016-09-21:
@msoltyspl, any thoughts?

msoltyspl, 2016-09-21:
Dunno yet, I'll doublecheck this later today and see how it looks in the code.

msoltyspl, 2016-09-29:
Ok, sorry for delay.

It seems that sdcmd_CS() doesn't do any wait in between commands (but still works fine if it's called twice manually).

We can add say arbitrary 3.5 seconds to wait after the first (force ob) command is being issued. This should work finein that case.

msoltyspl, 2016-11-17:
I've pushed the changes to master, please verify when you have the chance if it's working correctly for you.

dark-penguin, 2016-11-17:
Thank you!
I'm not sure when I'll be around that server, but I'll try it someday.
For the record, which version has the fix? I'd be interested in tracking when it makes its way into Debian Sid.

closed 2016-11-17 by dark-penguin

aquette, 2016-11-18:
@dark-penguin : since the changes are in master, the next stable (2.7.5) will have it... when published.