networkupstools/nut #326: Please expose remote UPS data through a win32_battery WMI driver.

mrudat, 2016-09-25:
It would be awesome if a WMI driver could be implemented that would present remote UPS information to windows using the win32_battery class (the way a directly attached UPS is exposed to native windows applications).

This would allow using the native power configuration control panel to configure behaviour when on battery, rather than having to have the monitor daemon configured to run the appropriate programs to achieve the same effect (where this is actually possible).

This would make it easier/possible to:
* reduce maximum clock speed to save power
* reduce sleep time for pc/monitor
* shut down applications (eg. Boinc) that can be configured to not run on battery

fragtion, 2017-01-31:
Agree this is needed. It would either need to be implemented as part of upsmon.conf, or as a separate module

jimklimov, 2017-02-01:
Now I wonder if there's equivalent functionality for Unix/Linux OSes -
exposing an external UPS similarly to laptop battery, and/or exposing a
local battery as a power device for NUT...

clepple, 2017-02-06:
@jimklimov I don't think we have an issue open for this, but UPower is a successor to HAL, and is a possibility for exposing NUT data to the *nix GUI. In the other direction, we have #319.

@mrudat and @fragtion : it's good to know that there is interest in more advanced features, but what we really need is help re-implementing basic Windows support on something more recent than the 2.6.x branch of NUT. If you know anyone with experience in developing open-source software on Windows, issue #5 is our starting point.

6XGate, 2017-03-03:
I don't know about licensing, but I've got a daemon (still rough) that does expose Win32_Battery to NUT clients.  Ya'll can borrow any code from it.

RoganDawes, 2020-09-01:
I'd support the use of a Win32_Battery driver. In the simplest case, everything should just work with it running as a netclient, with a 1-line MONITOR entry a la upsmon. That could be stored in the Windows Registry. If needed, POLLFREQ and similar variables can also be set in the same location. Actions such as shutting down and hibernation, etc become the responsibility of the OS.
To me this seems like the absolute minimum effort required to get a working Windows-based NUT client. And potentially, it could be added to the Windows core, much as APC did with their built-in Windows UPS driver.

RoganDawes, 2020-09-16:
This page seems to be a good starting point for learning how to write a WMI provider: https://docs.microsoft.com/en-us/windows/win32/wmisdk/creating-wmi-providers