networkupstools/nut #327: FTBFS on kfreebsd-*

bigon, 2016-09-28:
Hi,

It seems that nut FTBFS on the kfreebsd ports in debian.

They both have the same error:

```
/bin/bash ../libtool  --tag=CC   --mode=link gcc -I../include  -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -g -O2 -fdebug-prefix-map=/build/net-snmp-jatWcV/net-snmp-5.7.3+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security -isystem /usr/include/bsd -DLIBBSD_OVERLAY -DNETSNMP_USE_INLINE -Ukfreebsd -Dkfreebsd=kfreebsd -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -I/usr/lib/x86_64-kfreebsd-gnu/perl/5.24/CORE -Wdate-time -D_FORTIFY_SOURCE=2 -I/usr/include -I/usr/include/neon   -g -O2 -fdebug-prefix-map=/«PKGBUILDDIR»=. -fPIE -fstack-protector-strong -Wformat -Werror=format-security -Wall -Wsign-compare  -fPIE -pie -Wl,-z,relro -Wl,-z,now -Wl,-z,defs -Wl,-O1 -Wl,--as-needed -o al175 al175.o ../common/libcommon.la ../common/libparseconf.la main.o dstate.o  serial.o -lpthread 
libtool: link: gcc -I../include -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -g -O2 -fdebug-prefix-map=/build/net-snmp-jatWcV/net-snmp-5.7.3+dfsg=. -fstack-protector-strong -Wformat -Werror=format-security -isystem /usr/include/bsd -DLIBBSD_OVERLAY -DNETSNMP_USE_INLINE -Ukfreebsd -Dkfreebsd=kfreebsd -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -I/usr/lib/x86_64-kfreebsd-gnu/perl/5.24/CORE -Wdate-time -D_FORTIFY_SOURCE=2 -I/usr/include -I/usr/include/neon -g -O2 -fdebug-prefix-map=/«PKGBUILDDIR»=. -fPIE -fstack-protector-strong -Wformat -Werror=format-security -Wall -Wsign-compare -fPIE -pie -Wl,-z -Wl,relro -Wl,-z -Wl,now -Wl,-z -Wl,defs -Wl,-O1 -o al175 al175.o main.o dstate.o serial.o  -Wl,--as-needed ../common/.libs/libcommon.a ../common/.libs/libparseconf.a -lpthread
main.o: In function `main':
./drivers/main.c:511: undefined reference to `bsd_getopt'
collect2: error: ld returned 1 exit status
Makefile:1114: recipe for target 'al175' failed
```

https://buildd.debian.org/status/fetch.php?pkg=nut&arch=kfreebsd-amd64&ver=2.7.4-4&stamp=1475016742
https://buildd.debian.org/status/fetch.php?pkg=nut&arch=kfreebsd-i386&ver=2.7.4-4&stamp=1475016813

clepple, 2016-09-28:
NUT is not specifically requesting `bsd_getopt()`, and it builds those drivers correctly on FreeBSD and OpenBSD without any special `./configure` options. Do you have any additional information as to why it is not just using `getopt()`, which should be part of libc?

bigon, 2016-09-28:
The problem seems to be the fact that Net-SNMP CFLAGS are leaking in when building the drivers. Is that necessary?

bigon, 2016-09-28:
Same for libusb, I don't think that these flags should be passed when building serial drivers

bigon, 2016-09-28:
The root cause is the fact that LIBNETSNMP_CFLAGS contains "-isystem /usr/include/bsd" having this also requires to be linked against -lbsd