networkupstools/nut #329: Windows MSI package missing DLLs

pwetzel, 2016-10-01:
Attempting to run the service in netclient mode failed silently. The issue was two missing DLLs that prevented upsmon from starting: `libgcc_s_dw2-1.dll` and `libeay32.dll`. The `libgcc*` one was available in the NUT\bin directory, I just made another copy in NUT\sbin. The other is an openssl library, so I installed the 1.0.2j package from [Shining Light](http://slproweb.com/products/Win32OpenSSL.html). Now it works fine. Perhaps other DLLs are missing for upsd, but I don't have a means to test for it.

The first (`libgcc`) is probably a linker setting error, assuming you actually want static binaries in `**s**bin`. I don't blame you for not wanting to distribute openssl, but a note somewhere on the download page or (better yet) the installer dialog would be helpful.

I've found lots of examples of people running this port successfully, so perhaps something just got messed up in this last package? This was using Windows MSI installer 2.6.5-6 downloaded on 1 October 2016.

clepple, 2016-10-05:
Duplicate of #262 - please reopen if I missed something.

Also, if you know of any developers who could help out with the Windows port, that would certainly speed things along. Not much recent activity on issue #5, either.

closed 2016-10-05 by clepple

pwetzel, 2016-10-09:
You are correct, please forgive the noise. I didn't find that when searching.