networkupstools/nut #333: [HTC] drivers for DirectUPS products

mjsduncan, 2016-10-22:
hello, DirectUPS (www.directups.com) isn't listed as a manufacturer.
using either the blazer_usb and usbhid-ups drivers, upsc gives "Error: Driver not connected".
dmesg lists the unit's usb connection as serial-to-usb:
````
[    2.028015] usb 5-1: new low-speed USB device number 2 using uhci_hcd
[    2.213240] usb 5-1: New USB device found, idVendor=0665, idProduct=5161
[    2.214485] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.215678] usb 5-1: Product: USB to Serial
[    2.216864] usb 5-1: Manufacturer: INNO TECH
[    2.218044] usb 5-1: SerialNumber: 20100826
````
any suggestions for correct driver?

clepple, 2016-10-25:
hi, which version of NUT are you using? nutdrv_qx will match 0665:5161, but it is a fairly recent addition (relative to some distributions).

clepple, 2016-10-25:
Which version of NUT are you using? nutdrv_qx will match 0665:5161, but that does seem to be a generic USB-to-serial device.

mjsduncan, 2016-10-26:
thanks for the quick response!  i'm using nut 2.7.1 on current ubuntu 14.04.  in ups.conf i tried without and then with all the protocols:
````
 [directups1000]
       driver = nutdrv_qx
         port = auto
     protocol =   [mecer, megatec, megatec/old, mustek, voltronic, zinto]
````
and each time got 
````
# upsdrvctl start
Network UPS Tools - UPS driver controller 2.7.1
Network UPS Tools - Generic Q* USB/Serial driver 0.01 (2.7.1)
USB communication driver 0.32
Device not supported!
Device not supported!
Driver failed to start (exit status=1)
````
any other suggestions?

thx again

clepple, 2016-10-26:
For Ubuntu (and other Debian derivatives), you can launch the driver directly with an extra "-D" flag, e.g.:

   sudo /lib/nut/nutdrv_qx -D -a directups1000 

Note that when starting the driver this way, you can take out the "protocol=" line from ups.conf and try the various values with "-x protocol=megatec" on the command line (if that is easier).

There is also a version in this PPA which has a "q1" driver that might work better:

   https://code.launchpad.net/~dobey/+recipe/nut-daily

(Not sure it's rebuilt daily anymore, but it should be newer than NUT 2.7.1)

mjsduncan, 2016-10-26:
after installing nut v2.7.3.1+r260, nutdrv_qx still didn't work (there wasn't  a nutdrv_q1) but blazer_usb now works!

thanks.

zykh, 2016-10-26:
mmh.. nutdrv_qx is set to replace blazer_* drivers in the long run, so we better support this device. Since 2.7.3 is not exactly new, it could be an already-solved bug, but maybe it's something new...
@mjsduncan, can you please post the output of nutdrv_qx?

clepple, 2016-10-27:
@mjsduncan My mistake, I meant to write that there is a "q1" **sub**driver.

Also, can you double-check the version number? It should end in a number like 2604 (the latest in that PPA).

@zykh The version numbering from that PPA is a little strange, but the "2.7.3.1" was from `configure.ac` at the time of import, and then the numbers like `r2604` are bzr revisions from the git import. I think that revision corresponds to [git v2.7.4-35-g010bfe0](http://bazaar.launchpad.net/~vcs-imports/nut/master/revision/2604).

mjsduncan, 2016-10-27:
the package version is 2.7.3.1+r2601.1~ubuntu14.04.1, the previously posted version is a typo.

and nutdrv_qx works now, not sure why it didn't work before:
````
# /lib/nut/nutdrv_qx -D -a directups1000
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4.1)
USB communication driver 0.33
   0.000000	debug level is '1'
   0.000552	upsdrv_initups...
   8.239700	Using protocol: Q1 0.07
   8.239748	upsdrv_initinfo...
   8.495839	No charge time specified, using built in default [43200 seconds]
   8.495874	No idle load specified, using built in default [10.0 %]
   8.495898	upsdrv_updateinfo...
   8.495918	Quick update...
   8.751752	upsdrv_updateinfo...
   8.751786	Quick update...
 ...
````

clepple, 2016-10-30:
@mjsduncan glad to hear it is working. Since several Linux distributions are phasing out libusb-0.1 in favor of libusb-1.0, I updated the PPA to build against the newer version. I have only tested the 16.04/xenial build, but there were very few changes to the distribution-specific portions, so I have high confidence that the 14.04/trusty build will "just work".

Also, if you don't mind sending some of the `upsc` output, we'd like to update the HCL/DDL info: http://networkupstools.org/stable-hcl.html#footnotes

Any information on shutdown testing would be helpful as well.

mjsduncan, 2016-11-02:
````
# upsc directups1000
Init SSL without certificate database
battery.charge: 100
battery.runtime: 5380
battery.voltage: 13.90
battery.voltage.high: 13.9
battery.voltage.low: 10.5
battery.voltage.nominal: 12
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 5161
driver.parameter.runtimecal: 1200,100,6120,9.1
driver.parameter.synchronous: no
driver.parameter.vendorid: 0665
driver.version: 2.7.4.1
driver.version.data: Q1 0.07
driver.version.internal: 0.28
input.frequency: 60.2
input.voltage: 120.6
input.voltage.fault: 120.8
output.voltage: 120.8
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 11
ups.productid: 5161
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
ups.vendorid: 0665
````
ups.conf
````
 [directups1000]
                                driver = nutdrv_qx
                                  port = auto
                              vendorid = 0665
                             productid = 5161
          default.battery.voltage.high = 13.9
           default.battery.voltage.low = 10.5
       default.battery.voltage.nominal = 12
                            runtimecal = 1200,100,6120,9.1
````
the default high voltage value is from upsc.  the high runtime calculation value is from a newegg customer report claiming a battery run time of 20 minutes (i assumed this was a maximum load) and the low values are from my own test using upslog.  there was a weird glitch in the data where the time jumped by 67.5 minutes:
[image]
since the unit was unplugged i assumed the battery charge change was linear and and dropped that interval to get the 6120 sec value for `runtimecal`.  any other suggestions for generating reference data?