networkupstools/nut #337: [HCL] CyberPower CP1500EPFCLCD supported by usbhid-ups

living180, 2016-10-30:
CyberPower CP1500EPFCLCD (European version of the CP1500PFCLCD):  http://www.cyberpower-eu.com/products/ups_systems/pfc-sinewave/cp1500epfclcd.htm

Shutdown sequence tested successfully via USB using NUT 2.7.3.

`upsc` output:
> battery.charge: 100
> battery.charge.low: 10
> battery.charge.warning: 20
> battery.mfr.date: CPS
> battery.runtime: 2640
> battery.runtime.low: 300
> battery.type: PbAcid
> battery.voltage: 24.0
> battery.voltage.nominal: 24
> device.mfr: CP1500EPFCLCD
> device.model: CRJB103.551
> device.serial: CPS
> device.type: ups
> driver.name: usbhid-ups
> driver.parameter.pollfreq: 30
> driver.parameter.pollinterval: 2
> driver.parameter.port: auto
> driver.parameter.synchronous: no
> driver.version: 2.7.3
> driver.version.data: CyberPower HID 0.4
> driver.version.internal: 0.39
> input.transfer.high: 260
> input.transfer.low: 170
> input.voltage: 237.0
> input.voltage.nominal: 230
> output.voltage: 269.0
> ups.beeper.status: disabled
> ups.delay.shutdown: 20
> ups.delay.start: 30
> ups.load: 19
> ups.mfr: CP1500EPFCLCD
> ups.model: CRJB103.551
> ups.productid: 0501
> ups.realpower.nominal: 900
> ups.serial: CPS
> ups.status: OL
> ups.test.result: No test initiated
> ups.timer.shutdown: -60
> ups.timer.start: -60
> ups.vendorid: 0764

`upsrw` output:
> [battery.charge.low]
> Remaining battery level when UPS switches to LB (percent)
> Type: STRING
> Maximum length: 10
> Value: 10
> 
> [battery.runtime.low]
> Remaining battery runtime when UPS switches to LB (seconds)
> Type: STRING
> Maximum length: 10
> Value: 300
> 
> [input.transfer.high]
> High voltage transfer point (V)
> Type: STRING
> Maximum length: 10
> Value: 260
> 
> [input.transfer.low]
> Low voltage transfer point (V)
> Type: STRING
> Maximum length: 10
> Value: 170
> 
> [ups.delay.shutdown]
> Interval to wait after shutdown with delay command (seconds)
> Type: STRING
> Maximum length: 10
> Value: 20
> 
> [ups.delay.start]
> Interval to wait before (re)starting the load (seconds)
> Type: STRING
> Maximum length: 10
> Value: 30

`upscmd -l` output:
> Instant commands supported on UPS [cyberpower]:
> 
> beeper.disable - Disable the UPS beeper
> beeper.enable - Enable the UPS beeper
> beeper.mute - Temporarily mute the UPS beeper
> beeper.off - Obsolete (use beeper.disable or beeper.mute)
> beeper.on - Obsolete (use beeper.enable)
> load.off - Turn off the load immediately
> load.off.delay - Turn off the load with a delay (seconds)
> load.on - Turn on the load immediately
> load.on.delay - Turn on the load with a delay (seconds)
> shutdown.return - Turn off the load and return when power is back
> shutdown.stayoff - Turn off the load and remain off
> shutdown.stop - Stop a shutdown in progress
> test.battery.start.deep - Start a deep battery test
> test.battery.start.quick - Start a quick battery test
> test.battery.stop - Stop the battery test

clepple, 2016-10-31:
Thanks for a very complete compatibility report!

Do all the measurements look reasonable to you? I'm not too familiar with the voltage readings one might expect on a non-120V system, but the difference between `input.voltage` and `output.voltage` seems large to me. Which is closer to the LCD reading?

living180, 2016-10-31:
`input.voltage` is accurate - `output.voltage` is bogus.  When on mains power the LCD shows the input voltage as being the same as the output voltage.  It looks like the US version has a similar issue with `output.voltage`:  http://networkupstools.org/ddl/Cyber_Power_Systems/CP1500PFCLCD.html.  I just did a quick test running `upsc` with the UPS disconnected from mains power, and `input.voltage` was `0.0` while `output.voltage` was `262.0`, which is still bogus.

clepple, 2016-11-02:
@living180 one more quick question: does `device.model` correspond to the serial number on the label?

living180, 2016-11-02:
No, I only found one label on the exterior case.  It has no description - just a barcode with some characters underneath, but it is the only thing that resembles a serial number.  It is different than `device.model` however, and there doesn't seem to be any relation between them.

closed 2016-11-03 by clepple

WillyTP, 2025-01-15:
Hello, as 15/01/2025 I report issue of wrong Output Voltage reading.
Updated UPS manufacturer page: [https://www.cyberpower.com/eu/en/product/sku/cp1500epfclcd](url)

upsc output:
battery.charge: 2
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 51
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500EPFCLCD
device.serial: CRMMYxxx (checked, it's correct)
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 5
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: DSM7-2-1-69057-NM-nano-4-C1-240105
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 260
input.transfer.low: 170
input.voltage: 232.0
input.voltage.nominal: 230
**output.voltage: 264.0**
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 15
ups.mfr: CPS
ups.model: CP1500EPFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: CRMMY2000017
ups.status: OL CHRG LB
ups.test.result: Done and warning
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764

Kind regards

clepple, 2025-01-16:
@WillyTP ask your NAS vendor to update their version of NUT.