networkupstools/nut #341: Tripplite SNMPWEBCARD ups.load 10 times too high

toolage, 2016-11-20:
I am running 2.7.4 using the snmp-ups. The value reported by ups.load is 10 times higher than the actual amount. Currently it is showing 160% when it is actually only 16%

Tripp-Lite SNMPWEBCARD running version 12.06.0064.2791.2532
Device is a Tripp-Lite SU2200RTXL2Ua UPS.

toolage, 2016-11-20:
battery.charge			100.00
battery.runtime			3300.00
battery.temperature		32.00
battery.voltage			56.00
device.mfr			Tripp Lite
device.model			SU2200RTXL2Ua
device.type			ups
driver.name			snmp-ups
driver.parameter.mibs		ietf
driver.parameter.pollfreq  	15
driver.parameter.pollinterval	2
driver.parameter.port		x.x.x.x
driver.parameter.snmp_version	v2c
driver.parameter.synchronous	       no
driver.version			2.7.4
driver.version.data		ietf MIB 1.5
driver.version.internal		0.97
input.bypass.frequency		59.90
input.bypass.phases		1.00
input.bypass.voltage		124.00
input.frequency			59.90
input.phases			1.00
input.transfer.high		150.00
input.transfer.low		65.00
input.voltage			124.00
input.voltage.nominal		120.00
output.current			0.20
output.frequency		59.90
output.phases			1.00
output.realpower		253.00
output.voltage			124.00
ups.beeper.status		muted
ups.firmware			19
ups.firmware.aux		12.06.0064
ups.load			150.00  (this value should show as 15% load)
ups.mfr				Tripp Lite
ups.model			SU2200RTXL2Ua
ups.status			OL

clepple, 2016-11-20:
The [IETF MIB RFC](https://tools.ietf.org/html/rfc1628) seems to indicate that the units should be 1% rather than 0.1%. I saw what looks to be a newer SNMPWEBCARD firmware when doing a quick Google search, but I didn't check for a changelog to see if this has been fixed. I don't have any experience with these cards, so check around to see if there are any gotchas if you upgrade (losing other functionality; not being able to downgrade; etc.)

Also, if you just want to fix it locally, you can change the `1.0` scale factor in `drivers/snmp-ups.c` to `0.1`: https://github.com/networkupstools/nut/blob/v2.7.4/drivers/ietf-mib.c#L174

jimklimov, 2022-04-25:
HCL/DDL part of the issue was handled with https://github.com/networkupstools/nut-ddl/pull/21 and #1396

Special handling for 10x the load I think was not done, at least not in IETF MIB