networkupstools/nut #345: Python 2/3

clepple, 2016-11-28:
Portions of NUT rely on the first `python` command in `$PATH` being Python 2. (Fortunately, this tends to only affects those building from Git, not from a source tarball.)

Before we jump in and upgrade everything to Python 3, let's summarize what the various distributions ship by default:

* Debian 8 (jessie): minimal install does not provide Python
  * `python-minimal` depends on `python2.7` package, which provides `/usr/bin/python2` and `/usr/bin/python2.7` 
* Ubuntu 12.04 (precise): same as Debian, but `python-minimal` is installed by default
* Ubuntu 14.04 (trusty): same as Debian, but `python-minimal` is installed by default
* Ubuntu 16.04 (xenial): same as Debian (none installed by default)
* Arch (version?): `python2` in path; installed by default?
* Mac OS X 10.11: `/usr/bin/python` points to `python2.7`; **no `python2` symlink**
  * Fink provides a `python2` symlink as part of the `python` meta-package

The short answer is that `python2.7` seems to be the only name that won't cause confusion.

We could probably run 2to3 and/or use `six` to be able to run on whatever version is available. Thoughts?

(If you have proposed changes, please commit them to a branch other than `master` first, thanks!)

aquette, 2016-11-28:
@clepple: not fully exact

* Debian 8 (jessie): minimal install does not provide Python
  * `python-minimal`, which provides symlinks `/usr/bin/python2` and `/usr/bin/python`, depends on `python2.7-minimal` package, which provides `/usr/bin/python2.7` 

=> so `/usr/bin/python` *is* avail... once `python-minimal` is installed.

Other notes:
* the only failure is related to autogen.sh and Augeas ups.conf lens generation if "python" is not available, due to distribution issues.
* other python scripts calls (nut-snmpinfo.py and gitlog2changelog.py) simply generate warnings in case "python" is not available

Hence my crappy "sed" proposal, which ends up with creating a sane default scripts/augeas/nutupsconf.aug.in for distribution.

zykh, 2016-11-28:
As an archer.. Arch Linux:
- python is not installed (neither 2 nor 3) by default ('base' group),
- python 2 can be installed with 'python2' package, which provides `/usr/bin/python2` symlink pointing to `/usr/bin/python2.7`.
- [before 18/10/2010](https://www.archlinux.org/news/python-is-now-python-3/), `/usr/bin/python` pointed to python 2 (at that time `/usr/bin/python2.6`, not 2.7), now to python 3 (package 'python').

clepple, 2016-11-29:
@aquette and @zykh: thanks, I edited the original description. The "installed by default" part is fuzzy - we advise people to install the build-deps for their distribution, so they don't just have a minimal install at that point.

For our purposes, Python 2.7 is a superset of Python 2.6. I also don't think we need to worry about distributions more than, say, five years back for bootstrapping like this (given that we have release tarballs from that time).

The-Compiler, 2016-11-29:
Making the script work with python 3 is trivial, see #346.

PR #346 (2016-12-06): Make gen-nutupsconf-aug.py work with Python 3
This makes the script work with Python 2.6+.
Fixes #345.

cc @zman0900
commit eea0c5e9 (2016-12-06): Make gen-nutupsconf-aug.py work with Python 3

This makes the script work with Python 2.6+.
Fixes #345.
closed 2016-12-06 by clepple