networkupstools/nut #348: Smart-UPS 750 input.voltage

MurtFolak, 2016-12-16:
Hello,

i can't get the "input.voltage"
in [http://networkupstools.org/ddl/APC/Smart-UPS_750.html](url) it is listed
but here it looks like this:

`pi@raspberrypi:~ $ sudo upsc apc750
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 8760
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.3
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Smart-UPS 750
device.serial: Axxxx
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.version: 2.7.2
driver.version.data: APC HID 0.95
driver.version.internal: 0.38
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.firmware: UPS 08.3 / ID=18
ups.mfr: American Power Conversion
ups.mfr.date: 2012/10/12
ups.model: Smart-UPS 750
ups.productid: 0003
ups.serial: Axxxxx
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
`

how can I get the Value?

Thank you!

clepple, 2016-12-17:
(edited to adjust the formatting)

Note that the `ups.firmware` format is similar to the following models:

 * http://networkupstools.org/ddl/APC/Smart-UPS_3000_2U.html
 * http://networkupstools.org/ddl/APC/Smart-UPS_SMT3000RMI2U.html
 * http://networkupstools.org/ddl/APC/Smart-UPS_C1500.html

It is possible that your UPS has a MODBUS interface, which is not yet implemented in NUT (see issue #139)

dynek, 2018-02-12:
Same problem with my SMT750I - do we have more info since end 2016?

```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 3480
battery.runtime.low: 120
battery.type: PbAc
battery.voltage: 27.0
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion
device.model: Smart-UPS 750
device.serial: AS1243212098
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 10
driver.parameter.port: /dev/ttyS1
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: APC HID 0.96
driver.version.internal: 0.41
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: UPS 08.3 / ID=18
ups.mfr: American Power Conversion
ups.mfr.date: 2012/10/24
ups.model: Smart-UPS 750
ups.productid: 0003
ups.serial: AS1243212098
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d
```

Kh3nsu, 2023-03-19:
Hello,
Sorry for digging up this really really Old issue. But is there a fix for this? As the Problem still persists with my SMT750I. 
I know there is a smart card for the ups which is around 80€ (https://www.office-partner.de/apc-ap9620-1793700) which delivers legacy communication protocol. 

Any Otter solutions?

jimklimov, 2023-03-19:
Which NUT version are you running? Some PRs for better SNMP support landed last year, so a master-branch build may fare better. At least you can try a quick check from the build workspace, without replacing your installation: https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

jimklimov, 2023-03-19:
#1868 may be of interest...

As an APC customer, you can try asking for the protocol.

Seqway, 2024-01-04:
I will once again push this up here.
Any news to get voltage values via USB ? How can I help for USB connection ?